dresden-elektronik/deconz-rest-plugin #3648: Door/Window Sensors Dropping

majorsl, 2020-11-13:
## Describe the question or issue you are having

I'm having issues with door/window sensors just dropping out of deconz/phoscon and I don't understand why. Batteries are 100% and the particular sensor in my screen shot is ~15ft/4.5m from the Conbee II.

I have no repeaters (yet) but really only have 16 sensors so I would hope it could handle those. The sensors that are dropping are SNZB-04, if that makes a difference, but they are on the supported list.  I have a few CentraLite Series 3 that seem to be very solid as are my Temp/Flood Sensors.

I'm also careful to have my Wifi and Zigbee network at opposite end of the spectrum.  I live in the country and no other wireless networks are in range.

## Screenshots

[image]
## Environment

 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.05.86 / 10/15/2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

I'm also fairly new to this, so am unsure if there is a better way to reestablish communication other than re-pairing the device.

Smanar, 2020-11-13:
You have it inside the fridge ?

majorsl, 2020-11-13:
No, it is on the outside.  One of the others is on an inside door not too far away from the Conbee II as well.  That is what is confusing to me - I have sensors double or triple the distance and they seem to work well.  I have the same sensors on windows that go weeks without being opened or closed, but respond immediately if used.  I have some plugs on the way and they'll be repeaters and maybe help, but I just don't understand why so few devices are a problem.

Smanar, 2020-11-13:
I don't see the link on your capture, just by curiosity the faulties devices are not using an ikea router, and solid one direct connexion ?

majorsl, 2020-11-13:
I do not have the plugs yet and no repeater.  The only thing they are connecting to is directly to the Conbee II.

It had a link yesterday.  This morning no link.  Nothing changed and the other devices are working.

majorsl, 2020-11-14:
This morning I noted that it reconnected itself some time overnight. I didn't do anything that would have done this. Very strange.

[image]

Smanar, 2020-11-14:
So direct connexion, and the link looks good, not notmal you loose this device ...

closed 2020-12-31 by stale[bot]