dresden-elektronik/deconz-rest-plugin #3653: Problem after upgrade to 26660700

rasssta, 2020-11-14:
## Describe the bug

After upgrading to 26660700 my lights would not change status in reality, even though I could set them to Off via the API. But they would show as off in the UI, but the lights would still be on. Sensors would not connect. I've tried to upgrade and downgrade firmware and deCONZ without success. After restore of a backup I could get connectivity to 2 sensors, but no other devices. I've left it like this for 1 week and it did not improve.

## Steps to reproduce the behavior

Upgraded from 26580700 to 26660700

## Expected behavior

Sensors and lights should work, sensors and lights should be able to add again.

## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.05.88
 - Device: ConBee II 
 - Do you use an USB extension cable: no (worked flawlessly w/o extension before the upgrade)
 - Is there any other USB or serial devices connected to the host system? If so: USB-SSD

## deCONZ Logs

`21:15:04:941 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
21:15:04:949 CTRL. 3.27.221:15:05:079 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
21:15:05:079 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
21:15:05:080 dev /dev/ttyAMA0
21:15:05:080 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
21:15:05:282 parent process -bash
21:15:05:283 gw run mode: normal
21:15:05:283 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
21:15:05:285 DB sqlite version 3.27.2
21:15:05:286 DB PRAGMA page_count: 33
21:15:05:286 DB PRAGMA page_size: 4096
21:15:05:286 DB PRAGMA freelist_count: 0
21:15:05:286 DB file size 135168 bytes, free pages 0
21:15:05:287 DB PRAGMA user_version: 7
21:15:05:287 DB cleanup
21:15:05:288 DB create temporary views
21:15:05:288 DB view [0] created
21:15:05:289 DB view [1] created
21:15:05:289 DB view [2] created
21:15:05:290 DB view [3] created
21:15:05:290 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
21:15:05:291 sql exec SELECT key FROM config2
21:15:05:291 sql exec SELECT key,value FROM config2
21:15:05:292 Load config UTC: 2020-11-14T21:14:23 from db.
21:15:05:292 Load config announceinterval: 45 from db.
21:15:05:292 Load config announceurl: https://phoscon.de/discover from db.
21:15:05:292 Load config apiversion: 2.05.88 from db.
21:15:05:292 Load config bridgeid: 00212EFFFF05506D from db.
21:15:05:293 Load config datastoreversion: 93 from db.
21:15:05:293 Load config dhcp: true from db.
21:15:05:293 Load config disablePermitJoinAutoOff: false from db.
21:15:05:293 Load config discovery: true from db.
21:15:05:293 Load config factorynew: false from db.
21:15:05:294 Load config fwneedupdate: false from db.
21:15:05:294 Load config fwupdatestate: idle from db.
21:15:05:294 Load config fwversion: 0x26660700 from db.
21:15:05:294 Load config gateway: 127.0.0.1 from db.
21:15:05:294 Load config group0: 65520 from db.
21:15:05:295 Load config groupdelay: 50 from db.
21:15:05:295 Load config gwpassword: $1$8282jdkm$9h/SvacNU7rIrU629CBQp1 from db.
21:15:05:295 Load config gwusername: delight from db.
21:15:05:295 Load config homebridge: not-managed from db.
21:15:05:296 Load config homebridge-pin:  from db.
21:15:05:296 Load config homebridgepin:  from db.
[... 315 lines cut ...]
21:16:32:164 Idle timer triggered
21:16:33:364 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:35:662 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
21:16:35:762 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:38:162 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:40:555 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:42:963 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:45:363 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:45:662 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
21:16:47:762 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:50:163 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:52:562 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:54:962 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
21:16:55:663 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
21:16:57:166 CTRL db store gui node 00:21:2e:ff:ff:05:50:6d
21:16:57:173 saved node state in 0 ms
21:16:57:176 sync() in 3 ms
21:16:57:177 REST API plugin shutting down
21:16:57:178 DB save zll database items 0x000000C1
21:16:57:178 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('1', 'normal', '00:21:2e:ff:ff:05:50:6d-01', 'Configuration tool 1', '65520', '1', 'ConBee II', 'dresden elektronik', '0x26660700', '{"attr/id":"1","attr/lastannounced":null,"attr/lastseen":"2020-11-14T21:16Z","attr/manufacturername":"dresden elektronik","attr/modelid":"ConBee II","attr/name":"Configuration tool 1","attr/swversion":"0x26660700","attr/type":"Configuration tool","attr/uniqueid":"00:21:2e:ff:ff:05:50:6d-01","state/reachable":true}')
21:16:57:185 DB saved in 6 ms
21:16:57:189 sync() in 2 ms
`

## Additional context

Smanar, 2020-11-14:
But you have an empty database ? There is no device at all ?

rasssta, 2020-11-14:
Yes. I've started from scratch now, but I'm unable to add any device that was previously working fine.

indomus, 2020-11-15:
Similar. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3645

rasssta, 2020-11-15:
I’ve tested the conbee stick on rpi4 with usb-ssd and rpi3 with sd card. Same issue.

Smanar, 2020-11-15:
You are not using USB 3.0 ?
Can you make a try without the SSD / with extention / On a windows/Ubuntu machine ?

Mimiix, 2020-11-16:
@indomus Me and @kane610 both requested logs from you. You did not provide, neither did anyone else. Please don't bring up the assumption if it's related as we have nothing on your issue.

Mimiix, 2020-11-16:
@rasssta I suggest using a Extension cable. With a USB SSD it is very likely that's the issue.

rasssta, 2020-11-16:
Please read all of the information I have submitted. I have tried without USB SSD as well, same issue.

rasssta, 2020-11-22:
Upgrading the firmware to 26670700 did not make it work.
Using an extension cable on USB3 port did not work.

However, using an extension cable on USB2 port together with 26670700 works (FINALLY!).

closed 2020-11-22 by rasssta