dresden-elektronik/deconz-rest-plugin #3654: ADEO Lexman Télécommande (Leroy Merlin)

diliact, 2020-11-14:
https://www.leroymerlin.fr/v3/p/produits/telecommande-pour-ampoules-connectees-rgb-ctt-zigbee-lexman-enki-e1506791204

## Device

- Product name: Télécommande (https://www.youtube.com/watch?v=cdOHHqwMkzU#t=66)
- Manufacturer: ADEO
- Model identifier: LXEK-5
- Device type : 
  - Remote

## Screenshots

Mimiix, 2020-11-14:
Missing screenshots of all clusters.

diliact, 2020-11-14:
Ok, here they are.

[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-11-14:
No, it s perfect now ^^, thx.
BTW this remote is realy ugly ...

diliact, 2020-11-15:
Thank you. Yes, it is not as sleek and seamless as the dimmer switch from Philips.

Smanar, 2020-11-16:
@diliact do you have an Unix machine to test the code ?
I will explain how to do that, but you need Unix and a full OS, no HA for exemple

diliact, 2020-11-16:
I have an old Raspberry Pi, the first one, is it OK?

Smanar, 2020-11-17:
It seem yes

> upported Raspberry Pi models and distributions
>     Raspberry Pi 1, 2B, 3B, 3B+ and 4B
>     Raspbian Jessie
>     Raspbian Stretch
>     Raspbian Buster

Now the big part ^^ 

So you need to instal deconz on it then take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch adeo https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

And after that you will just need to swap the key from your dev machine to the raspberry for test, don't worry, devices are stored locally, not on the gateway.

So the modification don't do lot of thing yet, it just permit the device creation in the API. But the device can be invisible in phoscon, so to see it, 2 ways possibles
- third app
- using direclty the API

But it seem you have log in phoscon now (not tried on my side yet) in the "help" part, can be enought.

After that, you will see an error message every time you press a button, can you show me them.

diliact, 2020-11-18:
Thank you, I will try this at the end of the week. Do I need to connect a display to the Raspberry Pi or can I do everything on SSH?

Smanar, 2020-11-18:
All is doable with VNC, I m using it like this.

Don't be afraid by all the manipulations to do ^^, you will see, after a try, it s not so difficult.

I m on discord too (and french too), but have a new video game and need to find christmas gift, so it s posible need some time to answer :)

diliact, 2020-11-22:
Ok, thank you for your help, first results are:
```
pi@raspberrypi:~ $ sudo apt install deconz-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package deconz-dev
```

Smanar, 2020-11-23:
You have already installed deconz ?

https://www.phoscon.de/en/conbee2/install#raspbian

Then 
```
sudo apt-get update
sudo apt-get upgrade
sudo apt install deconz-dev
```

diliact, 2020-11-24:
Thank you, I've managed to install everything (raspbian full instead of lite is a better choice for TightVNC).
But deCONZ has no GUI?

[image]

In the first message, I used deCONZ on my desktop.

I've barely managed to launch Chromium on my old Raspberry, it is better on the desktop but I still cannot add the Lexmark remote in Phoscon App (phoscon.de/app on the desktop or on the Raspberry via TightVNC):

[image]

Smanar, 2020-11-24:
Ok so first you are lucky if you have the GUI, TightVNC is not compatible with deconz, you need to use the integrated VNC, this one you can enable using raspi-config cf https://github.com/dresden-elektronik/deconz-rest-plugin/issues/343

The usefull part for log are INFO_L2 and Error _L2

You have used all the command line to compile the code and replace the existing lib file by the compiled one ?

```
git clone --branch adeo https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

It seem the device is too in the API (on deconz the device have the model id as name), but easy to see.

If all is fine, you will have in log some error message when you press a button.

Smanar, 2020-11-29:
@diliact , still have problem ?

diliact, 2020-11-29:
I haven't tried again, I will give it a try tomorrow but I know the GUI was very slow on my raspberry 1, it was not easy to identify the right equipment.
If you want I can send you one remote that you can keep.

Smanar, 2020-11-30:
No, you will see it s not so hard, and the file to edit is a json one, so you will be able to make change yourself.

And you don't need to identify the right equipement, just include it, enable log, press a button and copy/paste log starting by
>[INFO] - No button map for
>[INFO] - No button handler for

diliact, 2020-11-30:
Thank you.

I have many sensors and 5 Osram switches, there are many messages.

By searching for "LEXK", you will find some interesting lines but not as expected.

23:06:25:910 Wait 5s till query finished
23:06:26:909 Wait 4s till query finished
23:06:27:356 Node data 0x84182600000f3452 profileId: 0x0104, clusterId: 0x0006
23:06:27:362 0x84182600000F3452: added ZCL value 0x03/0x0006/0x0000
23:06:27:372    payload: 00001000
23:06:27:668 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 7, node: 0xAAFB
23:06:27:909 Wait 3s till query finished
23:06:28:908 Wait 2s till query finished
23:06:29:908 Wait 1s till query finished
23:06:30:823 DB save zll database items 0x00000001
23:06:30:829 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('6', 'normal', '7c:b0:3e:aa:00:ae:74:f0-03', 'On/Off plug-in unit 6', '65520', '3', 'Plug 01', 'OS
RAM', 'V1.04.12', '{"attr/id":"6","attr/lastannounced":null,"attr/lastseen":"2020-11-30T23:06Z","attr/manufacturername":"OSRAM","attr/modelid":"Plug 01","attr/name":"On/Off plug-in unit 6","attr/swversion":"V1.04.12","attr/type":"On/Off
plug-in unit","attr/uniqueid":"7c:b0:3e:aa:00:ae:74:f0-03","state/alert":null,"state/on":true,"state/reachable":true}')
23:06:30:884 DB saved in 61 ms
23:06:31:008 poll node 84:18:26:00:00:0e:4e:ad-03
23:06:31:012 Poll light node On/Off plug-in unit 2
23:06:31:023 Idle timer triggered
23:06:31:032 create binding for attribute reporting of cluster 0x0006
23:06:31:043 queue binding task for 0x84182600000E4EAD, cluster 0x0006
23:06:31:047 Force binding of attribute reporting for node On/Off plug-in unit 2
23:06:31:491 read attributes of 0x84182600000E4EAD cluster: 0x0006: [
23:06:31:511 0x0000
23:06:31:517 ]
23:06:31:527 Poll APS request 221 to 0x84182600000E4EAD cluster: 0x0006
23:06:31:800 Poll APS confirm 221 status: 0x00
23:06:31:808 Erase task req-id: 221, type: 19 zcl seqno: 1 send time 0, profileId: 0x0104, clusterId: 0x0006
23:06:31:984 Node data 0x84182600000e4ead profileId: 0x0104, clusterId: 0x0006
23:06:32:004 0x84182600000E4EAD: added ZCL value 0x03/0x0006/0x0000
23:06:32:040 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
23:06:32:058 Wait 1s till query finished
23:06:32:359 Websocket 192.168.0.32:65101 send message: {"e":"changed","id":"1","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = -1098831000)
23:06:32:480 discard group state push for 1: state/any_on (already pushed)
23:06:32:525 configure reporting rq seq 2 for 0x84182600000E4EAD, attribute 0x0006/0x0000
23:06:32:768 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xAAFB
23:06:32:908 Wait 2s till query finished
23:06:33:658 poll node 84:18:26:00:00:0e:66:b8-03
23:06:33:663 Poll light node On/Off plug-in unit 3
23:06:34:384 Poll APS request to 0x84182600000E66B8 cluster: 0x0006 dropped, values are fresh enough
23:06:34:396 Idle timer triggered
23:06:34:411 create binding for attribute reporting of cluster 0x0006
23:06:34:427 queue binding task for 0x84182600000E73E3, cluster 0x0006
23:06:34:432 Force binding of attribute reporting for node On/Off plug-in unit 4
23:06:35:158 Wait 1s till query finished
23:06:35:942 Wait 1s till query finished
23:06:37:500 Node data 0x84182600000e4ead profileId: 0x0104, clusterId: 0x0006
23:06:37:512 0x84182600000E4EAD: update ZCL value 0x03/0x0006/0x0000 after 0 s
23:06:37:524    payload: 00001000
23:06:37:724 poll node 84:18:26:00:00:0e:73:e3-03
23:06:37:737 Poll light node On/Off plug-in unit 4
23:06:37:747 Idle timer triggered
23:06:37:758 create binding for attribute reporting of cluster 0x0006
23:06:37:770 queue binding task for 0x7CB03EAA00AE74F0, cluster 0x0006
23:06:37:785 Force binding of attribute reporting for node On/Off plug-in unit 6
23:06:38:533 read attributes of 0x84182600000E73E3 cluster: 0x0006: [
23:06:38:549 0x0000
23:06:38:562 ]
23:06:38:578 Poll APS request 246 to 0x84182600000E73E3 cluster: 0x0006
23:06:38:611 Wait 1s till query finished
23:06:39:373 Wait 0s till query finished
23:06:40:350 Idle timer triggered
23:06:40:387 Force binding of attribute reporting for node On/Off plug-in unit 3
23:06:41:674 Wait 1s till query finished
23:06:42:488 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
23:06:42:527 Idle timer triggered
[... 416 lines cut ...]
23:15:12:863 Poll APS request to 0x84182600000F3452 cluster: 0x0006 dropped, values are fresh enough
23:15:14:079 Mgmt_Lqi_req zdpSeq: 255 to 0x84182600000F3452 start index 0
23:15:14:085 APS-DATA.request id: 171, addrmode: 0x03, addr: 0x84182600000f3452, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
23:15:15:641 command queue EVENT_TIMEOUT, cmd: CMD_LINK_ADDRESS, seq: 30
23:15:16:583 Node 0x84182600000E4EAD is known by 5 neighbors, last seen 49 s
23:15:18:643 poll node 7c:b0:3e:aa:00:ae:74:f0-03
23:15:18:648 Poll light node On/Off plug-in unit 6
23:15:18:667 command queue EVENT_TIMEOUT, cmd: CMD_APS_DATA_CONFIRM, seq: 32
23:15:18:681 APS-DATA.confirm id: 171, status: 0x00 SUCCESS
23:15:18:697 APS-DATA.confirm request id: 171 -> confirmed, timeout 31864392
23:15:19:758 Poll APS request to 0x7CB03EAA00AE74F0 cluster: 0x0006 dropped, values are fresh enough
23:15:19:786 APS-DATA.confirm id: 0, status: 0x00 SUCCESS
23:15:19:798 APS-DATA.confirm id: 0, status: 0x00, match: 0
[... cut ...]
23:15:21:829     * neighbor: 0x00212EFFFF04F058 (0x0000), LQI: 94, relation: 0x02 rxOnWHenIdle: 1
23:15:22:708 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
23:15:24:524 Websocket 192.168.0.32:65101 send message: {"attr":{"id":"5","lastannounced":"2020-11-24T02:06:38Z","lastseen":"2020-11-30T23:15Z","manufacturername":"OSRAM","modelid":"Plug 01","name":"On/Off plug-in unit 5","swversion":"V1
.04.12","type":"On/Off plug-in unit","uniqueid":"84:18:26:00:00:0f:34:52-03"},"e":"changed","id":"5","r":"lights","t":"event","uniqueid":"84:18:26:00:00:0f:34:52-03"} (ret = 347)

KipK, 2020-12-01:
Hi. 
.I also got a bunch of those to test with. 
Tell me if you need more logs, I can provide .

Mimiix, 2020-12-01:
@diliact I've edited your last post to make it more readable 👍

Smanar, 2020-12-01:
And you are right, nothing special.
you have pressed a button during your tests ?

Perhaps when you enable "Info_L2", you need to enable "Info" too in same time, as you have osram, they need to react same when you use them (but not with same message)

diliact, 2020-12-02:
@Mimiix thank you.

Yes, I have pressed each button two or three times, on the second part (after 23h10) I have enabled nearly all messages (but OTA, TLINK and another one). Osram switches (230V sockets) are normally controlled by Jeedom, I cannot control them when plugging the Conzbee dongle on the Raspberry.

Smanar, 2020-12-02:
Ok, something have failed, there is realy nothing that come back.
Do you know how to make bind ?
If you can use the "bind dropbox"
- drag and drop the cluster 0x0006 (grey) from the remote in "source"
- drag and drop the "gateway titlebar" in "destination.

Then press the button that serve for on/off, and look in log if you can see something about this device.

vader781, 2020-12-08:
Hi, I also have this remote.
Was unable to pair it with deconz/Phoscon so far on my jeedom...
I hope it can be further integrated.

Smanar, 2020-12-09:
@vader781 you don't have an unix machine to make test by hazard ?

nono56, 2020-12-09:
J'essaierai de trouver le temps et de tester demain @Smanar 
En gros, il faut installer quoi ?
juste ça?
sudo apt install deconz-dev
git clone --branch adeo https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

Ensuite, je lance Deconz et je debig avec ca?
The usefull part for log are INFO_L2 and Error _L2

Merci

Smanar, 2020-12-09:
Yep, and log are visible in deconz now.
You need to have 
```
[INFO] - No button map for
[INFO] - No button handler for
```

It seem it don't work correclty for @diliact, so the bindind is probably not well done by my code.

And BTW, to speak french, you can pm me on discord (I have same username), issues are for all users, and  a lot don't speak french.

nono56, 2020-12-09:
OK, thanks.
My virtual machine is ready.
I will test it tomorrow.

nono56, 2020-12-09:
I have tested it quickly...
No success :
[image]
when i click on connect, nothing happens
If i click on Phoscon app, it ask me a password...
Any idea?

Smanar, 2020-12-10:
I think it s VM configuration, you have set the USB port ?
The conbee appear on the VM ?

nono56, 2020-12-10:
Hi.
Yes the conbee is OK.
Just the interface wich don't work.
I will prepare a Raspberry this weekend.

vader781, 2020-12-23:
I tried again to pair the remote in my fully functional Jeedom with Deconz/Phoscon without any success. Logs don't show anything. No device being added.
I could only do tests that won't interfere with configuration on the RPI, I don't want to mess everything up in Jeedom.

System : RPI 4
Jeedom : 4.0.61
Deconz plugin version : 2020-10-28 18:46:23
Phoscon : 2.05.88 / 15/10/2020
Firmware : 26680700

Smanar, 2020-12-24:
But you have made test with modified/recompiled code ?

vader781, 2020-12-27:
Hello !
I was able to set up a VM with Ubuntu and do the installation part.
Now in deCONZ I have the device "LXEK-5 2" showing.
Nothing relevant in the logs with Info / Info_L2 / Error / Error_L2 selected.
So I tried binding. But I have the error : Failed: timeout
[image]
what can I do now ?

Smanar, 2020-12-28:
If I m right you need to see it in the api too (in jeedom if you link it with this deconz version).

You can see it too in log 
>23:08:26:755 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('2', 'LXEK-5 2', 'ZHASwitch', 'LXEK-5', 'ADEO', '08:6b:d7:ff:fe:1b:c6
:80-01-1000', '2.8', '{"buttonevent":null,"lastupdated":null}', '{"battery":97,"group":null,"on":true,"reachable":true}', '{"d":2048,"ep":1,"in":[0,1,4096],"out":[6,8],"p":260}', 'normal', '1')

or at this url 
`http://IP:PORT/api/KEY/sensors`
IP and PORT same than for phoscon, and KEY is one api key (you can't the same used by jeedom, because this one is stored on the production machine.

To get a new api key
`curl http://127.0.0.1:80/api/ -X POST -d '{"devicetype": "TestKey"}'`

>So I tried binding. But I have the error : Failed: timeout
Remember it s battery device, so you need to awake it (just with using it) in same time you send request.

But it s strange  you haven't this kind of log
```
[INFO] - No button map for
[INFO] - No button handler for
```

vader781, 2020-12-30:
It's here in the API :
[image]

Binding is still failing even when pressing all kind of buttons. I'm not sure I do it correctly : 
I drag the 0006  On/Off entry to source
[image]
Then I drag the "HA  Home Automation" grey zone (of the CombeeII)  to destination
[image]

I also tried to bing the cluster 300  Color control and here it's successful

But still the only logs showing about the device are few : "Force read attributes for ZHASwitch SensorNode LXEK-5 2".
Whatever button pressed does not produce any log.

Thanks again for helping me out.

vader781, 2020-12-30:
Oh I looked back in the logs and I have one (I don't know how I managed to get that one) : 
_[INFO] - No button map for: LXEK-5 endpoint: 0x01 cluster: 0x0001 command: 0x0A payload[0]: 021_

And here is one finally : 
_[INFO] - No button map for: LXEK-5 endpoint: 0x01 cluster: ONOFF (0x0006) command: OFF (0x00) payload[0]: 000_
ON button doesn't seem to make any logs...

I cannot manage to get more, binding is always failing with timeout except some very rare times.

Smanar, 2020-12-30:
Don't worry for binding, for the moment, was a supposition.
You have make bind on cluster 0x0300, and it seem only 0x0006 is working.

Cluster 0x0001 is for battery.

Have added the button "on" and "off" according to your log.
To make try, have supposed the "set level" one.

The next try will be easy, if you have a recent deconz version.
You can continue yourself if it work

Just find the file "button_maps.json" in "/usr/share/deCONZ/devices" for me. And add that https://github.com/Smanar/deconz-rest-plugin/commit/188caaf3dd587519c6fef3c6991e18b2dd2203fd

vader781, 2021-01-06:
Hello,
My Ubuntu VM somehow crashed so I will have to setup a new environment :(  Will keep you informed.
Just to be sure, once I will have updated the JSON file, do I need to run any command then ?
How to continue testing ?
For info we can see all capabilities of the remote in the guide [here](https://media.adeo.com/marketplace/82473189/98916bb7-244d-4acc-87ca-9882c9990c8e.pdf ).

Smanar, 2021-01-07:
Hello, happy new year ^^

You need to use the recompiled version to have the device included in the API
You need to edit the json to use the command of the device.

Then just use the on/off command, or the set level one with pressing the good button on the remote, there is only thoses one ATM.

vader781, 2021-01-08:
Hi,
I was able to setup everything like before. I added the JSON part.
This time objects are only showing ID not the device name, anyaway when I click on OFF button it is blinking green on the device :
[image]
Other buttons don't do anything.
Logs are less important too this time as well (though I checked Info, Info_L2, error, error_L2).
Only logs about the remote is when I press OFF button : " 21:43:05:436 APS-DATA.indication from child 0x6507 " (works every time).

Smanar, 2021-01-09:
>his time objects are only showing ID not the device name

I think the device is invisible on the api too ? For me the inclusion have failed. But you don't need to include it every time, when you are trying a new recomplied code, you just need to restart deconz.

vader781, 2021-01-21:
Hello again, sorry for the delay.
I can confirm the remote is not included in the API this time.
How can I perform the inclusion ? (tried reseting the remote several times it doesn't work)

I confirm I did the steps: install deconz-dev, got the adeo git branch, compile and cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

Smanar, 2021-01-22:
From the documention, it s using the little hole whith a pin up to the led blink for 3 time.

If you are using the GUI for test, can you share the log during the inclusion ? (on deconz/help/log with "info", "info-l2" "error"

vader781, 2021-03-07:
Hello, sorry I had not much time to dig into this.
I confirm I have the same model, I removed the node and re-included it and it is now showing again with title "LXEK-5 2" plus showing in the API.

Looking at the logs and TADA :
_**18:22:01:413 [INFO] - No button handler for: LXEK-5 endpoint: 0x01 cluster: 0x0001 command: 0x0A payload[0]: 021**_

Other logs showing (I tried different buttons) : 
A buch of those : 
_18:22:01:412 Force binding of attribute reporting for sensor LXEK-5 2_
_18:22:24:993 Force read attributes for ZHASwitch SensorNode LXEK-5 2_
And also one : 
_18:22:07:358 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('2', 'LXEK-5 2', 'ZHASwitch', 'LXEK-5', 'ADEO', '08:6b:d7:ff:fe:1b:db:0e-01-1000', '2.8', '{"buttonevent":null,"lastupdated":null}', '{"battery":100,"on":true,"reachable":true}', '{"d":2048,"ep":1,"in":[0,1,4096],"out":[6,8],"p":260}', 'normal', '1')_

I hope we can continue the integration work.

Smanar, 2021-03-07:
Welcome back ^^.

I have just updated the code with the last version from official github

```
cd deconz-rest-plugin
git pull
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

No need to re-include the device (just close deconz during file replacement), you have no more logs when using button ?

> 18:22:01:413 [INFO] - No button handler for: LXEK-5 endpoint: 0x01 cluster: 0x0001 command: 0x0A payload[0]: 021

Unfortunately cluster 0x0001 is for battery, better to find log about 0x0006 0x0008 or 0x0300.
You don't see specific logs when using it, if not, try with enabling APS or ZCL.

vader781, 2021-03-12:
Hello,
I updated the code.
Tonight nothing at all in the logs (every kind of log enabled) when pressing all buttons. Just :
_01:25:42:677 SensorNode id: 2 (LXEK-5 2) available_
_01:26:08:867 Force read attributes for ZHASwitch SensorNode LXEK-5 2_

To confirm the device infos :
Device model : HR-C99C-Z-C045
NWK : 0x40ff
IEEE : 0x086bd7fffe1bdb0e
Manufacturer : 0x1277

Smanar, 2021-03-12:
I m sorry I have checked the code again, and I don't see what is missing, this device have nothing special for what I have see.

I m searching some log with 

> [INFO] - No button map for
> [INFO] - No button handler for

Can you share a more complete log during a button press ? using PasteBin for exemple, there is no sensibles informations in log, else I m on discord Smanar.

BTW
>Manufacturer : 0x1277

??? it s 0x0000 on the capture on the first post.

bemardev, 2021-03-12:
Hi guys,
Having my hands on this remote, I tried to pair it to deconz too.
**- Model name :** HR-C99C-Z-C045
**- IEEE :** 0x000D6FFFFEC8A078
**- NWK :** 0x8F8F
**- Manufacturer code :** 0x1277

@Smanar, after trying on my dev laptop with your latest adeo branch, I saw my remote did not have the same MAC prefix as mentionned here, so I add emberMacPrefix too in de_web_plugin.cpp and rebuild libde_rest_plugin in consequence.

<details>
<summary>This is the log file when I paired it on my controller and playing with all the remote buttons.
(log with Info, Info L2, Error, Error L2, APS, APS_L2 and ZCL)</summary>
<p>

```
18:00:26:671 send permit join, duration: 59
18:00:26:671 APS-DATA.request id: 159, addrmode: 0x02, addr: 0xfffc, profile: 0xA1E0, cluster: 0x0021, ep: 0xF2 -> 0xF2 queue: 1 len: 6 tx.options 0x00
18:00:26:671 send GP proxy commissioning mode
18:00:26:672 enqueue event config/localtime for /config/
18:00:26:813 APS-DATA.request id: 161, addrmode: 0x03, addr: 0x00212effff059d21, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 2 len: 2 tx.options 0x00
18:00:26:827 APS-DATA.confirm id: 161, status: 0x00 SUCCESS
18:00:26:828 APS-DATA.confirm request id: 161 -> confirmed, timeout 1615568426
18:00:26:831 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 217, rssi: 24
18:00:26:832 	asdu: 0d00070101219d05ffff2e2100333d2401008d15000f18120001eb
18:00:26:833 APS-DATA.indication request id: 161 -> finished
18:00:26:834 APS-DATA.request id: 161 erase from queue
18:00:27:189 APS-DATA.confirm id: 159, status: 0x00 SUCCESS
18:00:27:213 aps request id: 159 finished, erase from queue
18:00:27:332 APS-DATA.confirm id: 158, status: 0x00 SUCCESS
18:00:27:372 aps request id: 158 finished, erase from queue
18:00:27:709 enqueue event config/localtime for /config/
18:00:27:710 poll node 84:2e:14:ff:fe:38:7b:3e-01
18:00:27:711 Poll light node RubanLedLidl
18:00:27:712 Idle timer triggered
18:00:27:712 Force read attributes for ZHALightLevel SensorNode LightLevel 18
18:00:27:713 don't create binding for attribute reporting of sensor LightLevel 18
18:00:27:714 Force binding of attribute reporting for node LightLevel 18
18:00:28:582 poll node 60:a4:23:ff:fe:8c:ce:4c-01
18:00:28:583 Poll light node SalonAppliqueGauche
18:00:28:657 Skip idle timer callback, too early: elapsed 948 msec
18:00:29:475 poll node 84:2e:14:ff:fe:13:da:bb-01
18:00:29:477 Poll light node CouloirLed1
18:00:29:657 enqueue event config/localtime for /config/
18:00:29:693 APS-DATA.request id: 180, addrmode: 0x03, addr: 0x00212effff059d21, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
18:00:29:706 APS-DATA.confirm id: 180, status: 0x00 SUCCESS
18:00:29:707 APS-DATA.confirm request id: 180 -> confirmed, timeout 1615568429
18:00:29:712 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 217, rssi: 24
18:00:29:714 	asdu: 1300070201219d05ffff2e21004cce8cfeff23a4604fb8250101fc
18:00:29:716 APS-DATA.indication request id: 180 -> finished
18:00:29:716 APS-DATA.request id: 180 erase from queue
18:00:30:313 poll node 58:8e:81:ff:fe:fd:12:fa-01
18:00:30:314 Poll light node CouloirLed2
18:00:30:656 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
18:00:30:658 enqueue event config/localtime for /config/
18:00:30:659 poll node 58:8e:81:ff:fe:c6:ba:10-01
18:00:30:659 Poll light node CouloirLed3
18:00:30:660 Idle timer triggered
18:00:30:660 binding for attribute reporting of ep: 0x01 cluster 0x0001 seems to be active
18:00:30:661 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
18:00:30:661 Force binding of attribute reporting for node SYMFONISK Sound Controller 
18:00:31:482 poll node 5c:02:72:ff:fe:2d:e8:5c-01
18:00:31:484 Poll light node SalonAppliqueDroite
18:00:31:657 enqueue event config/localtime for /config/
18:00:32:324 poll node 84:2e:14:ff:fe:38:7b:3e-01
18:00:32:325 Poll light node RubanLedLidl
18:00:32:573 APS-DATA.request id: 196, addrmode: 0x03, addr: 0x00212effff059d21, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
18:00:32:585 APS-DATA.confirm id: 196, status: 0x00 SUCCESS
18:00:32:586 APS-DATA.confirm request id: 196 -> confirmed, timeout 1615568432
18:00:32:592 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 221, rssi: 30
18:00:32:593 	asdu: 1700070301219d05ffff2e210010bac6feff818e580dd2250101fc
[... 2293 lines cut ...]
18:02:40:660 enqueue event config/localtime for /config/
18:02:41:157 poll node 60:a4:23:ff:fe:8c:ce:4c-01
18:02:41:159 Poll light node SalonAppliqueGauche
18:02:41:657 enqueue event config/localtime for /config/
18:02:42:047 poll node 84:2e:14:ff:fe:13:da:bb-01
18:02:42:049 Poll light node CouloirLed1
```
</p>
</details>

<details>
<summary>Type of sequence seen when pressing a button</summary>
<p>

```
18:01:00:263 [INFO] - No button map for: LXEK-5, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 210020C8, zclSeq: 1
18:01:00:264 ZCL attribute report 0x000D6FFFFEC8A078 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:01:00:264 	payload: 210020c8
18:01:00:265 APS-DATA.request id: 121, addrmode: 0x02, addr: 0x8f8f, profile: 0x0104, cluster: 0x0001, ep: 0x01 -> 0x01 queue: 1 len: 5 tx.options 0x00
18:01:00:269 Websocket 10.0.10.111:51712 send message: {"config":{"battery":100,"on":true,"reachable":true},"e":"changed","id":"15","r":"sensors","t":"event","uniqueid":"00:0d:6f:ff:fe:c8:a0:78-01-1000"} (ret = 148)
```
</p>
</details>

@Smanar don't want to bother you, but it seems you're french speaking too, so if you want I will pm you on discord if needed more infos or debug

Smanar, 2021-03-15:
For information @bemardev have finished the support for it, if you want to try it > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4587

vader781, 2021-03-15:
Hi, thanks for the work accomplished.
I'm not sure how to confirm it works. 
I went to my Jeedom and hit Install Dependences on my deconz Plugin. First try including it didn't work. 
Should it be enough to be able to include it to Jeedom and have the commands ?

Smanar, 2021-03-16:
You need to compile the code, and replace the file to test the new code.

closed 2021-04-15 by github-actions[bot]

diliact, 2021-04-15:
Hello, is the issue fixed?

I cannot provide technical support as my raspberry pi is very slow and currently used for another purpose and my Conbee is also used by Jeedom. I can only send one remote to Smanar for example.

Smanar, 2021-04-15:
No need, this device can work, but the PR is not yet validated https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4587

bemardev, 2021-04-15:
@diliact Issue is fixed, just waiting for the PR mentioned by @Smanar to be included...
@Smanar do you know what is needed to finish validating that PR? Would be nice to have it in next stable...

Smanar, 2021-04-15:
Good question, 1 month old PR ...
Have updated files for merging, but there is no comment, so idk if there is a problem, I hope for PR will be validated on next session.

Smanar, 2021-04-19:
@bemardev I m asking you something on discord ^^

Smanar, 2021-04-23:
No news ?

Smanar, 2021-04-25:
@bemardev ?

mathoudebine, 2021-05-19:
Thanks for the work on this device, just bought it today and happy to see it is supported by Zigbee2Mqtt and now deCONZ ! Hope the PR will be merged soon.

The device already has [a dedicated page in the Zigbee Device Compatibility Repository](https://zigbee.blakadder.com/ADEO_HR-C99C-Z-C045.html), it would be great to edit it after the PR is merged to indicate deCONZ support.

FYI i opened a [Device support request on Zigpy](https://github.com/zigpy/zha-device-handlers/issues/889) for this device to be fully supported in Zigbee Home Automation as well.

Smanar, 2021-05-19:
For the moment I m  waiting for a last test before, but if someone else can test the device ?

[2 more replies]