dresden-elektronik/deconz-rest-plugin #3657: Dimmer Switch HGZB-04D

shsurury, 2020-11-15:
## Device

- Product name: Dimmer Switch HGZB-04D
- Manufacturer: Feibit Inc Co.
- Model identifier: FB56+ZSC05HG1.0
- Device type : Please remove all unrelated device types. 
  - Switch
The dimmer switches crasshes when requesting changing the brightness. Also the on/off state is not available when the device is physically turned on/off

 

## Screenshots

[image]
[image]

Mimiix, 2020-11-15:
You are missing the other cluster screens.

shsurury, 2020-11-15:
Hopefully I have included all. Please let me know if anything missing

[image]
[image]
[image]
[image]
[image]

shsurury, 2020-12-01:
Hi is there any update on this request. It will be a great help if this added

closed 2020-12-16 by SwoopX