dresden-elektronik/deconz-rest-plugin #3658: Feibit 1 Gang Switch HGZB-41

shsurury, 2020-11-15:
## Device

- Product name: 1 Gang Switch 
- Manufacturer: Feibit Inc Co.
- Model identifier: FB56+ZSW1GKJ2.5
- Device type : Please remove all unrelated device types. 
  - Switch
The on/off state is not available when the device is physically turned on/off. The device is not refreshing automatically

 

## Screenshots

[image]
[image]

Mimiix, 2020-11-15:
You are missing the other cluster screens.

shsurury, 2020-11-15:
Hopefully I have attached all the clustures. Please let me know if any missing
[image]
[image]
[image]

shsurury, 2020-12-01:
Hi is there any update on this request. It will be a great help if this added

SwoopX, 2020-12-16:
@shsurury already on the latest beta and had a try? Should work from thereon for your devices.

shsurury, 2020-12-16:
Thanks SwoopX,

Where do I get the beta firmware so I can install on my raspbeeII

SwoopX, 2020-12-16:
Depends on how you run deconz (it's not the firmware you need to update). Have you previously installed it via apt and if yes, which branch did you chose? Or are you running docker?

shsurury, 2020-12-16:
Hi SwoopX,

I was running Deconz as an add on for home assistant. Currently I am using zigbee2mqtt on home assistant as I was having too many issues with the Phoscon app maintaining my devices as I have almost 75 devices.

Please feel free to contact me for any further queries.

Regards

Sajid Husain Surury
Food Safety Auditor & Consultant
B. Sc., M. Sc., M. App. Sc., MAIFST
NFSA High Risk Auditor # 113236
DHHS Auditor # 11017153
Mobile – 0400571140<tel:0400571140>
Phone/Fax – 03 8839 0538<tel:03%208839%200538>
E mail – sajid@surury.com.au<mailto:sajid@surury.com.au>

closed 2020-12-16 by SwoopX