dresden-elektronik/deconz-rest-plugin #3660: Feibit 3 Gang Switch HGZB-043

shsurury, 2020-11-15:
## Device

- Product name: 3 Gang Switch HGZB-043
- Manufacturer: Feibit Inc Co.
- Model identifier: FB56+ZSW1IKJ2.5
- Device type : Please remove all unrelated device types. 
  - Switch
The on/off state is not available when the device is physically turned on/off. The device is not refreshing automatically

 

## Screenshots

[image]
[image]

Mimiix, 2020-11-15:
You are missing the other cluster screens.

shsurury, 2020-11-15:
Please find attached the rest of the clusters. Hopefully they are all there. Please let me know if any missing

[image]
[image]
[image]

shsurury, 2020-12-01:
Hi is there any update on this request. It will be a great help if this added

SwoopX, 2020-12-12:
What exactly should be done here, is it working?

shsurury, 2020-12-12:
Hi SwoopX,
It is working but the gateway is not able to request the state of the device so when you physically turn the switch on the gateway still shows that it is off

Regards

Sajid Husain Surury
Food Safety Auditor & Consultant
B. Sc., M. Sc., M. App. Sc., MAIFST
RABQSA # 113236
Mobile – 0400571140<tel:0400571140>
Phone/Fax – 03 8839 0538<tel:03%208839%200538>
E mail – sajid@surury.com.au<mailto:sajid@surury.com.au>

closed 2020-12-14 by manup