dresden-elektronik/deconz-rest-plugin #3661: Feibit Downlight HGZB-07A

shsurury, 2020-11-15:
## Device

- Product name: Downlight HGZB-07A
- Manufacturer: Feibit Inc Co.
- Model identifier: FB56-ZCW11HG1.2
- Device type : Please remove all unrelated device types. 
  - Light
 

## Screenshots

[image]
[image]

shsurury, 2020-11-15:
additional clusters

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

shsurury, 2020-11-15:
[image]

shsurury, 2020-11-15:
[image]

shsurury, 2020-12-01:
Hi is there any update on this request. It will be a great help if this added

SwoopX, 2020-12-12:
What exactly should be done here, is it working?

closed 2020-12-14 by manup