dresden-elektronik/deconz-rest-plugin #3663: Feibit RGBW Bulb HGZB-07A

shsurury, 2020-11-15:
## Device

- Product name: RGBW Bulb HGZB-07A
- Manufacturer: Feibit Inc Co.
- Model identifier: FB56-ZCW11HG1.2
- Device type : Please remove all unrelated device types. 
  - Light
 

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

shsurury, 2020-12-01:
Hi is there any update on this request. It will be a great help if this added

SwoopX, 2020-12-12:
What exactly should be done here, is it working?

shsurury, 2020-12-12:
Hi SwoopX,
It is working but the gateway is not able to request the state of the device so when you physically turn the switch on the gateway still shows that it is off

Regards

Sajid Husain Surury
Food Safety Auditor & Consultant
B. Sc., M. Sc., M. App. Sc., MAIFST
RABQSA # 113236
Mobile – 0400571140<tel:0400571140>
Phone/Fax – 03 8839 0538<tel:03%208839%200538>
E mail – sajid@surury.com.au<mailto:sajid@surury.com.au>

closed 2020-12-14 by manup