dresden-elektronik/deconz-rest-plugin #3664: The file /deCONZ.tar.gz couldn't be found.

Pirlobeck, 2020-11-15:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 264A0700
 - deCONZ version: 2.05.67
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Question: before I update the system, I want to make a Backup with the Phoscon Backup feature, but I got the Error
"This is not the page you are looking for
The file /deCONZ.tar.gz couldn't be found."
When I press the Backup button, I get " Backup successful ", but after pressing the Download button the Error appears.

Smanar, 2020-11-15:
Hello, I don't know exactly how work this feature but
- Have you take a look if you have the file on your system ?
`sudo find / -name "deCONZ.tar.gz"`
On my side I have it at
`/home/pi/.local/share/dresden-elektronik/deCONZ/deCONZ.tar.gz`

You can too make the backup file "home made" it s just a zip file that contain the files
- deCONZ.conf
- zll.db
- session.defaut

Pirlobeck, 2020-11-15:
> Hello, I don't know exactly how work this feature but have you take a look if you have the file on your system ?
> `sudo find / -name "deCONZ.tar.gz"`
> On my side I have it at
> `/home/pi/.local/share/dresden-elektronik/deCONZ/deCONZ.tar.gz`
> 
> You can too make the backup file "home made" it s just a compressed file that contain the files
> 
> * deCONZ.conf
> * zll.db
> * session.defaut

OK, the file is not on my system!
In the folder /home/pi/.local/share/dresden-elektronik/deCONZ/ are 
config.ini
gw-version
session.default
zcldb.txt
zll.db

Which files do I need vor the zip File?

Smanar, 2020-11-15:
Unfortunately the missing file "deCONZ.conf" is created by the code, so if you haven't it ....

But from my memory this file contain the gateway config, so it will not change if you just update deconz (without firmware update)
The more important file is zll.db, this one contain all your devices, and your setting. I think just backup this file is enought for a deconz upgrade.

But I don't see why your backup don't work, a permissions problem ?

Pirlobeck, 2020-11-15:
How can I change the permission?
I found an  "deCONZ.conf" in an older backup, but I am not able to copy it in the folder deCONZ...
Write Permission for the folder is given!

Smanar, 2020-11-15:
>I am not able to copy it in the folder deCONZ

So it s perhaps a way to explore.

With 
`sudo chmod -R 777 folder_name`

But take care this command give full right, so take the lower folder.

Pirlobeck, 2020-11-15:
[image]
Write and read permission is given!
I am able to download a file, but not to upload!
I can rename files, but I not able to change content.

Smanar, 2020-11-15:
I don't understand the pricture, but remember you haven't same right on the GUI than on console with sudo.
On my side I can do in console some thing impossible to do in the GUI.

closed 2020-12-25 by stale[bot]

snhnic, 2022-10-19:
The file /deCONZ.tar.gz couldn't be found

I'm on synology docker. I don't have /home/pi/.local/share/dresden-elektronik/deCONZ/ and my zll.db smmes old. Everytime I update deconz or synology conbee has lost new devices and resets it self to older stuff. Whart can i do?

Smanar, 2022-10-19:
You have error message when trying to make a backup using phoscon ?

snhnic, 2022-10-19:
No error, succes on creating backup. But url download not working htps://domain/deCONZ.tar

Smanar, 2022-10-19:
I mean error in deconz log (on deconz / help / debug view with "error" "error_l2" "info" and "info_l2"
What is your deconz version ?

Mimiix, 2022-10-20:
@snhnic Please open a topic on the forums .