dresden-elektronik/deconz-rest-plugin #3667: Develco Smart Cable

r0dh3773, 2020-11-15:
## Device

- Product name: Smart Cable
- Manufacturer: Develco
- Model identifier: SMRZB-143
- Device type : Please remove all unrelated device types. 
  - Other: Smart plug/relay w/ Power consumption

https://www.develcoproducts.com/products/smart-relays/smart-cable/
I only get the on/off functions in home-assistant / node-red (Shows up as a light in phoscon), 
it would be nice to have the power consumption readings from it.

## Screenshots

[image]
[image]
[image]
[image]
[image]

Smanar, 2020-11-16:
Hello, do you have an unix machine to test the code ?
Else the code is small https://github.com/Smanar/deconz-rest-plugin/commit/4ad6e3c56563665df147f446c64e2a463d666298 probably don't need testing.

SwoopX, 2020-11-22:
@Smanar You need to update calculation for the current, as the value displayed is in mA. Just add it to the already given 13 devices.

You also need to add it for the reporting configuration of voltage and current where reportable change is 100. After that, lemme have a quick look and it should be fine then.

Smanar, 2020-11-23:
done

SwoopX, 2020-11-23:
Yup. Then I'd say it's good to go.

Smanar, 2020-11-24:
Ok, because I don't realy like send code without test.

SwoopX, 2020-12-17:
@Smanar Can you please raise the PR? Please change the model identifier according to the other develco devices (that has been updated recently) prior to that.

Otherwise, the code is fine, no worries.

Smanar, 2020-12-17:
Oups sorry, I totaly forget this PR

But what is the problem with the model identifier ?

SwoopX, 2021-02-15:
Added in 2.10.00-beta

closed 2021-02-15 by SwoopX