dresden-elektronik/deconz-rest-plugin #3670: Aqara Opple 6 button switch in deconz 2.05.88

ruabo, 2020-11-15:
## Describe the question or issue you are having
Hi, I'm not able to receive events in HA after successful pairing in Phoscon. In developer tools deconz_event are not detected

## Screenshots
[image]

## Environment
Deconz 2.05.88 

 - Host system: Raspberry Pi 3
 - Running method: Home Assistant 0.117.6 deCONZ Add-on 6.5.0 HassOS 4.16 Supervisor 2020.11.0
 - Firmware version: 26390500
 - deCONZ version: (2.05.88)
 - Device: RaspBee I
 - Is there any other USB or serial devices connected to the host system? Yes, Cyberpower UPS

## deCONZ Logs

## Additional context
While pressing opple button, other zigbee lights get activated. I tried to remove opple, stop deconz, remove from HA the device by calling service deconz.remove_orphaned_entries, reboot host but nothing. No entity are created for opple.

Mimiix, 2020-11-16:
Probably pairing issues. I had this too on mine, after pairing it a few times it worked fine.

arnvre, 2020-11-16:
Same problem here with Aqara Opple Switch 4 button. Similar issue as described in 2.05.85 (two buttons turn on or off all the lights and smart plugs connected) and is found in the Phoscon App. I'm not able to change the behavior of the buttons in Phoscon App nor in HAS. I tried resets, reconnects, reboot etc. Problem occurred after update deCONZ and firmware Conbee II.

Host system: Ubuntu 20.04
Running method: Home Assistant Container 0.117.6, deCONZ container
deCONZ version: 2.05.88
Device: Conbee II
Firmware: 26660700

Maxtropos, 2020-11-17:
I‘m having the same issue in x.88 and x.85
New paired Aqara-Switches are controlling *all* lights paired to the Raspbee II, following the „standard“ Aqara-Opple Button Layout for direct light control. They do not trigger events anymore.
I re-paired 4 Switches ~10 Times to verify this behaviour.
I think, that deCONZ sets the switches from event-mode to command-mode and pairs them with every light this way. I don‘t know, how to reset this.

My home automation setup is broken by now, let‘s hope for a hotfix soon.

BTCanada, 2020-11-18:
Same issue here with a WXCJKG13LM 6 button Aqara Opple switch.  The device pairs successfully in Phoscon and it shows up as a device in Home Assistant.  Pressing a button successfully updates the "Last Seen" time in deCONZ, but nothing is received by HA.  The deCONZ addon log indicates that none of the buttons have handlers, and prints this when all six buttons are pressed in sequence:
```
12:08:07:051 [INFO] - No button handler for: lumi.remote.b686opcn01 endpoint: 0x01 cluster: ONOFF (0x0006) command: ON (0x01) payload[0]: 000
12:08:23:497 [INFO] - No button handler for: lumi.remote.b686opcn01 endpoint: 0x01 cluster: ONOFF (0x0006) command: OFF (0x00) payload[0]: 000
12:08:26:343 [INFO] - No button handler for: lumi.remote.b686opcn01 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP (0x02) payload[0]: 001
12:08:27:673 [INFO] - No button handler for: lumi.remote.b686opcn01 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP (0x02) payload[0]: 000
12:08:28:876 [INFO] - No button handler for: lumi.remote.b686opcn01 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: STEP_COLOR_TEMPERATURE (0x4C) payload[0]: 001
12:08:30:167 [INFO] - No button handler for: lumi.remote.b686opcn01 endpoint: 0x01 cluster: COLOR_CONTROL (0x0300) command: STEP_COLOR_TEMPERATURE (0x4C) payload[0]: 003
```

Host system: Windows 10
Running method: HassOS 4.17 in VM (Home Assistant 0.117.6, Supervisor 2020.11.0, deCONZ add-on 6.5.0)
deCONZ version: 2.05.88
Device: Conbee II
Firmware: 26670700

SwoopX, 2020-11-20:
@BTCanada In you case, a file from deconz appears to be missing or cannot be found. Can you please share some log containingdeconz' start? 15-20 secs should be enough

MarcelMertens, 2020-11-20:
Same issue here. After Updating to 2.06.xx from an older version (april/may 2020) most of my aqara switches are broken.
Some events are sent two times, reconnecting is close to impossible or same problems as  Maxtropos

BTCanada, 2020-11-20:
> 
> 
> @BTCanada In you case, a file from deconz appears to be missing or cannot be found. Can you please share some log containingdeconz' start? 15-20 secs should be enough

Here's my deCONZ log during startup.  Please let me know if you want the debugging log as well.  Note that I have 9 Aqara temperature/humidity/pressure sensors and 2 Aqara water sensors connected that seem to work fine, though I seem to be getting `No button handler for: lumi.weather` messages now as well, which I don't recall seeing before.

```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] firmware.sh: executing... 
[10:01:51] INFO: GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II 
[cont-init.d] firmware.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] novnc.sh: executing... 
[cont-init.d] novnc.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[17:01:51] INFO: Running the deCONZ OTA updater...
[17:01:51] INFO: Starting udevd...
[10:01:51] INFO: Websockify waiting for VNC to start
[17:01:51] INFO: Running the IKEA OTA updater...
[17:01:51] INFO: Running the OSRAM LEdvance OTA updater...
[10:01:51] INFO: Waiting for device...
[10:01:55] INFO: Starting VNC server (local/yes)...
[10:01:56] INFO: Starting websockify...
WebSocket server settings:
  - Listen on 127.0.0.1:5901
  - Flash security policy server
  - Web server. Web root: /usr/share/novnc
  - No SSL/TLS support (no cert file)
  - proxying from 127.0.0.1:5901 to 127.0.0.1:5900
  [10:01:58] INFO: deCONZ waiting for VNC to start
[10:01:58] INFO: Starting the deCONZ gateway...
QCoreApplication::arguments: Please instantiate the QApplication object first
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
10:01:59:287 HTTP Server listen on address 0.0.0.0, port: 40850, root: /usr/share/deCONZ/webapp/
10:01:59:289 CTRL. 3.27.210:01:59:296 COM: use stable device path /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2217191-if00
[17:01:59] INFO: Starting Nginx...
2020/11/20 17:01:59 [notice] 321#321: using the "epoll" event method
2020/11/20 17:01:59 [notice] 321#321: nginx/1.14.2
2020/11/20 17:01:59 [notice] 321#321: OS: Linux 5.4.77
2020/11/20 17:01:59 [notice] 321#321: getrlimit(RLIMIT_NOFILE): 1048576:1048576
2020/11/20 17:01:59 [notice] 321#321: start worker processes
2020/11/20 17:01:59 [notice] 321#321: start worker process 1475
10:01:59:305 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2217191-if00 / serialno: DE2217191, ConBee II
10:01:59:305 ZCLDB init file /data/.local/share/dresden-elektronik/deCONZ/zcldb.txt
10:01:59:325 parent process s6-supervise
10:01:59:326 gw run mode: docker/hassio
10:01:59:326 GW sd-card image version file does not exist: /data/.local/share/dresden-elektronik/deCONZ/gw-version
10:01:59:326 DB sqlite version 3.27.2
10:01:59:326 DB PRAGMA page_count: 39
10:01:59:326 DB PRAGMA page_size: 4096
10:01:59:327 DB PRAGMA freelist_count: 0
10:01:59:327 DB file size 159744 bytes, free pages 0
10:01:59:327 DB PRAGMA user_version: 7
10:01:59:327 DB cleanup
10:01:59:327 DB create temporary views
10:01:59:330 started websocket server at port 8081
10:01:59:330 [INFO] - Found file containing button maps. Parsing data...
10:01:59:331 [ERROR] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
10:01:59:332 [INFO] - Button maps loaded.
[... 60 lines cut ...]
10:02:13:317 [INFO] - No button handler for: lumi.weather endpoint: 0x01 cluster: 0x0405 command: 0x0A payload[0]: 000
10:02:13:318 ZCL attribute report 0x00158D00057912C9 for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:02:13:323 [INFO] - No button handler for: lumi.weather endpoint: 0x01 cluster: 0x0403 command: 0x0A payload[0]: 000
10:02:13:323 ZCL attribute report 0x00158D00057912C9 for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:02:20:403 [INFO] - No button handler for: lumi.weather endpoint: 0x01 cluster: 0x0402 command: 0x0A payload[0]: 000
10:02:20:404 ZCL attribute report 0x00158D000548A4AB for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:02:20:415 [INFO] - No button handler for: lumi.weather endpoint: 0x01 cluster: 0x0405 command: 0x0A payload[0]: 000
10:02:20:416 ZCL attribute report 0x00158D000548A4AB for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:02:20:420 [INFO] - No button handler for: lumi.weather endpoint: 0x01 cluster: 0x0403 command: 0x0A payload[0]: 000
10:02:20:421 ZCL attribute report 0x00158D000548A4AB for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:02:26:407 [INFO] - No button handler for: lumi.weather endpoint: 0x01 cluster: 0x0402 command: 0x0A payload[0]: 000
10:02:26:408 ZCL attribute report 0x00158D00054C6834 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:02:26:416 [INFO] - No button handler for: lumi.weather endpoint: 0x01 cluster: 0x0405 command: 0x0A payload[0]: 000
[... cut ...]
10:02:45:287 [INFO] - No button handler for: lumi.weather endpoint: 0x01 cluster: 0x0405 command: 0x0A payload[0]: 000
10:02:45:287 ZCL attribute report 0x00158D000578A8DE for cluster: 0x0405, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:02:45:292 [INFO] - No button handler for: lumi.weather endpoint: 0x01 cluster: 0x0403 command: 0x0A payload[0]: 000
10:02:45:292 ZCL attribute report 0x00158D000578A8DE for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
10:02:59:295 Current channel 15
10:02:59:307 Device TTL 1481 s flags: 0x7
```

Maxtropos, 2020-11-21:
Could we change the issue label to bug report then?

ruabo, 2020-11-21:
@ Mimiix 
> Probably pairing issues. I had this too on mine, after pairing it a few times it worked fine.

no matter how many times I try, It ends the same way

BTCanada, 2020-11-21:
I downgraded to 2.05.84 (HA add-on 6.4.1) and the switch now works properly.  It looks like either something got lost when I upgraded to 2.05.88, or some bug was introduced.

VoyteckPL, 2020-11-21:
How can I downgrade in HA? Any guides online? Thanks

ruabo, 2020-11-21:
I need those instructions too ! Thanks

MarcelMertens, 2020-11-21:
Also downgraded:

ConBee II Firmware 26660700 -> 26650700
Deconz 2.06.00 -> 2.05.84

Repairing now works, switch works.

Deconz downgrade for Ubunutu

```
wget http://deconz.dresden-elektronik.de/ubuntu/stable/deconz-2.05.84-qt5.deb
sudo systemctl stop deconz
sudo apt install /path/to/deconz-2.05.84-qt5.deb
sudo systemctl start deconz
```

VoyteckPL, 2020-11-21:
Is that even possible on Hassos?

BTCanada, 2020-11-21:
It is possible, though unfortunately the only way I know of doing it is by restoring an old HA snapshot, so you have to have a snapshot that includes that version.  Fortunately, you can just restore deCONZ from the snapshot rather than restoring the whole thing: 
[image]

VoyteckPL, 2020-11-21:
Nice. Unfortunately I don't have any backups... Any chance you could upload it do I could restore deconz only from your snapshot?

VoyteckPL, 2020-11-21:
Or perhaps make a backup with deconz only for your privacy? I would be trully grateful!

BTCanada, 2020-11-21:
There are quite a few files in the deCONZ snapshot.  Any idea what's core vs configuration?  I don't particularly want to stick my deCONZ database online.

VoyteckPL, 2020-11-21:
Understand. Maybe you could mail it? Wojciech.racek@gmail.com

BTCanada, 2020-11-21:
Not without anonymizing it, if we can figure out how.  Better yet would be for whatever is causing this to get fixed in a deCONZ update, since it looks like numerous people are affected.

VoyteckPL, 2020-11-21:
Sure

VoyteckPL, 2020-11-21:
How often are Deconz update released? Is there any chance for a quick hotfix from this company?

SwoopX, 2020-11-21:
Tough week. Releases are usually done on a monthly basis (check the repo readme). However, we can ofc shoot out hotfixes way faster if required. And for that, we first need to pinpoint the issue here.

I've seen things in @BTCanada's log data which doesn't really fit together. This comment here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3670#issuecomment-729898245 indicates the maps for the buttons are not loaded, so the zigbee commands cannot be translated. Here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3670#issuecomment-731290553 , deconz startup says it loaded the maps alright however (lines contain [INFO] after the timestamp).

The only reason for that I can think of is an older version of `button_maps.json`, which does not get updated due to whatever reason during the update. Can anybody having the issue with 2.06.00 search for the file withing the container and share it here to double check?

BTCanada, 2020-11-22:
@SwoopX Ok, dumb question: how do I get `button_maps.json` out of the container so I can post it here?  I found it and tried copying it directly from the `addon_core_deconz` container to the `homeassistant` container so I could access it though my HA samba share, but apparently docker doesn't support copying between containers.  Trying to copy it first to a directory on the HassOS host tells me my file system is read-only.  I tried `apt-get install sftp `and `apt-get install smbclient` in the deconz container and they both claimed the package was unable to be located.

BTCanada, 2020-11-23:
Some progress here: I managed to grab `button_maps.json` from the container (for anyone else trying this, you can't write to `/root` when logged in as `root` on HassOS, apparently, but you can write to `/tmp`).  After grabbing the 2.5.84 version of the file I re-updated to 2.5.88 and grabbed that one as well for comparison.  However, the switch now works in HA with no modifications, and is no longer throwing `No button handler` events.  It looks like something was messed up with my install, and downgrading to a previous snapshot and then re-upgrading to the latest fixed it.

VoyteckPL, 2020-11-23:
Really weird...can we expect a hotfix soon?

Mimiix, 2020-11-23:
@VoyteckPL I can arrange one, but i would need a good reason for it to push one ahead. I have one on the 30th this month, so i'd like to stick with that. But we need to have a PR done with a fix before that.

VoyteckPL, 2020-11-23:
Good reason is that these buttons do not work :) I had to go with Z2M instead of Deconz because this switch is crucial for my house. It's in the kitchen and serves of the lights and my wife is not happy ;) Wife is a good reason :D Please, fix it, I also have Lonsonho 4 gang zigbee switch and it also doesn't work as expected....

Mimiix, 2020-11-24:
@VoyteckPL I have one that is working perfectly fine. If there's a fix we can include at the 30th. Having a release just for 1 switch isn't worth the work imho. There's a lot of work for a release in terms of building the application.

VoyteckPL, 2020-11-24:
Ok, I can stick to Z2M for a week ;) Deconz has faster response to buttons so that's why I want to stick to it.

BTCanada, 2020-11-24:
@VoyteckPL, could you help the deCONZ team determine what is going wrong?  They can't fix what they don't know about.  Start by describing the issue you're having in as much detail as you can, and provide logs where possible.  In my case, downgrading and re-upgrading fixed the issue, so it looks like @SwoopX's suspicion that `button_maps.json` didn't get updated properly was accurate.  If you don't have any backups, you could try uninstalling deCONZ and reinstalling it (just make sure to take a deCONZ configuration backup first so you don't lose your setup; might as well make a HA snapshot as well for extra safety).

VoyteckPL, 2020-11-25:
My issue is: deconz_events in HA are not visible. Switch only works as direct control of all the lights. I did reinstall. It didn't help.

SwoopX, 2020-11-25:
@BTCanada thanks for your feedback and glad to see you got it resolved for you. Also was pleased to read some safe and sound words from you.

@VoyteckPL Sorry to say that, but you've given us almost nothing to work with here besides that it doesn't work and you moved to z2m. We can certainly prepare a fix, but for what exactly?

If it is only related to button events, then there's two reasons:

1. When a button press controlls all lights, then the switch is bound to zigbee group 0xfff0 (the all group)
2. The button maps are either not loaded our outdated

As for 1), this can happen for switches and is not an Aqara exclusive phenomenon. I had my very own bad experience with that shutting my whole place literaly completely down, no funat all. To overcome that, it might be required to reset and re-pair the switch a couple of time till working.

2) When deconz starts, it reports if it found the file containing the button mapping. If yes, good, if not, that file is missing or defective. No dice.
When the file was found, but still info messages are emitted stating no fitting button map has been found, then the file most likely does not contain any modelIDs and consequently is outdated.

So, apparently, I was right in @BTCanada's case putting my bet on the file.

VoyteckPL, 2020-11-26:
How can I factory reset this device? I will try repairing. Maybe I should take battery out?

mprajsler, 2020-11-26:
@VoyteckPL Hi I did that several times. Click “C” button five times to reset. But it did not resolve the problem. Tried battery removal too (not that simple). How about this?

09:03:24:157 [INFO] - Found file containing button maps. Parsing data...
09:03:24:159 [ERROR] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
09:03:24:165 [INFO] - Button maps loaded.

I think this is the problem...

BR,
M.

mikecli, 2020-11-26:
Can confirm this issue with all my Aqara switches after I upgrade to the latest deconz add-on in Hassio. It incl. Aqara Opple 1 gang and 2 gang switches, the Aqara 1 button Smart Switch and Aqara movement sensor --they do not appear to be firing events in "deconz_events" anymore despite being paired. The Aqara lights do seem to be working. A downgrade of the deconz add-on using a previous snapshot restored functionality.

BTCanada, 2020-11-26:
@mikecli, if you re-upgrade does it stay fixed?  It did for me.
@mprajsler, that looks like it's successfully finding the button map file (the error listed is specific to a certain device, which isn't the one in question).  Posting your button_maps.json file here would help.  Also, what version of deCONZ are you running?

I'd love to find out why this happened to a bunch of us and get it fixed so next time I update deCONZ my daughter doesn't start complaining that the switch for her LEDs no longer works.

Jan1503, 2020-11-26:
Just wanted to say I'm facing the same issues here:
23:45:27:192 [INFO] - No button handler for: lumi.remote.b286opcn01 endpoint: 0x01 cluster: ONOFF (0x0006) command: ON (0x01) payload[0]: 000

How to connect to the docker container containing the deconz-addon to be able to verify or replace the json-file? I could only ssh to the ha-container using the installed ssh-server..

The switch is supported after confirming with [https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/button_events/#xiaomi-aqara-opple-switches]

The switch is still binding to zigbee group 0xfff0 no matter how often i reset and re-pair it...
I don't have a backup of an older config cause this is my first installation :)

Thanks!

SwoopX, 2020-11-26:
You might try this procedure as a workaround https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061#issuecomment-567380475

Jan1503, 2020-11-27:
Ok, after trying this method [comment](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061#issuecomment-731493950) the error in deconz is gone:
01:02:57:242 [INFO] - Button 2002 On press lumi.remote.b286opcn01

But still no events coming thru to Home Assistant :( Tried a restart but that didn't help.

Jan1503, 2020-11-27:
Removed it completely using phoscon and deconz and readded using the normal method results in the same problem, no button handler found...very weird...

BTCanada, 2020-11-27:
> How to connect to the docker container containing the deconz-addon to be able to verify or replace the json-file?
@Jan1503 To connect to the docker container for deconz, you'll have to login to the machine (or VM) running HA.  I'm running HassOS on a VM, so if you're using some other method of running HA, YMMV.  The default HassOS login is `root` with no password.  Once you're logged in, you'll be greeted by some ASCII art and a command prompt:
[image]
At this point, issue the following commands:
1. `login` to get out of the ha-cli and get a shell on the host system.
2.  `cd /tmp` to change to a directory where you can write files
3. `docker cp addon_core_deconz:/usr/share/deCONZ/devices/button_maps.json .` to grab the button map from the deCONZ container and copy it to the current directory.
4. `docker cp button_maps.json homeassistant:/config` to copy the button map file to the HA container so you can access it through HA functionality.
5. At this point you can get the file off of the HA machine and do what you want with it (like upload it somewhere and link it here).  I think you could use SCP for this, or use the [`Samba share` HA add-on](https://github.com/home-assistant/addons/tree/master/samba) to create a network file share to HA.

If for some reason the button map isn't in the `/usr/share/deCONZ/devices/` folder in the deCONZ container, you can connect to the container and try to see where it's located:
1. `docker exec -it addon_core_deconz /bin/bash` to open a shell on the deCONZ container
2. `find / -name button_maps.json` to look for the file

VoyteckPL, 2020-11-27:
_login to get out of the ha-cli and get a shell on the host system._
What credentials should be used at this step?

Jan1503, 2020-11-27:
@BTCanada Thanks for that, but I don't think this is the issue anymore since I got the switch working in phoscon/deconz using the method accelle17 mentioned in the other issue.
Still not getting the events in HA. All other switches/sensors/bulbs working fine.

VoyteckPL, 2020-11-27:
@Jan1503 How do you control devices in HA then?

Jan1503, 2020-11-27:
> @Jan1503 How do you control devices in HA then?

Not at all :) I cannot use the Opple switch yet. Not a dealbreaker, only have one switch for testing. I use Philips and Sunricher switches around the house.

VoyteckPL, 2020-11-27:
Thanks man. So the issue still exists :(

Jan1503, 2020-11-27:
> Thanks man. So the issue still exists :(

Yes. If you only need to control some lights with it, maybe its sufficient to configure it in phoscon only until the HA-events work again?!

Mimiix, 2020-11-27:
It's odd. I'm on 2.6 and i have no issues with it. Is this limited to docker versions only then?

Jan1503, 2020-11-27:
I'm on latest HassOS (5.5 Dev) and DeConz 6.5.0 (2.05.88). Don't know how to update the add-on to 2.6.

dabku, 2020-11-27:
Had the same problem. What worked for me was mashing the buttons in pairing mode. With OPPLE still in Phoscon (did not delete anything) I clicked on 'Add a new switch', selected 'other'. Hold the pairing button on the back of the switch and I think after it flashed I began to press the buttons, might take few tries to figure out when to press them. It's possible that the mentioned linked [workaround](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061#issuecomment-731493950) is working and I just did not press the button in the correct moment. Now I receive events in HA. Similar thing happened to MiCube, had to rotate it constantly to get both flip and rotate events.

mprajsler, 2020-11-27:
@BTCanada Hello thank you for a guide how to extract "button_maps.json" here is mine 
[button_maps.json.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5606944/button_maps.json.txt)

I am running Hassio 0.118.0 on rpi4 and the deconz add-on version 6.5.0 which is deCONZ 2.05.88

Jan1503, 2020-11-27:
@mprajsler The file looks valid to me, the Opple-switches are correctly mapped. This must be another issue and the log-error about the missing events is a side effect of it.

mprajsler, 2020-11-27:
@Jan1503 pairing issues? But how to correctly pair it? I tried so many times and all workarounds mentioned here... :(

Jan1503, 2020-11-27:
@mprajsler The pairing seems ok and I got the battery-level in HA too but no events using the listener in HA.
[image]

DeConz-Log:
[image]

Jan1503, 2020-11-27:
Ok, after re-pairing the damn switch for about 15 times, event are coming thru:
[image]
Will never touch this thing again ;)

SwoopX, 2020-11-27:
Will share my theory on all this later on.

Jan1503, 2020-11-27:
Ok, can't get my hands off of this...deleted it again and got it to work now:
So basicly what you need to do is paring the switch with the method mentioned above. It seems vital to wait until deconz has fully initialized the switch before you close the popup in phoscon. Then you should get the battery-sensor in HA but no events...
Then restart HA and the events will be there...

VoyteckPL, 2020-11-27:
I'm trying this method - no luck. When I hold 6 seconds, led blink 3 times. The after few seconds it blinks again 3 times. I press random buttons on the switch. It pairs but turns on all the lights. When I'm in deconz app I select opple, click f5, nothing happends. I don't see battery readout in integrations.

VoyteckPL, 2020-11-27:
I get this in log:

13:55:05:053 [INFO] - No button handler for: lumi.remote.b686opcn01 endpoint: 0x01 cluster: LEVEL_CONTROL (0x0008) command: STEP (0x02) payload[0]: 000

[38 more replies]

closed 2020-12-12 by Mimiix