dresden-elektronik/deconz-rest-plugin #3674: Repair devices keeping the original names plus Xiaomi "fixed" router question

ReX1983, 2020-11-15:
## Describe the question or issue you are having

I read that the Xiaomi devices remain linked to the router / coordinator joined at the time of pairing, so any new router added to the network post pairing is ignored. Based on that, it seems that they must be paired from their target location, with all the routers already setup.

Does it apply only to Xiaomi sensors or all the Xiaomi devices? E.g. Aqara no neutral wall switches.

Secondly, assuming the above is correct, I must repair a large number of devices (more than 50). To speed up the process and avoid renaming the repaired devices in both deCONZ and Home Assistant, I was planning to do the following:
- turn off Home Assistant
- create a backup of deCONZ 
- **not** remove the devices from deCONZ
- reset the devices while having deCONZ in discovery mode (repair)
- restore the deCONZ backup
- restart Home Assistant

Would the above procedure allow the Xiaomi devices to use the new routers while keeping the names assigned in deCONZ?

Thanks

## Screenshots

## Environment

 - Host system: (Raspberry Pi 3)
 - Running method: (Marthoc Docker container)
 - Firmware version: (26660700)
 - deCONZ version: (2.05.88)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II

## deCONZ Logs

## Additional context

Smanar, 2020-11-16:
> Does it apply only to Xiaomi sensors or all the Xiaomi devices? E.g. Aqara no neutral wall switches.

For me it s all end device, the no neutral version is not a router from my memory.

For your problem, there is something I have not tested on my site but it seem if you short press the button, the device can choose a new router, it s an easy test to do.

If you don't delete the device, and if you make a new inclusion, the device kee the same name no ?

rvitch, 2020-11-16:
To force end-devide to take the way of new included router (xiaomi or other) , just re-include it after the new router without delede anything.

ReX1983, 2020-11-16:
@Smanar, I can try the short press of reset button of the sensors, but what would I short press with the wall switches no neutral? You are right, they are end devices, but the pairing requires long press of the actual switch rockers. A short press wouldn't do much.

Regarding the repairing of an existing device, @rvitch, last time I tried, deCONZ assigned a new default name. Have you seen otherwise?

Smanar, 2020-11-16:
>@Smanar, I can try the short press of the reset button of the sensors, but what would I short press with the wall switches no neutral? You are right, they are end devices, but the pairing requires long press of the actual switch rockers (there is no reset button). A short press wouldn't do much.

Ha right, haven't ask for this kind of device ....

ReX1983, 2020-11-20:
So, I have tried to repair some devices, here my findings:

1. Repair an existing Xiaomi sensor (e.g. door/temperature/motion) keeps the original name
2. Short press of the reset button of the Xiaomi sensors (e.g. door/temperature/motion) forces them to find new routers. I have seen this in 2 occasions with door sensors, but I am not sure if it works also for motion sensors (in my case they didn't change but it might be because the current connection is already green)
3. Repair a Xiaomi no neutral wall switch generates new devices and **does not keep the original name(s)**. Is this behaviour expected? 
4. In some cases, even if the router (Tradfri repeater) is next to the sensor and ConBee 30 meters away, the Xiaomi devices continue link directly to the ConBee II. 
5. Finally, the Xiaomi no neutral wall switch (both single and double) create a bunch of useless switches and lights under the light sections of Photoscon, the same useless devices are also exported as entities linked to the same device, under Home Assistant. This is a known problem discussed in various occasions but not yet addressed. I think @ebaauw explained that this happen because the devices are added while their features are yet to be discovered. 

Unfortunately the combination of finding 3 and 5 makes the repair process a tedious activity. The fact that it doesn't solve the routing issue (see 3) makes is a waste of time.

Smanar, 2020-11-20:
3. Hu, nope, you have the generic name used by deconz ? to find more easyly the used part in code pls ?

closed 2020-12-25 by stale[bot]