dresden-elektronik/deconz-rest-plugin #3676: Support lost for Schlage Connect Smart Deadbolt B3468

ericvandamme, 2020-11-16:
## Describe the bug
Upgrading to the latest stable release causes Schlage Connect Smart Deadbolt B3468 to no longer be a supported device.

## Steps to reproduce the behavior
  1. Update to v2.5.88 (possibly v2.5.87)
  2. No longer recognized in Phoscon nor name will properly display in deCONZ network map

## Expected behavior
Support expected to continue through versions.

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26390500)
 - deCONZ version: (2.5.88)
 - Device: ConBee I
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context
Attempted to revert to last known version (I am 99% sure it is v2.5.84) but it is no longer available via apt. *This might be a nice feature to have as versions have broke support for devices before, and being able to revert to a last known working config would be delightful*

Smanar, 2020-11-16:
Hello, do you have the cluster list for this device ?

ericvandamme, 2020-11-16:
I have already submitted a Pull Request to add support again. The new stricter check was what made it unsupported.

Attached is the basic cluster information from deCONZ.

[image]

ericvandamme, 2020-11-16:
Suppose I should also include the node info as well.

[image]

closed 2020-12-25 by stale[bot]