dresden-elektronik/deconz-rest-plugin #3679: ICASA rotary dimmer - ICZB R12D

kamiel93, 2020-11-16:
Hi,

Would it be possible to add the ICASA rotary dimmer + (ICZB R12D) as a supported device?
The ICASA dimmer (ICZB-IW11D) is already available.

I tried adding the ICZB R12D to deconz using a RPI + RASPBEE, but unfortunately it's not detected. Also not in Deconz via VNC. 
I also don't see it getting recognized in the logs when I turn it on/off. 

## Device

- Product name: Dimmer+ (ICZB R12D)
- Manufacturer: ICASA
- Model identifier: ICZB R12D
- Device type : Rotary Dimmer

## Screenshots

Mimiix, 2020-11-16:
Without any screenshots we can't integrate anything.

kamiel93, 2020-11-16:
But then i think it can't be added, because it's not recognized via VNC..

Mimiix, 2020-11-16:
You need to search for switches in Phoscon while the switch is in pairing mode.

Then it should appear in deconz. Also: Try to click the buttons while you are pairing.

kamiel93, 2020-11-16:
Ok.., I had issues with the pairing mode of the dimmer. But after like 20 attempts it was recognized in Deconz and Phoscon 👍 !
Sorry for taking your time, but i think it can be closed. Thanks!

closed 2020-11-16 by kamiel93

mauve, 2022-01-02:
Hi @kamiel93 I also have problems pairing this dimmer, do you remember what you did to get it to pair?

(sorry to ping on this old issue)

kamiel93, 2022-01-02:
No problem. After a lot of trying i finally got it in pairing mode as
described in the manual. So thay was my way to add it...