dresden-elektronik/deconz-rest-plugin #3682: New Version of Trust ZYCT Remote/Switch

Puntoboy, 2020-11-16:
## Device

- Product name: Trust ZYCT-202 (updated)
- Manufacturer: Trust International B. V.
- Model identifier: ZLL-NonColorController
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

This looks like a newer version of the same remote listed here. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/261

Mimiix, 2020-11-16:
@Puntoboy Missing screens! You need to screenshot all clusters.

Puntoboy, 2020-11-16:
> 
> 
> @Puntoboy Missing screens! You need to screenshot all clusters.

What does that actually mean? When I submitted a new device request previously I only submitted the above.

Mimiix, 2020-11-16:
The other clusters. Groups, Identity, On/off etc.

Please read https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to-read-Clusters

Puntoboy, 2020-11-16:
[image]

[image]

[image]

[image]

Mimiix, 2020-11-16:
I think one might be missing. Add that one if you can. 
Other than that, your ready i think.

Puntoboy, 2020-11-16:
I tried all of the ones that had attributes sections.

Mimiix, 2020-11-16:
Now we needa wait for a dev to integrate it 👍

Puntoboy, 2020-11-16:
Hopefully it's not too long otherwise I might return it.

Mimiix, 2020-11-16:
I am not going to push anyone just for 1 switch. Be adviced that next beta is expected atleast the 15th of next month. I for a fact know that there's gonna be one at the 30th this month. But  no guarantees on this switch.

wilsonlspacheco, 2020-11-16:
I have one of this device (updated version).
Waiting news, thks everyone.

Puntoboy, 2020-11-17:
> 
> 
> I am not going to push anyone just for 1 switch. Be adviced that next beta is expected atleast the 15th of next month. I know for a fact that there's gonna be one at the 30th this month. But no guarantees on this switch.

Not suggesting you should :) I was hoping that this was a easy fix as there is an existing switch with the same model name already in the system.

Mimiix, 2020-11-18:
The fix itself isnt that hard i think, it's the release schedule in this case :)

I'll ask a dev to check this out for you.

SwoopX, 2020-12-12:
Closing as supported by 2.07.01

Puntoboy, 2020-12-20:
So finally been able to test this and it's not working. I can see the device in VNC but if I reset it and then re-pair I don't see it in the Web UI. Also, Home Assistant doesn't see it either. I can see this in the logs.

```
13:07:14:628 CTRL skip polling while permit join is set
13:07:14:629 unknown neighbor 0x00158D00038E614F (0xE71A), lqi: 143, rssi: -76
13:07:14:629 FP indication 0x0104 / 0x0008 (0x00158D00038E614F / 0xE71A)
13:07:14:630                       ...     (0x00158D00038E614F / 0xE71A)
13:07:14:635 [4] Skipping additional attribute read -  Assumed Trust remote (ZYCT-202)
13:07:14:735 CTRL skip polling while permit join is set
13:07:14:737 CTRL skip polling while permit join is set
13:07:14:737 unknown neighbor 0x00158D00038E614F (0xE71A), lqi: 143, rssi: -76
13:07:14:738 FP indication 0x0104 / 0x0006 (0x00158D00038E614F / 0xE71A)
13:07:14:738                       ...     (0x00158D00038E614F / 0xE71A)
13:07:14:744 [4] Skipping additional attribute read -  Assumed Trust remote (ZYCT-202)
13:07:14:753 unknown neighbor 0x00158D00038E614F (0xE71A), lqi: 143, rssi: -76
13:07:14:754 FP indication 0x0104 / 0x0008 (0x00158D00038E614F / 0xE71A)
13:07:14:754                       ...     (0x00158D00038E614F / 0xE71A)
13:07:14:761 [4] Skipping additional attribute read -  Assumed Trust remote (ZYCT-202)
```

wilsonlspacheco, 2020-12-20:
> So finally been able to test this and it's not working. I can see the device in VNC but if I reset it and then re-pair I don't see it in the Web UI. Also, Home Assistant doesn't see it either. I can [...]

Same here 😔

Mimiix, 2020-12-20:
> So finally been able to test this and it's not working. I can see the device in VNC but if I reset it and then re-pair I don't see it in the Web UI. Also, Home Assistant doesn't see it either. I can [...]

It won't show in the web ui as they need to add support themselves. 

It would show in ha tho. What does the switch look like in vnc?

Puntoboy, 2020-12-20:
Sorry I thought Phoscon and Deconz was part of the same software? In VNC is appears as per the screenshot at the top.

Puntoboy, 2021-01-04:
Any update on this?

Mimiix, 2021-01-04:
@Puntoboy Missed your comment. However: Yes they are, but the Rest API is open source. Phoscon is not. 

https://github.com/dresden-elektronik/phoscon-app-beta/issues

That's the place to ask support for this switch in phoscon 😄

Puntoboy, 2021-01-04:
So how do I fix this? It doesn't appear in the Web UI or Home Assistant.

Mimiix, 2021-01-04:
Can you check the restapi if it shows up?

Also: Did you re-pair after updating?

Puntoboy, 2021-01-04:
Sorry, I don''t really know rest, how do I check please? 

I completely deleted it from VNC and reset the switch.

Mimiix, 2021-01-04:
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

 :)

Puntoboy, 2021-01-04:
I ran GET /api/<apikey> and I can't see it as a device in the list.

Mimiix, 2021-01-04:
Should be under sensors.

Puntoboy, 2021-01-04:
Not sure why it would be under sensors, it's a switch. Anyway I noticed that I must have deleted it again after trying to connect it before, so I re-added it and it now appears in the Rest API.

`      "ep": 1,
      "etag": "6bc8db628882d7668d3ca1f3c56667c1",
      "lastseen": "2021-01-04T12:06Z",
      "manufacturername": "Trust International B.V.",
      "mode": 1,
      "modelid": "ZLL-NonColorController",
      "name": "ZLL-NonColorController 31",
      "state": {
        "buttonevent": 4002,
        "lastupdated": "2021-01-04T12:06:12.361"
      },
      "swversion": "20150212",
      "type": "ZHASwitch",
      "uniqueid": "00:15:8d:00:03:8e:61:4f-01-1000"`

I have it working in Node-Red as that talks to Phoscon directly. I just need to set up the group light controls now as it's currently only able to output one set of buttonevents. 

Thank you :)

Mimiix, 2021-01-04:
Nice! Great to hear that.

Case closed?

Puntoboy, 2021-01-04:
I think so!

Kind regards,
Chris Jeffrey

Get Outlook<https://aka.ms/qtex0l> for iOS

closed 2021-01-05 by SwoopX