dresden-elektronik/deconz-rest-plugin #3683: Random lights can not be controlled

Lindansaren, 2020-11-17:
## Describe the bug

Random lights can not be controlled from time to time. Pressing 0 in VNC on the node gets it o work and respond properly again.
Blinking red light on node in VNC when it happens. Have about 80 nodes in total. About half of them powered by mains and spread out evenly through the house.

## Steps to reproduce the behavior

Happens sporadic several times a week and even sometimes several times a day.

## Expected behavior

Lights should react to commands. =)

## Screenshots

## Environment

 - Host system: Raspberry Pi4
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.05.84
 - Device: ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? If so: Zwave.me stick

## deCONZ Logs

01:02:19:924 delay sending request 227 dt 8 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:023 delay sending request 227 dt 8 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:123 delay sending request 227 dt 8 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:223 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:324 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:423 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:524 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:623 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:724 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:824 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:20:924 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:023 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:124 delay sending request 227 dt 9 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:224 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:324 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:424 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:523 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:624 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:751 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:825 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:21:923 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:023 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:123 delay sending request 227 dt 10 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:267 delay sending request 227 dt 11 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:323 delay sending request 227 dt 11 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:423 delay sending request 227 dt 11 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:523 delay sending request 227 dt 11 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:623 delay sending request 227 dt 11 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:724 delay sending request 227 dt 11 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:823 delay sending request 227 dt 11 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:22:924 delay sending request 227 dt 11 ms to 0x90FD9FFFFE33A6D9, ep: 0x01 cluster: 0x0006 onAir: 2
01:02:23:013 0x90FD9FFFFE33A6D9 error APSDE-DATA.confirm: 0xA7 on task

## Additional context

Mimiix, 2020-11-17:
Can you update to the latest version? There are some fixes in there.

Also, are you able to provide a screenshot of your deCONZ map?

Markkuuss, 2020-11-17:
@Lindansaren: If the lights do not react to your command, try switching the corresponding group. What happens then? Do the lights still not react or do they then switch on/off with the group?

Lindansaren, 2020-11-17:
I do not have groups set up in deconz only in HA. Do you mean groups in Deconz?

Lindansaren, 2020-11-17:
[image]

Mimiix, 2020-11-17:
Did you update to the latest version?

Lindansaren, 2020-11-17:
Yes, I have. Still issues.

Lindansaren, 2020-11-17:
Pressing a 0 on the node gets it back to life.

DerOetzi, 2020-11-17:
Same issue here. Having about 60 nodes at the moment. There are 3 IKEA TRADFRI bulbs and 1 osram plug in different rooms of the house, which not reacting randomly on any REST call by phoscon or openHAB. Yesterday I had not the time to directly go to deconz. After about a hour the bulb which is triggered together with 5 other bulbs by a motion sensor via openHAB starts reacting again without any interaction from my site.

Version: 2.05.88
Firmware: 26660700

Lindansaren, 2020-11-17:
@DerOetzi I have experienced the same. Leave the light for an hour or so also gets it to react. According to my mesh the node have several "lines" connected. Should not be an issue.

Lindansaren, 2020-11-17:
My wife is not impressed when her bedside lamp decides to not communicate anymore. Or when the nightlamp in the bathroom connected to a motion sensor does not turn on. =(

DerOetzi, 2020-11-17:
@Lindansaren are those always the same nodes not reacting like on my side?

Lindansaren, 2020-11-17:
I have done some more investigations today and noted that it's boiling down to maybe 10 different nodes. 8 IKEA (all versions including outlet) and 2 Philips Hue (E14)

SwoopX, 2020-11-17:
@Lindansaren You seem to have too much in the air simultaneously. With all those nodes, you sure you're using unicasts (multiple lights going on only one by one e.g.)? Using zigbee groups and therefore groupcasts should ease such situations and also be resilient to routing issues.

I mostly only notice after a couple of days that my smart home system is down (middleware seems to have a memory leak somewhere) while primarily using groupcasts.

Lindansaren, 2020-11-17:
@SwoopX The problem is that this is occuring not only when sending multiple commands. This is also an issue when trying to reach individual nodes manually from HA. As stated before. Going in to VNC and pressing 0 on the node make it controllable again.

Markkuuss, 2020-11-17:
> I do not have groups set up in deconz only in HA. Do you mean groups in Deconz?

Yes, I mean groups in Deconz. Add the affected light in Deconz to a group and try to switch the group via API or WebApp.

I have a similar problem with a Ubisys S1-R. It doesn't react anymore when I control it directly. But it can always be switched via the group. Can you please check if you have the same behavior.

Lindansaren, 2020-11-23:
Hmm, ok. Will try groups. But how will that help me when I'm having issues controlling even single bulbs? Seems like the problem are getting worse over time until I restart deconz. After a while multiple bad connections appear (many more "red lines") and I start to loose connection to lights. The connection is always restored if I press 0 in the vnc interface on that specific node. Please help. =)

MattWestb, 2020-11-23:
@Lindansaren If using light groups with lights and light controlling devices (like on/off remote) and not ZH switches that is bounded to the group they is sending commands to the group also the deCONZ is odd line.  One example: IKEAs motion sensor is bounded to one group with 3 bulbs. Then its triggering motion is sending one light command "ON with off 180(secunds)". All devices is the group is getting  the command and is going on for 180 seconds and then off if not one more motion is triggered and getting one new "on with off 180". All groupe commands is broadcast so its being sended to all routers in the network and is not depending if some routers is not working well or not if other routes still working.
Then using HA switches they is sending unicast that must being addressed and routes thru the network and also finding the right route if it not known / or for the moment is not working.
The information the HA switch is not commands that can controlling lights only "click click" that must being translated to "Light on / off / toggle" so the light can understanding it and executing the command.

In the zigbee works HA profiles is for sensors and switches that sending "click click" that can being funny, and the light profile is controlling devices (remotes, dimmer. light switches) is sending real commands like On. Dimm down, Colour move XY and so on that lights can reacting on with or without nodered or HA. If like controlling lights from HA or somthing with individual unicast commands (no zigbee groupe broadcast commands) it is working if not having so much devices that need being controlled.
But its 110% the wrong way and abuse of the zigbee network funktionality.

Its like live chatting with one person that is talking mongolian and im (is not) talking spanish and both must using google translate, Its not going very fast for getting the message thru. If both can speaking the same language example Icelandic then no google translate is needed and mutch misunderstandings is avoided and its much FASTER.

The Zigbee and ocupacion profile is not perfect but its still very good if being used right and it is part of the zigbee 3 standard as the HA profile and can being used parallel and doing great things together but dot do one big mess of it by abusing the functionality then you getting all problems you can getting.

I wishing you all the luck with the HA light steering in the future.

Lindansaren, 2020-11-24:
@MattWestb Frankly I don't really understand what you meant with that text? Do you imply that my usage of Zigbee is incorrect, and how should I change my setup in order to make it work?

Example of current usage:

- I use node red for creating flows that tell my zigbee bulbs and plugs to turn on or off.

- In node red i listen to a deconz event from a timer or a button

- If an event is triggered I send a command to ONE light or several that is grouped together (grouped in HA).

- Result is that 90% of the time the commands are performed without issues, but in some cases the bulb wont respond at all. It does not respond in any way. Have been trying the group function in phoscon now as wel but it behaves in the same manner. The light is even greyed out in phoscon. That is until I press 0 in vnc deconz on that particular node OR re-pair the light OR reboot the server OR just the deconz (which triggers a rebuild of the mesh).

- I have noticed that after a while i'm getting more and more "red lines" in the vnc interface between nodes. If I reboot it is stable for some time but after a day/half a day the communication starts to break again.

Please help, someone.

Lindansaren, 2020-11-24:
> Yes, I mean groups in Deconz. Add the affected light in Deconz to a group and try to switch the group via API or WebApp.
>
> I have a similar problem with a Ubisys S1-R. It doesn't react anymore when I [...]

Have tried now. It wont work with the light included in a group. The bulb shows exactly the same behavior. The rest of the lights in that particular group is indeed turning on/off as expected. Not even when I try to control the group from webapp....

MattWestb, 2020-11-24:
@Lindansaren If you is having red lines between devices in deCONz GUI you is very likely having network problems or interference that is blocking the radio communication in the network.
Sending commands to one device is unicast and must going thru one known route and its not so reliable then the network is unstable. Grupe commands is broadcasts and can going around blocked routes if not hole network is being completely blocked.

Use one WiFI analyser app for getting one picture of the wifi channels and adjust wifi / zigbee so they is having best mapping. Be aware Zigbe and WiFi channels numbers is not the same.

Lindansaren, 2020-11-24:
Wifi and Zigbee are not interfering with each other in my network. 

1. Could someone explain to how how I can use group commands instead och unicast when using HA and node red?
2. How would the unicast/group issue explain that random nodes become "unreachable" until i press 0 in VNC or reboot?

MattWestb, 2020-11-24:
For point 1: use the old web app / Wireless Light Control and on tab groups making new light groupes and adding light to them. If you have some real light switch its being there 2 as one light groupe.
deCONZ is sending the new groups to HA so you can using them there without doing anything more.
[image]

For point 2: i don't knowing then i don't have that problem longer then i have moving most devices to ZHA.

Lindansaren, 2020-11-25:
Found one setting in deconz vnc. "Destination settings". Changed that to "Broadcast routers", and now I haven't had a single issue in 24 hours! Whats the catch? =)

MattWestb, 2020-11-25:
I think the option "Destination settings" is for then you is GUI is sending commands to the aktive (marked) device. Example you is marking one light and the on / off cluster and clicking toggle for sending one toggle command to the device. The default the command is sent as unicast to the device. If changing to broadcast the toggle command is sent as one broadcast.
@Mimiix can you please correcting my if am wrong !!

I think you is having routing problem in your mesh then broadcast command looks working better then unicast. The problem only using broadcasts is that the underlying network layer is limiting broadcast commands to ca 1,2 / second for eliminating broadcast storms that is killing the network.
IKEA is using nearly broadcast for all but that is not working if building large systems (that tradfri is not constructed for).

Mimiix, 2020-11-25:
> I think the option "Destination settings" is for then you is GUI is sending commands to the aktive (marked) device. Example you is marking one light and the on / off cluster and clicking toggle for [...]

I am not sure on this subject :( @SwoopX knows better :)

SwoopX, 2020-11-25:
@Lindansaren The point we're trying to make here is that if you would exclusevly use zigbee group commands here, you wouldn't have that issue any more. However, I understand this is not the final solution to this behaviour and you might not be able to solve all applicable use cases for your environment.

Btw, also have seen that you apparently identified another way to dance around it.

closed 2020-12-25 by stale[bot]