dresden-elektronik/deconz-rest-plugin #3684: BenexMart 3 gang switch

ContraNL, 2020-11-17:
This switch is similar to issue 3524. But on version 2.06.00 it does not show up in Phoscon, it does in deconz GUI and within Home Assistant I only see a battery sensor.

**Device:**

- Product name: Benexmart Smartlife and Tuya Zigbee Smart Switch Touch 1 2 3 Gangs Realize Multiple Ways Of Control The Smart Home Equipment
- Manufacturer: empty
- Model identifier: empty, but Aliexpress says model number 86
- Device type: switch

Purchased here: [https://www.aliexpress.com/item/4000793877542.html?spm=a2g0s.9042311.0.0.6b6e4c4dTU0UO2]

I have the three gang version

**Screenshots:**

Screenshot of the Node Info Panel:
[image]

Screenshot of the Basic cluster:
[image]
[image]

Screenshot of Cluster Info - Switch 01 Power Configuration
[image]

Screenshot of Cluster Info - Switch 01 On/Off
[image]
[image]

Screenshot of Cluster Info - Switch 01 OTAU
[image]

Screenshot of Cluster Info - Switch 02 Power Configuration
[image]

Screenshot of Cluster Info - Switch 02 On/Off
[image]
[image]

Screenshot of Cluster Info - Switch 03 Power Configuration
[image]

Screenshot of Cluster Info - Switch 03 On/Off
[image]
[image]

Hope you have all the information you need. If you need more or this device should already work on version 2.06.00 and I can help on issue 3524 please let me know.

Smanar, 2020-11-17:
Missed ^^, the 3 th capture haven't value. Yo uneed to press to press the "read" button and make the device awake (if it need)

ContraNL, 2020-11-17:
I can't get more of this out of that capture. I have removed the node and after adding it again I get this:

[image]

Smanar, 2020-11-17:
No model identifier ?
You have pressed the "read" button with making the device awake in same time ?

ContraNL, 2020-11-17:
[image]

I've tried again and when I continuously press a button on the device and than the read button I get more info.
See capture.

Smanar, 2020-11-17:
Lol hourra another tuya device ....

But will be fast, do you have Unix machine to compile and test the code ?

ContraNL, 2020-11-17:
Sorry mate, I now you hate them :) 

Yes, I have an Ubuntu machine where I run deconz (docker) on. So if you run me though the steps we need to take, I can use it to compile and test. Will be on a production environment but seeing it is docker it shouldn't be a problem.

Smanar, 2020-11-17:
Noo, and you are lucky, the device is already in code > "_TZ3400_key8kk7r"

You have it in beta version https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.6.0-beta

Smanar, 2020-11-17:
The battery level work ? You have already included the device ? with a bad mac adress ?

ContraNL, 2020-11-17:
Yes:

[image]
[image]

Smanar, 2020-11-17:
Ha ok, it is included as "light" so the blacklist haven't worked.
On a normal inclusion it will included as "sensor", and you can have a "light" sensor, but totaly useless

But for the battery I relay don't understand, can i have the device Json if you can, or just the uniqueID, pls ?

If you want to make a try I can explain you how to compile the code, you can use it immediatly and can be usefull for next device ^^.

To resume you have the procedure here  https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So for you
```
sudo apt install deconz-dev
git clone --branch tuya_benexmart https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Then re include the device, will be easier if you don't delete the previous one (tuya devices need 2 inclusions  ATM)

ContraNL, 2020-11-18:
I'm running the docker image from marthoc. So far I can see I can't install anything on the image when I open it with bash.
How to procede?

I can create a VM with ubuntu and install deconz the way you explained in the previous comment.
I can install a new NUC with Ubuntu and install deconz on it.
I can install deconz-dev on the root of my device, but I don't know if it will play nice with the docker image.

What is the best way to go?

Smanar, 2020-11-18:
Not the third ^^, you are right, avoid using docker.

ContraNL, 2020-11-18:
I’m installing a Ubunto VM and will install deconz en deconz-dev on it. Will report back when done.

ContraNL, 2020-11-19:
I've created a Ubuntu VM with DeconZ, added the sensor, installed deconz-dev and your plugin, included sensor again.
What do you want me to do next? How to get the device Json or uniqueID?

ContraNL, 2020-11-19:
Googling around a bit I think you want this information from the API?

[image]

Smanar, 2020-11-19:
Exactly.
And if the code is good, the field "buttonevent" will represent the button pressed

X00Y
- X is the button number.
- Y is the event.

And you can too configure HA to use this deconz, but if the fied value change, for me it s good.

ContraNL, 2020-11-21:
The field buttonevent in the API changes. 

**first button:**
1002 = single press
1003 = long press
1004 = dubble press

**second button:**
2002 = single press
2003 = long press
etc.

Is this all you need to know?

Smanar, 2020-11-21:
I think yes ^^
So all is working ? I can make the PR, you don't miss something ?

ContraNL, 2020-11-21:
As far as the button events, all show up. I haven’t tried HA seeing I don’t have it installed on the VM. You want me to test it or wait for beta? Seeing I don’t have any devices to trigger on button event within the DEV envrinment. But I can create it if you want.

Smanar, 2020-11-21:
It s for you, but for me it's ok, HA will use the same API, so if you see the good modification when looking yourself, HA will do same.

ContraNL, 2020-11-21:
Thanks for adding the support. When it gets to beta I will test in my production environment and report back if everything works well together.

Smanar, 2020-11-21:
The update is minot, I think it will be in the stable (earlier)

ContraNL, 2020-11-21:
I hope so, when you know please tell me.

Smanar, 2020-11-21:
Easy to know, when the PR will be validated, the issue wil be closed automaticaly.

ContraNL, 2020-11-21:
Perfect, thanks.

closed 2020-11-30 by manup

ContraNL, 2020-12-08:
@Smanar , I'm using 2.07.00 of the deconz plugin but I'm still seeing the units being added as light or battery only.
This ticket has been mentioned in the changes (Support Benexmart 3 gang switch variant (TS0043) #3724) but for now it's the same result as before the change.

Can you help? What am I doing wrong?

Smanar, 2020-12-08:
>After checking the API I see the button event values changing as we saw in the DEV environment.
But Home Assistant shows the same as before.

So the value change in the API, but not in HA ? You have not same kind of problem for other sensors ? not a mistake beetwen light and sensor ?

And no, the bug with light device creation is still here. On the code if you have at least 1 sensor with the good uniqueid and a correct manaufacture name, it can prevent the light device creation, but it seem not working.

On my side I just select the node and press delete.
Or you can just delete the api entry using for exemple

`curl -X DELETE http://192.168.1.50:8080/api/1GF311EB72/sensors/5`

ContraNL, 2020-12-08:
It’s not that the value is not changing. There are no values in HA to change.

[image]
[image]

I only have a battery for the one unit. And still three lights for the other. Even when I delete the nodes from deconz I don’t see them disappearing in HA. When looking in deconz I see the units reappear if I wait long enough even without pairing/re-adding them.

ContraNL, 2020-12-08:
@Smanar, I have run the service remove orphaned entities and that removed the old light entities.

Seeing this is my first switch I found that these switches don’t make entities but events. I will check this out and report back if everything works.

Source of information:
Remote control devices

Remote controls (ZHASwitch category) will not be exposed as regular entities, but as events named deconz_event with a payload of id and event. Id will be the device name from deCONZ and Event will be the momentary state of the switch.

ContraNL, 2020-12-09:
@Smanar, sorry m8. Everything is working great. Thought it would create entities but needed to use deconz_event.

Smanar, 2020-12-09:
Ok, so except the boring useless light device, all is working ?

ContraNL, 2020-12-10:
Yes, and even the boring useless light devices are gone after a HA service run of deconz.remove_orphaned_entries