dresden-elektronik/deconz-rest-plugin #3687: What are the effects of "Read Bind Table" in deCONZ 2.05.88 ?

yves19, 2020-11-17:
## Describe the question or issue you are having

Hello. Since release 2.05.88 of deCONZ a new function is avaliable through Edit Menu/Read binding table (CTRL + B).

My question is : 
- what does this function do exactly and what is its expected result ? 
- Is there any visual return to the user or is it a silent task which does not report any visual asset ?

## Screenshots

  No required screen shot

## Environment

 - Host system: Intel computer running Win10 as well as NUC Intel 8i5BEH HA box running Debian 10.5 
 - Running method:  Win10 as well as Debian 10.5 
 - Firmware version: (26660700)
 - deCONZ version: (2.05.88)
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

  No applicable logs

## Additional context

  No additional content

SwoopX, 2020-11-18:
You currently can leverage this only by sniffing the zigbee traffic. However, as I recall, a graphical representation of the currently available device bindings should be implemented in deconz GUI in near future.

yves19, 2020-11-18:
Thank you so much.
I am very impatient to have the implementation integrated.

Regards

closed 2020-11-18 by yves19