dresden-elektronik/deconz-rest-plugin #3689: TUYATEC-r9hgssol RH3001 Door/Window Contact Sensor

panenka, 2020-11-17:
Still not fully supported, previous case - Tuya Smart ZigBee Door Window Contact Sensor adding/pairing #2806

## Device

- Product name: The device name as shown on the product or package.
- Manufacturer:TUYATEC-r9hgsso
- Model identifier: RH3001
- Device type : 
  - Sensor

[image]
[image]
[image]
[image]

panenka, 2020-11-17:
edited

SwoopX, 2020-11-18:
The device is fully supported and sensor(s) has been created as depicted in your screenshot. For visibility in Phoscon, please raise a corresponding request in the phoscon-beta repository.

closed 2020-11-18 by SwoopX