dresden-elektronik/deconz-rest-plugin #3692: Lidl/Silvercrest Smart Motion Sensor

panenka, 2020-11-17:
Not possible to rename, not shown in Phoscon, found other thread Device name: SilverCrest Smart Motion Sensor #3628

Motion Sensor, tamper sensor

## Device

- Product name: Lidl/Silvercrest Smart Motion Sensor
- Manufacturer: _TZ1800_fcdjzz3s
- Model identifier: TY0202
- Device type : 
  - Sensor

## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-11-17:
Hello, you haven't the cluster list, pls ?

panenka, 2020-11-17:
> Hello, you haven't the cluster list, pls ?
> (and same for other device pls ^^), We realy need it for tuya devices, to know if we have a chance the device is using standard one.

Updated. Please could you open #3689 too, it has been automatically closed and I don't know how can I could open it again?

Smanar, 2020-11-17:
It seem this device is standard.

Do you have an unix machine to compile and thest the code ?

panenka, 2020-11-17:
No, Windows 10.

Smanar, 2020-11-17:
Ha bad luck ^^, 5 requests and not possible test them at all ...

panenka, 2020-11-17:
And what about virtual machine? It is enough?

Which OS and such guide?

Smanar, 2020-11-17:
It works on VM, but a little harder, Take Ubuntu or Raspbian (with desktop if you can).
And if you can have deconz running and are able to install dev package, it will be fine to continue

`sudo apt install deconz-dev`

ensey, 2020-11-18:
@Smanar I'm making a live usb with ubuntu on it, to test this through my laptop - what should I do after installing deconz-dev?

Smanar, 2020-11-18:
You can take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch lidl https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

After that you can run deconz and include the device. It will probably not be visible in phoscon, so you can use third app or direclty the api to check the device creation.

Not sure for battery return yet, but I think the rest will be ok, realy few modifications.

ensey, 2020-11-19:
@Smanar I am getting a "Unable to locate package deconz-dev"
Do you have any tips? I have tried the info from here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2811

ensey, 2020-11-19:
@Smanar I can confirm that I can add it in deconz. I have done this will out-of-home, so I havn't been able to confirm it with HA yet.
I will confirm with HA when I return home, to see if it shows up as an entity.

ensey, 2020-11-19:
[image]

Mimiix, 2020-11-19:
That's looking good. Should be working with HA.

Smanar, 2020-11-19:
You need to include it with phoscon, not deconz.
And generaly if it work, you will have a message.

davidsantin, 2020-11-21:
Thanks Smanar. Following your indications I can say that the sensor works perfectly for me in HomeKit. It is not visible in Phoscon, but I don't need it since with Homebridge it is visible in Homekit and I can set rules that work without any problem.

Smanar, 2020-11-22:
I haven't checked for battery, it is working on your side ?

davidsantin, 2020-11-22:
Yes, the battery level is displayed perfectly. Everything is correct. Thank you @Smanar

Smanar, 2020-11-22:
Pr done, thx to you.

marval, 2020-11-22:
> 
> 
> Yes, the battery level is displayed perfectly. Everything is correct. Thank you @Smanar

Could you change the name of the node? Maybe I did it wrong but I built the project on my local machine and imported it into a docker container. It seems to be connecting, but then I cannot change the name (of just this node) and I cannot really see it in node-red or HA.

davidsantin, 2020-11-22:
I can neither change the name nor the default room in which it appears in the HomeKit. I suppose this is a side effect of the fact that the sensor does not appear in the Phoscon application. This is not a serious problem for me because I can create automations even if it is not in the correct room or appears with the default name.

marval, 2020-11-23:
It would not be a dealbreaker for me either, but it seems I cannot see it in HA/node-red with the default name which is0xC432 in my case and not "TY0202" as shown by @ensey. @davidsantin, are you running deConz natively or in a docker container?

davidsantin, 2020-11-23:
@marval , I am running deconz natively on a Raspberry Pi. From what you say, the sensor has not been added correctly, as it should appear with the name TY0202. It took me several synchronization attempts to detect it properly. 

What I did was to press the reset button on the sensor for about 10 seconds, until the light went out, then I activated the search for new sensors in Phoscon and finally I pressed the sensor reset button again until the green light flashed quickly. At that point, you just have to wait for the correct name to appear in deconz.

After several attempts to reset the device and delete the node in the deconz, I managed to get it to appear with the correct name.

Mimiix, 2020-11-23:
@davidsantin @marval  A re-pair is needed after support was added.

marval, 2020-11-23:
Hmmm .. I have tried a couple of times now without luck. I delete the device via VNC, unpair it via the button, start the search on Phoscon and press the button on the device a second time. It is added, the clusters button gets available, but it stays like this:
[image]
If I wait a bit longer (5-10min) it will show the connection line to the Conbee and maybe a second line to a bulb that is close to it. Any ideas as to what might be wrong? I am sure that my .so in the docker image is updated with the newly compiled one.

davidsantin, 2020-11-23:
Try one last test. Deactivate the automatic end-point search in deConz, delete the device and start a new sensor search in Phoscon. Then go to the device and press the reset button for more than 15 seconds and wait. 

[image]

Smanar, 2020-11-23:
@marval, your was not suported (have just updated the code) look at your mac adress, they don"t start by same value.

marval, 2020-11-24:
Well spotted! I built the new .so from your repo but still have issues. Does this flow make sense. Is it missing something?
`
15:14:25:711 APS-DATA.indication from unknown node 0xEC1BBDFFFEE7AA8F

15:14:25:713 new node - ext: 0xec1bbdfffee7aa8f, nwk: 0xF982

15:14:25:714 device announce 0xEC1BBDFFFEE7AA8F (0xF982) mac capabilities 0x80

15:14:25:714 set fast probe address to 0xEC1BBDFFFEE7AA8F (0xF982)

15:14:25:747 FP indication 0x0000 / 0x0013 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:25:747                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:25:747 device announce 0xEC1BBDFFFEE7AA8F (0xF982) mac capabilities 0x80

15:14:26:601 [1] get node descriptor for 0xec1bbdfffee7aa8f

15:14:27:898 FP indication 0x0000 / 0x8002 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:27:898                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:27:898 ZDP indication search sensors 0xEC1BBDFFFEE7AA8F (0xF982) cluster 0x8002

15:14:27:898 ZDP indication search sensors 0xEC1BBDFFFEE7AA8F (0xF982) clear timeout on cluster 0x8002

15:14:27:945 [2] get active endpoints for 0xec1bbdfffee7aa8f

15:14:28:751 FP indication 0x0000 / 0x8005 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:28:751                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:28:751 ZDP indication search sensors 0xEC1BBDFFFEE7AA8F (0xF982) cluster 0x8005

15:14:28:751 ZDP indication search sensors 0xEC1BBDFFFEE7AA8F (0xF982) clear timeout on cluster 0x8005

15:14:28:758 [3] get simple descriptor 0x01 for 0xec1bbdfffee7aa8f

15:14:30:758 FP indication 0x0000 / 0x8005 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:30:758                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:30:758 ZDP indication search sensors 0xEC1BBDFFFEE7AA8F (0xF982) cluster 0x8005

15:14:30:764 wait response fastEnddeviceProbe() 0xEC1BBDFFFEE7AA8F, elapsed 2005 ms

15:14:31:758 FP indication 0x0000 / 0x8004 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:31:758                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:31:758 ZDP indication search sensors 0xEC1BBDFFFEE7AA8F (0xF982) cluster 0x8004

15:14:31:758 ZDP indication search sensors 0xEC1BBDFFFEE7AA8F (0xF982) clear timeout on cluster 0x8004

15:14:31:764 [3.1] get IAS Zone type for 0xec1bbdfffee7aa8f

15:14:36:846 FP indication 0x0104 / 0x0500 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:36:847                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:36:847 Clear fast probe timeout for cluster 0x0500, 0xEC1BBDFFFEE7AA8F

15:14:36:853 [4.2] get basic cluster attr 0x0004 for 0xec1bbdfffee7aa8f

15:14:38:830 FP indication 0x0104 / 0x0000 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:38:830                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:38:830 Clear fast probe timeout for cluster 0x0000, 0xEC1BBDFFFEE7AA8F

15:14:38:836 [4.2] get basic cluster attr 0x0005 for 0xec1bbdfffee7aa8f

15:14:40:833 FP indication 0x0104 / 0x0000 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:40:833                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:14:40:833 Clear fast probe timeout for cluster 0x0000, 0xEC1BBDFFFEE7AA8F

15:15:13:357 FP indication 0x0000 / 0x0013 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:15:13:357                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:15:13:358 device announce 0xEC1BBDFFFEE7AA8F (0xF982) mac capabilities 0x80

15:15:13:358 FP indication 0x0000 / 0x0013 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:15:13:358                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:15:13:358 device announce 0xEC1BBDFFFEE7AA8F (0xF982) mac capabilities 0x80

15:16:14:502 FP indication 0x0000 / 0x0013 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:16:14:502                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:16:14:502 device announce 0xEC1BBDFFFEE7AA8F (0xF982) mac capabilities 0x80

15:16:14:502 FP indication 0x0000 / 0x0013 (0xEC1BBDFFFEE7AA8F / 0xF982)

15:16:14:502                       ...     (0xEC1BBDFFFEE7AA8F / 0xF982)

15:16:14:502 device announce 0xEC1BBDFFFEE7AA8F (0xF982) mac capabilities 0x80`

Smanar, 2020-11-24:
Can you check  the model and the manufacture number ?
Model id = "TY0202"
Manufacture number = 0x120B

But no, nothing special in log, it s the normal procedure for new device, even not complete.

marval, 2020-11-24:
[image]
[image]

The logs are after leaving it for like 10 minutes so this is the last thing in the log before trying to get the clusters read. If it is not complete it might be failing during the adoption process? What would be the completion log entry?

Smanar, 2020-11-24:
All seem fine, a trick you can do:
- set phoscon in permit join (add new sensor)
- read again basic attribute (cluster 0x0000, button "read") with making the device awake if needed.

Without doing inclusion procedure, the device need to be already in deconz.

Jos-77, 2020-11-27:
mine is found in deconz but not in phoscon, i read that some people deactivate the automatic end-point search in deConz, how?

i'm using a rpi 4 + conbeeII usb stick.

any ideas??

Smanar, 2020-11-27:
You have recomplied the code too @Jos-77 ?

Jos-77, 2020-11-27:
New in deconz, how is it done? Do you have a Guide?

Smanar, 2020-11-27:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3692#issuecomment-729936950

But you need an Unix OS, else you need to wait for the PR will be validated.

Jos-77, 2020-11-27:
I use ha io, so I have to wait, any idea how long?

Smanar, 2020-11-27:
>Next Beta: v2.7.0 expected at the 15th of December. Next Stable: v2.6.2 expected at the 8th of December.

As the modification is minor, perhaps the stable.

closed 2020-11-28 by manup

marval, 2020-12-01:
Just wanted to update this. Seems the sensors are now working (detected as Presence XX) with the latest update of the deConz docker image using the latest beta.

joostvanmourik, 2021-01-28:
I'am a bit new to this but i can't add the sensor, wich version should support it and do i need to doe something

Mimiix, 2021-01-29:
Since 2.7.1. What button are you using to reset it @joostvanmourik ? Make sure to use the small pinhole.

OsiMood, 2021-01-31:
I am trying to connect the Smart Wireless Door Bell but it is a no go.... it does not connect. Anything special to do?