dresden-elektronik/deconz-rest-plugin #3693: Tuya Double Switch TS0012 turns itself off again.

schumi2004, 2020-11-17:
## Describe the bug
Connected the wired double Tuya switch (no-neutral) switch as described and paired it with Deconz (Home Assistant).
When turned on manually or via Deconz the switch tunrs itself off again after approximatly 2 min.
Looks like some timer got activated but can't find any documentation about it.

And sometimes, after it happened for a few times, it becomes unresponsive and only manual activation makes it work again.

## Steps to reproduce the behavior
Connect the switch as instructed, pair it with Deconz and just turn it on, manually or using Deconz/Home Assistant.

## Expected behavior
Just stay on.

## Screenshots
n/a

## Environment
 - Host system: Raspberry Pi 3B+
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26660700
 - deCONZ version: 2.05.88 / 15-10-2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? SSD

- switch used: 2 gang switch TS0012

## deCONZ Logs
Need to figure out how to save the logs to a file in HassIO.

## Additional context
n/a

Smanar, 2020-11-17:
Hello, can you make a try with the api disabled ? Plugin / rest API plugin.

schumi2004, 2020-11-17:
I'm not sure how to do this?
I checked the Home Assistant Add-on: deCONZ in my HassIO setup but can't find anything to disable api?

I removed the port in HA deCONZ addon configuration that disables API backend, is that what you mean?
[image]

For now, I unpaired the switch and it functions normal again as a regular switch.
Will pair it again later this evening when I have more time.
Is it also possible I caused this behavior using a combination of button presses?
At first, I thought it did not pair so I tried it again to find out they were added underneath the lights section. Maybe a reset of the switch (if possible?) will help also?

Smanar, 2020-11-17:
Ha, sorry it s only possible using the GUI.

It s possible too with removing the lib file from the folder, but I don't want to broke your setting. Was to make the same test than you, the device in the network but without the API, to check if it s from zigbee or api side.

For the rest, idk, never see a timer feature in this kind of switches. Perhaps something visible in log (you have it in phosocn too now, in "help")

megawubs, 2020-11-18:
@schumi2004 Not really related to the topic, but where did you got this switch? I've been searching for something like this for a while now. But can't really find any shop selling this when I search for it on the web.

schumi2004, 2020-11-18:
@megawubs https://nl.aliexpress.com/item/4000298926256.html?cv=47843

@Smanar 
The HassIO deCONZ addon does have a GUI.
I can choose from Phoscon or Deconz, i'll check again when i have some more time inbetween work.

Smanar, 2020-11-18:
>The HassIO deCONZ addon does have a GUI.
I can choose from Phoscon or Deconz, i'll check again when i have some more time inbetween work.

Ha ? are you sure of that ?
But if you have the deconz GUI
[image]

schumi2004, 2020-11-18:
Got it unchecked, going to test it now.
[image]

Smanar, 2020-11-19:
You can pair it, and disable the switch after, or just let it already paired.
It s just to check if you have the same issue with the api disabled, only with the zigbee network alone. The important part is not pairing, but to check if you still have the "timer" bug.

schumi2004, 2020-11-19:
Switch still functions as expected since yesterday. The only difference is that I only did the pairing procedure once and not twice as the first time. 
I think that the very first time I paired the switch and thought it did not work I activated some feature during second attempt that caused this auto off. I took 2 minutes every time it turned itself off (used a timer to measure) and because of that I think something got activated during pairing when it already was.

Yesterday when I wanted to test I already had deleted the device and I needed to enable REST API again for it to get paired and since then it works flawless.

Maybe we should close this issue for now and reopen when I have new devices with similar behavior.

Smanar, 2020-11-20:
This is no documentatin with the device ? even in chinese ?

schumi2004, 2020-11-20:
Documentation is shipped with it but doesn't describe "advances" features like auto off or something like that (if any). It only describes how to wire and pair it for different versions (neutral, no neutral, wifi, zigbee, rf433)

[image]
[image]
[image]

kind3r, 2020-12-08:
Had the exact same issue, obviously because of the 'double pairing'. I thought that you have to pair each gang individually.

Deleted the switch from Phoscon app (removing one light removes the other also) and then repaired and now it's ok. I'm thinking I should do this with some of the other sensors that sometimes missbehave.

kind3r, 2020-12-08:
I think I spoke too soon, after some time they started automatically turning off again ...

Smanar, 2020-12-08:
I have perhaps an idea for tuya, but not sure It's that, because I have the ame bug on different brands.
Can you confirme me, this device use tuya cluster ?
Just enable log in deconz, press a button, and give me the lines with "debug tuya", all lines pls,  even if there are in double.

kind3r, 2020-12-08:
I'm not sure what log types need to be activated and what version (I'm running 2.5.88), but my switch is identified as `_TZ3000_fvh3pjaz` or `Heiman`:

```
20:21:50:962 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) 
VALUES ('13', 'normal', 'bc:33:ac:ff:fe:f5:b4:db-01', 'Baie', '65520', '1', 'TS0012', '_TZ3000_fvh3pjaz', '', '{"attr/id":"13","attr/lastannounced":null,"attr/lastseen":"2020-12-08T18:21Z","attr/manufacturername":"_TZ3000_fvh3pjaz","attr/modelid":"TS0012","attr/name":"Baie","attr/swversion":null,"attr/type":"On/Off light","attr/uniqueid":"bc:33:ac:ff:fe:f5:b4:db-01","state/alert":null,"state/on":true,"state/reachable":true}')
20:21:50:965 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) 
VALUES ('14', 'normal', 'bc:33:ac:ff:fe:f5:b4:db-02', 'Hol', '65520', '2', '', 'Heiman', '', '{"attr/id":"14","attr/lastannounced":null,"attr/lastseen":"2020-12-08T18:21Z","attr/manufacturername":"Heiman","attr/modelid":null,"attr/name":"Hol","attr/swversion":null,"attr/type":"On/Off light","attr/uniqueid":"bc:33:ac:ff:fe:f5:b4:db-02","state/alert":null,"state/on":false,"state/reachable":true}')
2
``` 

What I noticed is that after pairing it works fine for a few hours and then randomly switches to the auto-off behaviour.

Smanar, 2020-12-08:
You haven't the GUI ?

Yo have not line starting with "Tuya debug X" when using a button ?

kind3r, 2020-12-08:
As most of the people out there I'm using a Conbee 2 via the Home Assistant addon. More specifically HASS.io on a PI 4. I do have access to the deCONZ app via VNC but can't copy paste from there and finding something in there is a pain. I can use docker logs from the host, but I need to enable the logs in the addon's config. I have enabled `dbg_info: 2` but I found nothing Tuya related.

k0rn3l, 2020-12-14:
I have just installed two new switches: https://www.aliexpress.com/item/4000298926256.html
I have the same issue with auto-off after 2-3 minutes on both (X711A and X712A models). I have paired them only once.
Deconz plugin 2.07.01 / 8.12.2020 with home assistant
Conbee 2 FW: 26660700

k0rn3l, 2020-12-15:
I did some further investigation...
I did factory reset of the node from deconz vnc app; paired it again; and at first it looked like the problem was gone when I was switching the light ON from home assistant. It stayed ON. But as soon as I used the switch button, the problem returned and now the light turns itself off after 2 minutes no matter if I use home assistant or the button to turn the light on.
Deconz log from self turn off event:
`01:30:39:194 DeviceAnnce of LightNode: 0xbc33acfffeedfc0f Permit Join: 0
01:30:40:197 ZCL attribute report 0xBC33ACFFFEEDFC0F for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
01:30:40:289 verified group capacity: 255 and group count: 1 of LightNode 0xbc33acfffeedfc0f
01:30:40:290 0xbc33acfffeedfc0f found group 0xFFF0
01:30:41:758 binding for cluster 0x0006 of 0xBC33ACFFFEEDFC0F exists (verified by reporting)
01:30:41:758 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0xBC33ACFFFEEDFC0F (seems to be active)
01:30:54:023 ZCL attribute report 0xBC33ACFFFEEDFC0F for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000
`

k0rn3l, 2020-12-15:
Workaround:
Following info on https://zigbee.blakadder.com/Lonsonho_X711A.html
I forced entering pairing mode (10s button press) when already paired with ZB network, then I clicked "Add new lights" in Phoscon app, switch repaired with the same network and is now working correctly.

kind3r, 2020-12-15:
I noticed the same behaviour with the latest firmware, I left the switch unpaired for a week, yesterday I paired it again, after that I rebooted my PI and it was gone, so did a re-pair (altho the switch lights showed that it was paired) and it showed up again in Phoscon and HA with the names that I previously set and it's been working fine since. Usually it takes up to 2h for the auto-off behaviour to kick in, but now it seems to be fine for the past 20h.

Smanar, 2020-12-15:
Ha yesssss, thx a lot, It help so much, I was totaly lost, no idea at all.

> It seems like there’s two versions of this switch coming from AliExpress. The TS0011 appears to turn off by itself after two minutes after being paired for the first time with ZHA. To get it to function correctly after pairing the first time, press and hold the switch for 10s to enter pairing mode and then press ‘Add Devices Via This Device’ in your Zigbee Coordinator in Home Assistant. After that, it should function correctly.

gregskh, 2020-12-21:
Hello,

I have TS0011 and TS0012 switches and They turned off after 2 minutes no matter if I use home assistant or the button to turn the light on. After this procedure everything works ok:

It seems like there’s two versions of this switch coming from AliExpress. The TS0011 appears to turn off by itself after two minutes after being paired for the first time with ZHA. To get it to function correctly after pairing the first time, press and hold the switch for 10s to enter pairing mode and then press ‘Add Devices Via This Device’ in your Zigbee Coordinator in Home Assistant. After that, it should function correctly.

Original source: https://zigbee.blakadder.com/Lonsonho_X711A.html

parowa92, 2020-12-26:
Same here.
`01:18:38:486 DeviceAnnce of LightNode: 0xbc33acfffe7720c2 Permit Join: 0
01:18:38:487 DeviceAnnce of LightNode: 0xbc33acfffe7720c2 Permit Join: 0
01:18:39:470 verified group capacity: 255 and group count: 1 of LightNode 0xbc33acfffe7720c2
01:18:39:470 0xbc33acfffe7720c2 found group 0xFFF0
01:18:40:219 verified group capacity: 255 and group count: 1 of LightNode 0xbc33acfffe7720c2
01:18:40:219 0xbc33acfffe7720c2 found group 0xFFF0
01:18:40:904 ZCL attribute report 0xBC33ACFFFE7720C2 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
01:18:41:025 ZCL attribute report 0xBC33ACFFFE7720C2 for cluster: 0x0006, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
01:18:47:604 Current channel 15
01:18:47:612 Device TTL 4506 s flags: 0x7
01:18:53:965 ZCL attribute report 0xBC33ACFFFE7720C2 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
01:18:54:084 ZCL attribute report 0xBC33ACFFFE7720C2 for cluster: 0x0006, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
01:19:03:090 binding for cluster 0x0006 of 0xBC33ACFFFE7720C2 exists (verified by reporting)
01:19:03:091 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0xBC33ACFFFE7720C2 (seems to be active)
01:19:34:091 binding for cluster 0x0006 of 0xBC33ACFFFE7720C2 exists (verified by reporting)
01:19:34:092 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0xBC33ACFFFE7720C2 (seems to be active)`

Smanar, 2020-12-26:
And have you done the procedure ?

parowa92, 2020-12-26:
Yes, but if i disable api i cant use Tradfri 5 button remote, so it's not a solution for me.
`10:47:07:613 Device TTL 2865 s flags: 0x7
10:47:15:112 DeviceAnnce of LightNode: 0xbc33acfffe7720c2 Permit Join: 0
10:47:15:113 DeviceAnnce of LightNode: 0xbc33acfffe7720c2 Permit Join: 0
10:47:16:099 verified group capacity: 255 and group count: 1 of LightNode 0xbc33acfffe7720c2
10:47:16:099 0xbc33acfffe7720c2 found group 0xFFF0
10:47:16:845 verified group capacity: 255 and group count: 1 of LightNode 0xbc33acfffe7720c2
10:47:16:845 0xbc33acfffe7720c2 found group 0xFFF0
10:47:17:280 ZCL attribute report 0xBC33ACFFFE7720C2 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
10:47:17:396 ZCL attribute report 0xBC33ACFFFE7720C2 for cluster: 0x0006, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
10:47:31:127 ZCL attribute report 0xBC33ACFFFE7720C2 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
10:47:31:249 ZCL attribute report 0xBC33ACFFFE7720C2 for cluster: 0x0006, ep: 0x02, frame control: 0x08, mfcode: 0x0000 
10:47:39:091 binding for cluster 0x0006 of 0xBC33ACFFFE7720C2 exists (verified by reporting)
10:47:39:092 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0xBC33ACFFFE7720C2 (seems to be active)
10:47:47:658 	0xBC33ACFFFE7720C2 force poll (2)
10:47:47:663 ZCL attribute report 0xBC33ACFFFE7720C2 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 `

Smanar, 2020-12-26:
Disable the API ?
Nope, the procedure to remove this situation, on this model.

parowa92, 2020-12-26:
I tried to pair it again, after pairing. Next i tried this on ZHA, stil same problem. I just moved to z2mqtt and its gone.
When i disabled api the switches still was working, but with this problem.

kind3r, 2020-12-26:
After 10 days of working fine they suddenly started to turn off again (after a reboot, but it was not the first during those 10 days). I have tried double re-pairing (3 times) but still the issue of them turning off by themselfes comes back after some time (1-2 hours).

parowa92, 2020-12-26:
> After 10 days of working fine they suddenly started to turn off again (after a reboot, but it was not the first during those 10 days). I have tried double re-pairing (3 times) but still the issue of them turning off by themselfes comes back after some time (1-2 hours).

Try zigbee2mqtt, today i switched from deconz and it is working flawless.

kind3r, 2020-12-28:
Indeed, I have also tried ZHA and get the same auto-off behaviour, but from the start (in deConz most of the time it takes a few hours for the switch to start turning off).
I installed Z2M yesterday and it's been working flawlesly ever since, even pairing all the other devices was a smoother experience. The only thing I noticed that it's a bit less responsive than deConz, but I think this is because of the timeouts (`adapter_delay`) that are not properly set in the driver so I have created a PR for it.

notovich, 2021-01-01:
having the same issue, seems hardware issue, as switch was auto off after 2m and 2s exactly, no matter what i'm doing it is turning off, tried deleting/rairing/re-pairing nothing helps, it drives me crazy.

Smanar, 2021-01-01:
And that don't  work for you too ?

> It seems like there’s two versions of this switch coming from AliExpress. The TS0011 appears to turn off by itself after two minutes after being paired for the first time with ZHA. To get it to [...]

Rem-me, 2021-01-03:
Hello,
My two cents about this issue : 
I have baught several zigbee tuya switches from amazon/yagusmart and aliexpress/zemismart and it seems I got two familly of switches: 

- one with _TYZB01_* manufacturer, which has no problem, has no light when it's switched off (even if paired) and has a red light when switched on
- one with _TZ3000_* manufacturer, which works as expected for a couple of hours after been paired, and then switches off by itself after two minutes and 2 or 3 seconds. When not paired, it shows no light if it is off, and red light if on (like the TYZB01). When paired, it shows green light if off, and orange if on. 

I've got both TYZB01 and TZ3000 from amazon/yagusmart (in one sending !) and only TZ3000 (2 month later) from zemismart. The two kinds have the exact same packaging. Maybe it's a change of internal design for the same product.

[image]

Under deConz, TZ3000 has  a "time" cluster not present on TYZB01
[image]
 
I noticed several things about the TZ3000 : 

- In deConz or Phoscon Web UI, unpair the switch ("delete device") removes the green light (ok, not a surprise) and removes the auto-off behavior.
- Once the auto-off appeared, physically disconnecting the gateway (plug off the conbee 2 usb key) doesn't change anything. It's like the switches had been put in a sticky state (even with power supply cut)
- After an auto switch off, if you switch on for, say, 30 secs, then switch off and on again, the switch will turns itself off after 1 min 30 sec... As if there were a counter reset after 2min.
- And last but not least : as soon as paired through zigbee2mqtt (deConz stopped before), the auto-off disapears

Smanar, 2021-01-03:
Perhaps need to make something with the time cluster ....

ffragnol, 2021-01-10:
Hello i have exactly the same problem with TZ3000 version. Did you solved it ?

notovich, 2021-01-10:
Fixed by moving from zha to z2m - please note you can't run both, moving to z2m you will need to re-discover all devices

ffragnol, 2021-01-10:
It's not a fix for the inital problem. 👎  How get it worked with Deconz ?

kind3r, 2021-01-11:
I would give Z2M a chance. After the initial accomodation you'll find it much easier to use and it supports more devices and they seem to work better. I had the same reluctance at first but now I'm happy I switched.
It does seem to add a tiny bit of lag, but I think that's because of some artificially introduced delays in their Conbee driver implementation (I have submited a PR to properly configure those).

Mimiix, 2021-01-12:
@notovich @kind3r  This doesn't help here at all to mention that... Stick on-topic please :)

@Smanar What's the status?

notovich, 2021-01-12:
@Mimiix did you try to run it on Z2M? meaning disabling deCONZ/ZHA completely and installing Z2M? in my case it fixed the issue, have no idea how, but now over 2 weeks switch working as expected (nocking on wood)

kind3r, 2021-01-12:
@Mimiix The point is it works in Z2M so @Smanar can check how they handle pairing this device and what exceptions they put in for it (if that's the case) and take inspiration from there. Debugging Deconz stuff is especially difficult for those of us running HASS on PIs.

Mimiix, 2021-01-12:
@notovich Z2M is not deCONZ... So your reply is not related to this issue. 

@kind3r I agree on that. It might help Smanar 👍

kind3r, 2021-01-12:
@Mimiix it is not but it works with their adapter (I'm using a ConBee II), so at least this excludes the firmware.

Smanar, 2021-01-12:
@Mimiix, I realy don't have idea, It seem this issue is only present on deconz. The device can work for some time, then have the bug.

And if user disable the bind, the device continue to send request.

I think I will make a hack to disable the second request if there is 2 time the same request in less than 1s. No better idea ATM.

We are working on the time cluster too on anther issue, but I don't see how this cluster can provoke this kind of issue.

kind3r, 2021-01-12:
@Smanar the issue is also present in ZHA.

To sum up (all tested with ConBee II):
- **ZHA** - auto off behaviour right after pairing
- **deConz** - auto off behaviour starts from pairing or after some time (few hours)
- **Z2M** - no auto off behaviour

So deConz is somewhere in between, not good, not bad, 3.6 :D :D :D

Smanar, 2021-01-12:
Ha yep, sorry I mix 2 issue.
The auto off and the double event. I was speaking about the double event one.

To disable the auto off the solution in link don't work at all  ?

> It seems like there’s two versions of this switch coming from AliExpress. The TS0011 appears to turn off by itself after two minutes after being paired for the first time with ZHA. To get it to [...]

parowa92, 2021-01-12:
> Ha yep
> To disable the auto off the solution in link don't work at all ?
> 
> > It seems like there’s two versions of this switch coming from AliExpress. The TS0011 appears to turn off by itself after two minutes after being paired for the first time with ZHA. To get it to function correctly after pairing the first time, press and hold the switch for 10s to enter pairing mode and then press ‘Add Devices Via This Device’ in your Zigbee Coordinator in Home Assistant. After that, it should function correctly. 

This solution is for ZHA, tell me where i can find 'add devices via this device' in deconz. 

Now its 17 days since i moved from deconz to z2mqtt. 
This bug is only on ZHA and Deconz.

kind3r, 2021-01-13:
For me in ZHA re-pairing trick did not work. It did sometimes work in deConz but only for a few hours.

expend20, 2021-01-17:
Hello all! 

I have exactly the same problem but with 3 gang device TS0013. The device turns off by itself after ~2 minutes. The seller says the solution is to rejoin, but it didn't work to me either. Any ideas?

Here are my logs of joining, enabling and auto off parts: https://pastebin.com/JxsjVU40

arislaz, 2021-01-21:
I am not sure if I am helping a lot but the guys over the smartthings community has a similar issue with this switches. Either solution is around choosing the type of the switch when pairing. Maybe someone with more experience can check it out.

degrammont, 2021-01-30:
Hello I'm facing the same issue with my two 3 Gang switches (TZ3000_wyhuocal / TS0013)
I had the issue 7 days ago but it was working well for a week (using deconz docker container). I tried to install another device in my config and the issue came back.
Reseting the device (searching for new lights without removing it from the config) helps for the moment but I don't know how long it will stay like that.

cellfride, 2021-01-31:
Hi, I have the same issue with TS0012 even if i'm using zigbee2mqtt, the switch turns off randomly after some time, i tried the 10 seconds reset and pairing the switch again but it didn't solve the issue

HakunaMatmata, 2021-01-31:
Same issue here, with Deconz and ZHA did not try zigbee2mqtt yet as i do not have a clue how to set it up on home-assistant (no extension catalog with zigbee2mqtt).
I tried repairing it worked once but after a power cut problem came back and impossible to get it working with a new pairing.

Smanar, 2021-01-31:
There will be soon a corrective for button event for switch > https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4265

It correct the "double event" bug, and perhaps the "auto power off" one, if someone can test it ?

HakunaMatmata, 2021-01-31:
> There will be soon a corrective for button event for switch > #4265
> 
> It correct the "double event" bug, and perhaps the "auto power off" one, if someone can test it ?

would be glad to do so, how to download beta version?

Smanar, 2021-01-31:
It s little bit more complicated to test it ^^
The PR is not validated, so you need an unix machine and complile the code (I can explain how to to do)

HakunaMatmata, 2021-01-31:
Ok let me know then.
I use a QNAP with Deconz and Home assistant both installed via docker.
I began switching to ZHA (home assistant embedded version) but same problem there.

Smanar, 2021-01-31:
The procedure is described here > https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

```
sudo apt install deconz-dev
git clone --branch tuya https://github.com/sebres/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

But it s realy hard to do that on docker, and I m not able to do that.
You need a linux machine with a linux OS complete.

k0rn3l, 2021-02-01:
I managed to compile the plugin from  the mentioned PR  #4265  (I've compiled it inside of docker container running the deconz plugin in my hassio instance :P )
I replaced the original plugin, and after restarting deconz and repairing my light switch ( Vendor _TZ3000_9hpxg80k
Model TS0011 ) I am getting normal behaviour for now. But this is fresh, I need to wait for a few hours to confirm the auto-off behaviour is really gone.

[97 more replies]