dresden-elektronik/deconz-rest-plugin #3695: deconz Connections / Mesh

DrDoom666, 2020-11-17:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian deCONZ Add-on
 - Firmware version: latest
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

-->

## Additional context

-->

Smanar, 2020-11-17:
Lol, you are not the first one with this questions ^^.
On my side I have remarked devices always prefer direct connection, and the direct connexion don't look so bad on your side.

The color are just an indication, but I think the connection is better in direct than using "Steckdose Dachzimmer"

It s xiaomi devices ? If yes you can use trick, Xiaomi never (almost) change their parent, so turn off all the router you don't want, put the sensor close the router you want, and remake an inclusion (let it here for 1 hour to be sure), after that you can move it again.

DrDoom666, 2020-11-17:
Hehe. Okay, thank you.
Yes, it is Xiaomi Aqara.

dergreg, 2020-11-18:
> It s xiaomi devices ? If yes you can use trick, Xiaomi never (almost) change their parent, so turn off all the router you don't want, put the sensor close the router you want, and remake an inclusion (let it here for 1 hour to be sure), after that you can move it again.

That sounds like a neat workaround, maybe also for my issue #3529 

I repaired my Xiaomi door contact sensor to the nearest repeater and verified it in the deconz application. Let's see if the connection remains 😊

closed 2020-12-25 by stale[bot]