dresden-elektronik/deconz-rest-plugin #3697: Tuya Radiator Valve TS0601 missing config parameters

Arquiteto, 2020-11-17:
## Describe the bug

After updating deCONZ to 2.06.00 some of my Tuya TRV's lost connection to Conbee. I paired them again, but one of the valves first didn't appear in Phoscon (not sure if it matters) and most importantly it's missing some of the config parameters in REST API.
This is weird as I have 7 of those valve, and the valve I have problem with now was working correctly in previous version (2.05.85). I think the issue actually started with 2.05.88 but I'm not 100% sure, I made the update quite fast after another. All of the valves have same manufacturer and model.

Config from the valve that has all variables:
```
{
    "config": {
        "heatsetpoint": 2050,
        "locked": null,
        "mode": "off",
        "offset": 0,
        "on": true,
        "preset": "manual",
        "reachable": true,
        "schedule": {
            "W124": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T06:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ],
            "W3": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T134:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ]
        },
        "schedule_on": null,
        "setvalve": null,
        "windowopen_set": false
    },
    "ep": 1,
    "etag": "8d4d9e7e35d74fd0ce3f5e7d7670d64e",
    "lastseen": "2020-11-17T19:12Z",
    "manufacturername": "_TZE200_ckud7u2l",
    "modelid": "TS0601",
    "name": "Sypialnia lewy",
    "state": {
        "lastupdated": "none",
        "lowbattery": null,
        "on": false,
        "temperature": 2000,
        "valve": 0
    },
    "type": "ZHAThermostat",
    "uniqueid": "bc:33:ac:ff:fe:57:11:78-01-0201"
}
```

Config from the valve that's missing (locked, mode, preset, setvalve, state/lowbattery, state/valve):
```
{
    "config": {
        "heatsetpoint": 2800,
        "offset": -100,
        "on": true,
        "reachable": true,
        "schedule": {
            "W127": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T06:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ]
        },
        "schedule_on": null
    },
    "ep": 1,
    "etag": "e6115f13edbffa174cb34b1583d4d1d7",
    "lastseen": "2020-11-17T20:57Z",
    "manufacturername": "_TZE200_ckud7u2l",
    "modelid": "TS0601",
    "name": "Łazienka grzejnik",
    "state": {
        "lastupdated": "none",
        "on": true,
        "temperature": 2400
    },
    "type": "ZHAThermostat",
    "uniqueid": "84:2e:14:ff:fe:5a:31:45-01-0201"
}
```

## Steps to reproduce the behavior

1. Delete the device in deCONZ
2. Pair it again

## Expected behavior

Expected to be able to set mode, preset, see valve position, etc.

## Screenshots

n/a

## Environment

 - Host system: QNAP NAS
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.06.00
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? Yes, UPS

## deCONZ Logs

n/a

## Additional context

Arquiteto, 2020-11-17:
@Smanar can you take a look at this?

Smanar, 2020-11-18:
I have checked the code again all seem fine.
I have too compared your device, not the same MAC adress root, but both are in code and they have same manufacture name and modelid.

Have you tried to re-include it but without deleting the previous one ?

Arquiteto, 2020-11-22:
This one was purchased a month later, so it's not from the same shipment as my other 6. But it has the same box, model and looks the same. I tried to pair it again without deleting - no change.
It doesn't show up in lights domain either in  REST calls.

But I found something interesting - I "sacrificed" a thermostat in a guest room which is idle right now anyway. First I paired it again without deleting and the config parameters didn't change. But then I deleted it and added it again and the new sensor doesn't have preset, mode, valve state, etc. as well. So it looks like there is a problem with creating this values in deConz 2.06.00

Arquiteto, 2020-11-22:
Ok, fixed it. The values came back without need for re-pairing after I updated firmware to 26670700
Is this expected?

Smanar, 2020-11-22:
Yes, ATM the deconz core need some modifications to avoid that.
Tuya device need 2 inclusions, the first one to reconize the device, and you can have bad entries at this moment, and a second one that use data collected on the first one.

ShintaroGr, 2020-12-12:
Hello,
Before the update my TRV reported 30°c and now I've the same issue as @Arquiteto.
I've tried updating deconz and reseting the device.

How can I try to get back my TRV ? @Smanar 

Thanks,

Smanar, 2020-12-12:
You have update the firmware too or just deconz ?
Do you have the same JSON than him ? (with model id and manufacture name)
Have you tried to re  include it, without deleting it.

ShintaroGr, 2020-12-12:
So I'm a bit new with deconz.
How can I get the JSON and how to update the FW of one device ?

Yes I tried to repair it but it didn't work

Smanar, 2020-12-12:
Ha yes 
`"manufacturername": "Heiman",`

Easy to solve, but I prefer first found some user have this bug.
For the moment pls avoid to use the GUI.

Wich one deconz version have you ? Do you have the manufacture number of your device ? (in node information panel)

ShintaroGr, 2020-12-12:
On docker : 
Version 2.07.01 / 08/12/2020
Firmware 264A0700

Manufacturer Code 0x1002

Smanar, 2020-12-13:
Ok So 2 things, it have worked for another guy, so it will help you.

- This bug is realy boring and I can't find where it s from, so 1 month I m asking for some log during inclusion. I just need log 10s before the permit join and 1 mn after. You can use directly phoscon/help or command line https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags usefull flag are "info,error,info_L2/error_l2"
- As workaround, you can just read again the basic attribute of the device in deconz, it will solve the bug, but you will never be able to provide log after that.

ShintaroGr, 2020-12-13:
Sorry if the logs aren't clear, I've 80 nodes on my network so a lot of them are polling...

Hope it can help you : 
https://pastebin.com/2fPLjBL5

ShintaroGr, 2020-12-13:
After sending you the log I tired the workaround, but it didn't work.

The new JSON show a different Manufacturer, and on phoscon app their is a "Thermostat" sensor
```
{
    "config": {
        "heatsetpoint": 2000,
        "offset": -100,
        "on": true,
        "reachable": true,
        "schedule": {
            "W124": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T06:00"
                },
                {
                    "heatsetpoint": 16,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ],
            "W3": [
                {
                    "heatsetpoint": 20,
                    "localtime": "T134:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T08:00"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T11:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T12:30"
                },
                {
                    "heatsetpoint": 20,
                    "localtime": "T17:30"
                },
                {
                    "heatsetpoint": 15,
                    "localtime": "T22:00"
                }
            ]
        },
        "schedule_on": null
    },
    "ep": 1,
    "etag": "5e350694770b4cf3653801b3d0c75f7c",
    "lastseen": "2020-12-13T09:54Z",
    "manufacturername": "_TZE200_ckud7u2l",
    "modelid": "TS0601",
    "name": "Thermostat",
    "state": {
        "lastupdated": "none",
        "on": true,
        "temperature": 2150
    },
    "type": "ZHAThermostat",
    "uniqueid": "60:a4:23:ff:fe:a0:fb:cb-01-0201"
}
```

The basic attribute :
[image]

Smanar, 2020-12-13:
Big thx for log, I have something to search now ^^.

Now a little explanation about how work tuya device.
Deconz create the device before having the manufacture name, so the code is searching the deconz database according the mac addres of the device, if it s the first time you include it, it can found the good one, so you have a bad inclusion.
But now you have the correct manufacture name "_TZE200_ckud7u2l" so deconz can now create the good entries.

You can try to-reinclude it, but without deleting it, like that deconz can use previous informations
Or according to the code, just restart deconz will be enought, during the loading, deconz read the database and create sensor according to data stored in it.

ShintaroGr, 2020-12-13:
Thanks after a reboot everything works !

Smanar, 2020-12-13:
Yeah sorry there is ATM 2 problem for tuya device
- Avoid the "heiman" name
- Have the correct name stored in database

One time you have the manufacture name in the DB, deconz will be able to reconize the device.

easybeat, 2020-12-13:
Hi
I followed this becaus I just got 7 of them devices but was away over weekend. Coming home and realized that I have the same problem. I did follow your steps and it worked:

- Update Deconz to 2,06.00
- Update Firmware of ConbeeII to 26670700
- Powered off the devices and added them again without delete
- Read the Basic Cluster after the device was recognized in Phoscon

Now all 7 devices are showing on and start creating connections.

I will get another 5 next week. @Smanar : if you need more logs, please let me know.

Thank you so much for all your help!

Kind regards

Beat

Smanar, 2020-12-13:
Lol, you don't need all of that ^^. And BTW the new stable is the version 2.7.1.

This bug only happen if your device have "heiman" as name.
Just read the basic attribute is enought to correct the manufacture name, and restart.

But yes If you have "heiman" as manufacture name on new device I m interested by log.

I have take a look on ShintaroGr log, it seem the bad name was  from a previous deconz version, but if you have it on new device, it s another story.

easybeat, 2020-12-13:
@Smanar Now after a few hours the devices lost connection again. And yes if I look on manufacture name, it says "Heiman":

[image]

Please tell me again how I get the correct name? If I press the read button to read basic attribute it does update the name in deconz GUI:
[image]

But then I get "Heiman" again out of API...

Thanks

Smanar, 2020-12-14:
Ha ?
So the manufacture name is the good one and after a while he change itlself to "heiman" ?

BTW try with the last deconz stable  2.7.1. instead of 2,06.00, on this one an "heiman" was removed https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3457/files#diff-0413f3c7449569e1c20fdc58374b37e056b213b2113369e3d3b81bbec2a593b3

But there is no reason for the manufacture name change, you have USB disconnection ?

easybeat, 2020-12-14:
Hi @Smanar 

Ok I will try when I'm back home today.

I did update last night to 2.7.0 as I have deconz installed on ubuntu. I couldn't find a 2.7.1 for ubuntu here: https://deconz.dresden-elektronik.de/ubuntu/beta/

Is there another way to do it on ubuntu?

Sorry but I'm not an expert on this.

I will come back with an Update when I'm back home and checked what state the devices are.

Thanks for your help.

Beat

Smanar, 2020-12-14:
Ok, so right, the package is not available in download, probably only possible using package installation.
But BTW I have checked the code, it seem the heiman removal was done in version 2.6.0, so you need to have it too, or I have missed something ...

But you have too the problem on version 2.7.0 ?

easybeat, 2020-12-14:
Well it looks now that the 3 devices on upper floor losing connection:

[image]

which is very weird as there are a lot of devices they could connect to. Could this be related too or a new problem or just a problem with these "cheap" devices?

They 4 devices on the main floor seem to work ok now.

The manufacturer name is now ok!

easybeat, 2020-12-14:
Ahhh and the 3 devices on the upper floor do not react to change on heatsetpoint at all now...

Smanar, 2020-12-14:
Perhaps the connexion quality ? But it don't seem too bad for "BAD OG"

easybeat, 2020-12-14:
I can't understand why they don't route to other devices. In this bath I have another 5 Philips GU10 Bulbs and another devices which have a lot connections to other devices:

[image]

So maybe something is wrong with these "cheap" devices and they have a crap firmware?

And yes I'm using Conbee II USB but really don't want to move the router as this should be the advantage of Zigbee...and it stands pretty central in the house.

easybeat, 2020-12-14:
Ok so I think this must be a connection problem of the device. Did 2 tests now:

1. Unmounted the device from upstairs and took down to main floor. Restarted and connection is back.
2. Got the new batch today, so paired a new device to the system (it does not show all config params!!) but anyway it did accept to set heatsetpoint. Mounted it upstairs and it immediately lost connection!

Do you have an idea what I could do?

easybeat, 2020-12-14:
And you see. It is flashing red when I try to red the basic cluster:

[image]

Smanar, 2020-12-14:
>And yes I'm using Conbee II USB but really don't want to move the router as this should be the advantage of Zigbee...and it stands pretty central in the hous

No, don't need to move it, just use an 30/50 cm USB cordon extension to move the conbee away from perturbations.

And they never use the closest router (Philips GU10 Bulbs) ? What is the parent they prefer ? On xiaomi, you can cheat for the device choose another parent, but I think tuya can change it itself.

easybeat, 2020-12-17:
I will get the extension cable today. But I don't think this will change the problem.

I found this today:

[image]

So it seems to be reachable but I can't read the basic cluster in Deconz

[image]

I've read through #3452 and there was a discussion that the devices lost connection after Deconz restart. Could this be related?

easybeat, 2020-12-17:
OK just did an Update to 2.08.00 and after about 1 hour the TRV upstairs just created an connection to a router!!

Will see if they stay connected!

easybeat, 2020-12-17:
OK things did improve extremly!!
[image]
Now they even have 2! connections. They never did that before at all.
So fingers crossed. Will check again tomorrow morning.

Smanar, 2020-12-17:
Don't worry, you will have soon only 1 connexion ^^. The double is not permanent.
There is a new firmware too ?

easybeat, 2020-12-18:
@Smanar You were right! They not only lost 1 connection, the lost all connections again this morning!

I did an update to 2.08.00 yesterday. Firmware of Conbee II was already on latest.

I did enable error logging and got some entries in there. Does this give any information?

What is this entry: 10:23:50:778 CTRL got nwk update id 2
Comes up every minute!
[Deconz Errorlog 20201218.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5714895/Deconz.Errorlog.20201218.txt)

Mimiix, 2020-12-21:
@easybeat 
19:03:29:815 unhandled APS-DATA.confirm id: 78 status 0xE1

Seems like interferrence.

Guus-H, 2020-12-21:
> Lol, you don't need all of that ^^. And BTW the new stable is the version 2.7.1.
>
> This bug only happen if your device have "heiman" as name.
> Just read the basic attribute is enought to correct the [...]

Do you still need logs? I have the same symptoms (Vendor showing Heiman, unable to control device, no reporting of valve positions)
Do I understand you'd need them a few seconds before and 10 seconds after (re) pairing the device?

easybeat, 2021-02-08:
@Smanar : Do you still need some information to solve this bug?

Mimiix, 2021-02-08:
@easybeat As long as you have 0xe1, we can't confirm wany bug.

easybeat, 2021-02-08:
> @easybeat As long as you have 0xe1, we can't confirm wany bug.

Ok...then my next question is how to resolve the interferrence problem? I'v added a 3m USB Cable to get away from box but problem still exists. See Log.
[Deconz Errorlog 20210208.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5944563/Deconz.Errorlog.20210208.txt)

Smanar, 2021-02-08:
The device "_TZE200_ckud7u2" is working from other tests.
For the problem with heiman, long time I have see it again, you are using wich one decone version ?

But your log not seem so bad if you have only one error by day. You are still not able to read the basic attribute on deconz ?

Mimiix, 2021-02-08:
> Ok...then my next question is how to resolve the interferrence problem? I'v added a 3m USB Cable to get away from box but problem still exists. See Log.
> [Deconz Errorlog [...]

Is there a SSD involved?

easybeat, 2021-02-08:
> The device "_TZE200_ckud7u2" is working from other tests.
> For the problem with heiman, long time I have see it again, you are using wich one deconz version ?

I'm using deconz 2.08.00

> 
> But your log not seem so bad if you have only one error by day. You are still not able to read the basic attribute on deconz ? Remember you can have to wake up the device.
> 
> The problem can come from the device too.

Yes, I can read basic attribute now but still losing connection to each device after a few days if they are far away from hub and also with Ikea range extender it does not work.

[image]

This error was only from recording for about 1 hour.

easybeat, 2021-02-08:
> Is there a SSD involved?

SSD built in the NUC not connected by USB

Smanar, 2021-02-08:
>Yes, I can read basic attribute now but still losing connection to each device after a few days if they are far away from hub and also with Ikea range extender it does not work.

Not sure if it will solve your problem, but on the next deconz version it will be lot of improvement for tuya, one of them is the missing defaut response.

It will not solve your perturbations problems but can improve tuya working mode. For exemple without it some devices skip randomly some requests.

easybeat, 2021-02-08:
> Not sure if it will solve your problem, but on the next deconz version it will be lot of improvement for tuya, one of them is the missing defaut response.
>
> It will not solve your perturbations [...]

OK let's see if this will have an impact.

But I still don't understand why these devices do not connect to lights and use them as routers...but I opened another user question for this....thanks for your help.

Smanar, 2021-02-08:
>But I still don't understand why these devices do not connect to lights and use them as routers...but I opened another user question for this....thanks for your help.

Good question, no one can use router ? The connection don't look so bad on your capture (green line).

No one have tuya TRV/thermostat device that use routers ?

SebLz, 2021-02-12:
> > But I still don't understand why these devices do not connect to lights and use them as routers...but I opened another user question for this....thanks for your help.
> 
> Good question, no one can use router ? The connection don't look so bad on your capture (green line).
> 
> No one else have tuya TRV/thermostat device that use routers ?

I have 3 Moes HY368 TRV and here they do connect to routers (in my case Ikea smart plugs, ikea usb range extender and Blitzwolf shp13):
[image]

easybeat, 2021-02-16:
@Smanar Since I've updated to 2.9.3 they "changed" behaviour and do now connect to other routers! And also none lost connection...no idea if this is related to changes in 2.9.3.

Smanar, 2021-02-16:
No, not from my memory, perhaps the network need some time to be optimised ..

closed 2021-03-19 by github-actions[bot]