dresden-elektronik/deconz-rest-plugin #3699: heating plan for Danfoss Ally fails

Einstein2150, 2020-11-18:
## Describe the bug
Danfoss Ally:
Setting a heating Plan is not possible. It is rejected by the API.

## Steps to reproduce the behavior
Setting/unsetting a heating plan but the setting/unsetting fails

## Expected behavior
Heating plan is set / unset

## Screenshots
[image]
[image]

## Environment
RaspBee I
2.05.88 / 15.10.2020
Firmware
26390500
Raspberry Pi 4 - Raspbian
 

## deCONZ Logs
none

## Additional context
Set:
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: 00 begin
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: 13 schudule_on true
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: F4 temp: 21.5°C, 15.0°C, 21.5°C
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: FC time: 2020-11-18T07:03
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: Mon 05:00 17:00
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: Tue 05:00 17:00
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: Wed 05:00 17:00
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: Thu 05:00 17:00
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: Fri 05:00 17:00
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: Sat 05:00 17:00
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: Sun 05:00 17:00
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: Free 05:00 17:00
[18.11.2020, 07:03:47] [Hue] Temperaturregler Büro: 06 end
[18.11.2020, 07:03:47] [Hue] Phoscon-GW: gateway request 18683: put /sensors/9/config {"schedule_on":true}
[18.11.2020, 07:03:47] [Hue] Phoscon-GW: gateway request 18683: error 6 parameter, schedule_on, not available

Unset:
[18.11.2020, 07:07:14] [Hue] Temperaturregler Büro: 00 begin
[18.11.2020, 07:07:14] [Hue] Temperaturregler Büro: 13 schudule_on false
[18.11.2020, 07:07:14] [Hue] Temperaturregler Büro: F4 temp: 64.0°C, 15.0°C, 15.0°C
[18.11.2020, 07:07:14] [Hue] Temperaturregler Büro: 06 end
[18.11.2020, 07:07:14] [Hue] Phoscon-GW: gateway request 18768: put /sensors/9/config {"schedule_on":false}
[18.11.2020, 07:07:14] [Hue] Phoscon-GW: gateway request 18768: error 6 parameter, schedule_on, not available

API-Output for the Danfoss Ally:
{
"config": {
"battery": 76,
"displayflipped": false,
"heatsetpoint": 2150,
"offset": -50,
"on": true,
"reachable": true
},
"ep": 1,
"etag": "0cc16513f8a1ade11415dc82595f9639",
"lastseen": "2020-11-18T06:02Z",
"manufacturername": "Danfoss",
"modelid": "eTRV0100",
"name": "Temperaturregler Büro",
"state": {
"lastupdated": "2020-11-18T06:06:58.667",
"mountingmodeactive": false,
"on": true,
"temperature": 1885,
"valve": 58,
"windowopen": "Closed"
},
"swversion": "20200916",
"type": "ZHAThermostat",
"uniqueid": "cc:cc:cc:ff:fe:7c:a8:0b-01-0201"
}

Einstein2150, 2020-11-18:
Hello @ebaauw 
I saw you helped here before [https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2393](url)
Any solution from your expert knowlage for my Problem? 😎

SwoopX, 2020-11-18:
The thermostat simply does not support on-device schedules.

closed 2020-11-18 by SwoopX

SwoopX, 2021-01-07:
You need to upgrade your devices to firmware 1.08 for this to be supported as well as deconz version 2.10.00 (not yet released).

https://files.danfoss.com/download/Heating/Ally/Ally%20eTRV%201.08%20OTA%20file.zip

Einstein2150, 2021-01-07:
I'm looking forward to deCONZ 2.10 to test it 🎉