dresden-elektronik/deconz-rest-plugin #3702: Add color attribute to Alert

D34DC3N73R, 2020-11-19:
## Feature request type
Add color to Alert field for use with Home Assistant [light.turn_on service](https://www.home-assistant.io/integrations/light/#service-lightturn_on) `flash`

## Description
I'd like to be able to flash a hue bulb different colors with home assistant. Currently, HA supports adding color with `light.turn_on` `flash:short/long` but the color attribute doesn't appear to be passed on through the deconz api

## Considered alternatives
I've considered a script in home assistant that repeatedly turns the bulb off, but I'd prefer to use `flash`

## Additional context
When I call the service the hue rgb bulb flashes white instead of the specified color
```
  - service: light.turn_on
    data:
      transition: 0
      brightness: 255
      color_name: orangered
      flash: long
    entity_id: light.hue_lamp
```

ebaauw, 2020-11-19:
Zigbee lights don't support colour flashing; the _Identify_ command causes the light to flash in white.  The API exposes this through PUTting a body of `{"alert": "lselect"}` or `{"alert":"select"}` to the light `state`.

The closest approximation is the _Trigger Effect_ command (in the _Identify_ cluster), but not all lights support it.  This is exposed by the API and Hue lights support:
- `{"alert": "breathe"}`: blink the light for ~15 seconds in the current colour.  You might combine this with setting the colour in the same body, but you'll have to save the previous colour and revert manually.  Send `{"alert": "finish"}` to stop the blinking prematurely;
- `{"alert": "okay"}`: blink the light green once, and then revert to the previous colour;
- `{"alert": "channelchange"}`: turn the light orange for ~15s, and then revert to the previous colour.

D34DC3N73R, 2020-11-19:
That's interesting because hue bridge users can use `light.turn_on` with `flash` and `color` in home assistant.

ebaauw, 2020-11-19:
What API commands does that send to the Hue bridge?

D34DC3N73R, 2020-11-19:
I'm not exactly sure. Is it the API command somewhere in here? https://github.com/home-assistant/core/blob/61d9067a49bdb1df8c5b7024f563b7ec480d65fe/homeassistant/components/hue/light.py

ebaauw, 2020-11-19:
Looks like it's sending an `{"alert": "select"}` or `{"alert": "lselect"}`, which the Hue bridge translates to the same _Trigger Effect_ command as deCONZ generates for `{"alert": "breathe"}`.

D34DC3N73R, 2020-11-19:
I'll put in a feature request there. Thanks for the direction.

D34DC3N73R, 2020-11-20:
Related discussion for anyone who finds this in the future. https://community.home-assistant.io/t/update-deconz-integration-to-support-flash-with-color/246904

closed 2020-11-20 by D34DC3N73R