dresden-elektronik/deconz-rest-plugin #3704: deCONZ Update to 2.05.88 --> homebridge-hue very slow response

DrDoom666, 2020-11-19:
## Describe the bug

Not really a bug, but i did an update on my Raspberry with apt update, apt upgrade and now i'm on deCONZ 2.05.88.

I've realized that my devices are now shown with a battery symbol in deCONZ, what is very nice, but the response of the homebridge-hue plugin which controls my light when I toggle my wall switch is very slow.
It takes 3 or 4 seconds, till my lights turn on or off. That's way more slow than in the previous version.
Perhaps you can fix that.

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.05.88
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No.

## deCONZ Logs

## Additional context

Mimiix, 2020-11-20:
The only time this happens if when the SD card is slow or full or broken. I suggest checking there.

DrDoom666, 2020-11-23:
Okay. Everytime I restart my Raspberry the reaction time is very slow.
After some time the response time is in a normal range about one second.
So I close this. Thank you.

closed 2020-11-23 by DrDoom666