dresden-elektronik/deconz-rest-plugin #3712: USB Conbee II is restart every 5s (driver disconnect/connect).

lukinno1, 2020-11-20:
## Describe the question or issue you are having
Describe the bug

I use deCONZ with the Home Assistant integration: https://www.home-assistant.io/integrations/deconz/

I think upgrading to last firmware 26670700 from addon HA. Before upgrades was working.
Now my USB Conbee II is still disconnect/connect every 5s and not connected.
I checked devices in windows 10 but still is restart every 5s.
I think problem is in firmware.

## Screenshots

## Environment
Use Home Assistant
Install a version after 2.05.85

Host system: Windows 10
Running method: vmware virtual hassio with home assistant
Firmware version: (26660700)
deCONZ addon version: 2.05.85
Device: ConBee II
Do you use an USB extension cable: no
Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context
Has been working with Home Assistant up to this point. I

What can i solve this problem?

Smanar, 2020-11-20:
Hello, pls try that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3262

lukinno1, 2020-11-21:
Hello i downgrade firmware and now is working. Thanks

closed 2020-11-21 by lukinno1

yves19, 2020-11-21:
You should try flashing the Conbee2 again with the latest firmware. 
It seems like the flash operation was not completely achieved . 
It's a common bug linked to bad flash of the firmware