dresden-elektronik/deconz-rest-plugin #3713: Sonoff SNZB-03 Motion Sensor can#t connect to phoscon

bruesche, 2020-11-20:
## Describe the question or issue you are having
Hi there,

today my Sonoff SNZB-03 Motion Sensor arrived. I tried to connect it to phoscon, but it says "connection failed".
In the forum the motion sensor is marked as supported in version 79beta.

What can I do to use the motion sensor with phoscon.

Please gibe me some advice.

Thx

## Screenshots

## Environment

 My phoscon version is 2.6.00beta from 15.11.2020. Firmware from Conbee II Stick is 26660700.

## deCONZ Logs

## Additional context

SwoopX, 2020-11-20:
Can you please provide a screenshot of the basic cluster while the attributes have been read? For that to succeed, make sure the device is awake (e.g. triggering motion).

Eventually, your device has a yet unknown modelID. We've seen that Sonoff kinda random with that.

bruesche, 2020-11-20:
Hi,

Could get the data from deconz-gui over VNC. 

0x0004     Manufacturer Name     eWeLink
0x0005.    Model Identifier            66666
0x0006     Data Code                    20191107

I can get the infos in the basic node. Phoscon says connected. No sensor is available. At next try the 3 Point in decent-gui are empty.

Thx for help.

SwoopX, 2020-11-20:
There we have it, different and completely nuts modelID 🤦‍♂️

bruesche, 2020-11-20:
Can it be fixed? My other option is: Motion Sensor will be sent back...

bruesche, 2020-11-20:
_

SwoopX, 2020-11-21:
Yes, we need to whitelist it.

eberna, 2020-11-21:
Hi, i confirm I'm having the same issue with a brand new SNZB-03.
[image]

bruesche, 2020-11-22:
What have I to do to whitelist it?

damajor, 2020-11-30:
Please re-open, I just built the lastest beta and SNZB-03 are still not well identified.
[image]
[image]
[image]
[image]
[image]
[image]

SwoopX, 2020-11-30:
Everything's fine...

damajor, 2020-11-30:
On my second sensor I got a different "Basic Cluster" values :
[image]

damajor, 2020-11-30:
Well, they do not appear in Phoscon, but I seem to be able to reach them via node-red. I will do additional tests.

SwoopX, 2020-11-30:
> On my second sensor I got a different "Basic Cluster" values

Because the values have now been read in contrast to the former screenshots, which only show defaults.

Phoscon is just another REST API client and hence, not deconz. If you want to have it added to Phoscon, you have to raise an issue in the Phoscon beta repository.

damajor, 2020-12-01:
Well I confirm this is working. thanks for the patches.