dresden-elektronik/deconz-rest-plugin #3714: ConBee II: USB disconnecting/reconnecting in a continuous loop

indomus, 2020-11-20:
## Describe the bug
While searching for a solution [for this other issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3645), I discovered some that could be correlated.
This is my log:

> Nov 20 12:41:19 raspberry kernel: [  150.538077] usb 1-1.3: USB disconnect, device number 12
> Nov 20 12:41:20 raspberry kernel: [  150.888317] usb 1-1.3: new full-speed USB device number 13 using dwc_otg
> Nov 20 12:41:20 raspberry kernel: [  151.022733] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
> Nov 20 12:41:20 raspberry kernel: [  151.022757] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> Nov 20 12:41:20 raspberry kernel: [  151.022771] usb 1-1.3: Product: ConBee II
> Nov 20 12:41:20 raspberry kernel: [  151.022785] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
> Nov 20 12:41:20 raspberry kernel: [  151.022799] usb 1-1.3: SerialNumber: DE2119893
> Nov 20 12:41:20 raspberry kernel: [  151.025332] cdc_acm 1-1.3:1.0: ttyACM1: USB ACM device
> Nov 20 12:41:23 raspberry kernel: [  154.380158] usb 1-1.3: USB disconnect, device number 13
> Nov 20 12:41:24 raspberry kernel: [  154.728193] usb 1-1.3: new full-speed USB device number 14 using dwc_otg
> Nov 20 12:41:24 raspberry kernel: [  154.862601] usb 1-1.3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
> Nov 20 12:41:24 raspberry kernel: [  154.862618] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> Nov 20 12:41:24 raspberry kernel: [  154.862630] usb 1-1.3: Product: ConBee II
> Nov 20 12:41:24 raspberry kernel: [  154.862642] usb 1-1.3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
> Nov 20 12:41:24 raspberry kernel: [  154.862654] usb 1-1.3: SerialNumber: DE2119893
> Nov 20 12:41:24 raspberry kernel: [  154.865111] cdc_acm 1-1.3:1.0: ttyACM1: USB ACM device

and so on, and on.

## Steps to reproduce the behavior
-

## Expected behavior
-

## Screenshots
-

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.05.88
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

lukinno1, 2020-11-20:
hello

I have the same problem.

Smanar, 2020-11-20:
Hello, pld try that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3000#issuecomment-730714241

fudu, 2020-11-22:
Same issue here, 
DeConz 2.05.84
Conbee II firmware latest - 26660700
0.117.6
HassOs on Raspberry pi 4B

I had same issue on a old version installed on a VM in Qnap Virtualization, so I went on Raspberry, thinking of virtualization problem.
Now in these day I get this issue again with different hardware ad OS version ....

fudu, 2020-11-22:
> Same issue here,
> DeConz 2.05.84
> Conbee II firmware latest - 26660700
> 0.117.6
> HassOs on Raspberry pi 4B
>
> I had same issue on a old version installed on a VM in Qnap Virtualization, so I went on [...]

Update:
After to HassOS 4.17 and reboot, issue seems no more present .... 
???

manup, 2020-11-23:
Hi, I've edited the OP to adhere to the template.

Usually there are following reasons for looping and deCONZ not able to connect to ConBee problems:

#### No firmware is installed, or a firmware update wasn't completed.
Note: When deCONZ can connect and the blue node is shown, a firmware *is* installed.

One way to check the installed firmware version is to use GCFFlasher while deCONZ is not running.

```
$ GCFFlasher_internal -l

GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE1995634  | ConBee II 

$ GCFFlasher.bin -r -sn DE1995634

GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device  (ConBee II)
deCONZ firmware version 26670700
R21B18 Bootloader
Vers: 2.07
build: Jun 17 2019
```

Here the output `deCONZ firmware version 26670700` verifies the installed firmware.

#### No USB extension cable

In some setups this can have dramatic impacts, beside poor Zigbee signal one of them is boot looping.
To rule that out as a reason and in general we **highly** recommend a good USB extension cable.

#### SSD or other USB 3.0 peripherals

Same suggestion here: Use an USB extension cable for ConBee II.

#### Docker and USB re-enumeration

In case the firmware reboots while the device like /dev/ttyACM0 is still registered under Linux,
a new device name like /dev/ttyACM1 will be assigned. This new name is currently not visible inside containers which are often started to use a specific device.

Altough we have stable device paths as symlinks like:

    /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE19

Which will point to the new device name, this isn't forwarded inside Docker container. This is tricky problem, which will hopefully be fixed via Docker `cgroups` as mentioned in https://github.com/marthoc/docker-deconz/issues/298

#### Linux Modem Manager

The Modem Manager attempts also to connect and probe newly attached USB devices. Which can lead to problems in some setups.

To rule that out as problem, either stop the service:

`sudo systemctl stop ModemManager.`

Or exclude ConBee II from being probed via UDEV rules on the host system:

(requires a os restart or reload of UDEV rules)

```
$ cat /etc/udev/rules.d/69-conbee.rules 
# ConBee II disbale modem manager and mtp probing
ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{ID_MM_DEVICE_IGNORE}="1"
ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{MTP_NO_PROBE}="1"
```

fudu, 2020-11-24:
> Hi, I've edited the OP to adhere to the template.
> 
> Usually there are following reasons for looping and deCONZ not able to connect to ConBee problems:
> 
> #### No firmware is installed, or a firmware update wasn't completed.
> Note: When deCONZ can connect and the blue node is shown, a firmware _is_ installed.
> 
> One way to check the installed firmware version is to use GCFFlasher while deCONZ is not running.
> 
> ```
> $ GCFFlasher_internal -l
> 
> GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
> Path             | Vendor | Product | Serial     | Type
> -----------------+--------+---------+------------+-------
> /dev/ttyACM0     | 0x1CF1 | 0x0030  | DE1995634  | ConBee II 
> 
> $ GCFFlasher.bin -r -sn DE1995634
> 
> GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
> Reboot device  (ConBee II)
> deCONZ firmware version 26670700
> R21B18 Bootloader
> Vers: 2.07
> build: Jun 17 2019
> ```
> 
> Here the output `deCONZ firmware version 26670700` verifies the installed firmware.
> 
> #### No USB extension cable
> In some setups this can have dramatic impacts, beside poor Zigbee signal one of them is boot looping.
> To rule that out as a reason and in general we **highly** recommend a good USB extension cable.
> 
> #### SSD or other USB 3.0 peripherals
> Same suggestion here: Use an USB extension cable for ConBee II.
> 
> #### Docker and USB re-enumeration
> In case the firmware reboots while the device like /dev/ttyACM0 is still registered under Linux,
> a new device name like /dev/ttyACM1 will be assigned. This new name is currently not visible inside containers which are often started to use a specific device.
> 
> Altough we have stable device paths as symlinks like:
> 
> ```
> /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE19
> ```
> 
> Which will point to the new device name, this isn't forwarded inside Docker container. This is tricky problem, which will hopefully be fixed via Docker `cgroups` as mentioned in [marthoc/docker-deconz#298](https://github.com/marthoc/docker-deconz/issues/298)
> 
> #### Linux Modem Manager
> The Modem Manager attempts also to connect and probe newly attached USB devices. Which can lead to problems in some setups.
> 
> To rule that out as problem, either stop the service:
> 
> `sudo systemctl stop ModemManager.`
> 
> Or exclude ConBee II from being probed via UDEV rules on the host system:
> 
> (requires a os restart or reload of UDEV rules)
> 
> ```
> $ cat /etc/udev/rules.d/69-conbee.rules 
> # ConBee II disbale modem manager and mtp probing
> ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{ID_MM_DEVICE_IGNORE}="1"
> ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ENV{MTP_NO_PROBE}="1"
> ```

- I have latest firmware installed
- I have extension cable
- No SSD or other USB devices
- I don't see new /dev/ttyACM1 
- I'm on HassOS and Linux Modem Manager is not loaded.

After a day going fine (only 1 disconnection per hour or so ...), on its own this morning again every few seconds USB disconnect/connect.
Updated HA from 0.117.6 to 0.118.3, and at the moment issue is gone ......

Smanar, 2020-11-24:
>I have latest firmware installed

Even with that, you can have the last one, but badly installed, if you have a disconnection every second, even with the last one, you can try to update it again.

fudu, 2020-11-24:
> > I have latest firmware installed
> 
> Even with that, you can have the last one, but badly installed, if you have a disconnection every second, even with the last one, you can try to update it again (but if it work during 1 hour, it s another story)

Yes, it's strange, some time it starts to make continuos disconnection/connection and after a restart/reboot it disconnects only few times.

For updating firmware: I got error through Phoscon Interface in HassOS, so I tried plugging ConbeeII on a Windows10 pc.
Tried several times, I got success messages, but key was not working-not recognized anymore. Then I updated it through an Ubuntu VM installed on same Win10 pc

Smanar, 2020-11-24:
You are using the command with "-t 60" ?

fudu, 2020-11-24:
> You are using the command with "-t 60" ?

yes

closed 2020-12-25 by stale[bot]