dresden-elektronik/deconz-rest-plugin #3719: Evaluation of rules broken in v2.05.88

ebaauw, 2020-11-20:
## Describe the bug

The evaluation of rules in v2.05.88 is broken.  It still works in v2.04.84.  Consequently my toggle buttons no longer work.

## Steps to reproduce the behavior

Create a CLIP sensor and a pair of rules to toggle a light state when pressing a button:
```
$ ph get /rules/172
{
  "actions": [
    {
      "address": "/sensors/281/state",
      "body": {
        "status": 0
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/285/state/buttonevent",
      "operator": "eq",
      "value": "1002"
    },
    {
      "address": "/sensors/285/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/sensors/281/state/status",
      "operator": "gt",
      "value": "0"
    }
  ],
  "created": "2019-12-08T18:46:29",
  "etag": "774523b5d284c23c40cadb85ec36d772",
  "lasttriggered": "2020-11-20T19:12:43",
  "name": "Balcony Switch On/Off Press (1/2)",
  "owner": "3635D7B19D",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 1
}
$ ph get /rules/173
{
  "actions": [
    {
      "address": "/sensors/281/state",
      "body": {
        "status": 1
      },
      "method": "PUT"
    }
  ],
  "conditions": [
    {
      "address": "/sensors/285/state/buttonevent",
      "operator": "eq",
      "value": "1002"
    },
    {
      "address": "/sensors/285/state/lastupdated",
      "operator": "dx"
    },
    {
      "address": "/sensors/281/state/status",
      "operator": "lt",
      "value": "1"
    }
  ],
  "created": "2019-12-08T18:46:29",
  "etag": "3561bd37e71b6d57d31099ac5a4c9200",
  "lasttriggered": "2020-11-20T19:12:41",
  "name": "Balcony Switch On/Off Press (2/2)",
  "owner": "3635D7B19D",
  "periodic": 0,
  "status": "enabled",
  "timestriggered": 1
}
```
In v2.05.88 both rules trigger when pressing the button; in v2.05.84 only one of them.

## Expected behavior

Only one of the rules should tigger, as in v2.05.84 or on the Hue bridge.

## Screenshots

n/a

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: (26390500)
 - deCONZ version: (2.05.88)
 - Device: RaspBee I
 - Do you use an USB extension cable: n/a
 - Is there any other USB or serial devices connected to the host system? n/a

## deCONZ Logs

## Additional context

In v2.05.88:
```
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:834 trigger rule 173 - Balcony Switch On/Off Press (2/2)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:842 trigger rule 172 - Balcony Switch On/Off Press (1/2)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:856 trigger rule 170 - Balcony Status <1
Nov 20 19:57:37 pi2 deCONZ[3980]: 19:57:37:147 trigger rule 173 - Balcony Switch On/Off Press (2/2)
Nov 20 19:57:37 pi2 deCONZ[3980]: 19:57:37:154 trigger rule 172 - Balcony Switch On/Off Press (1/2)
Nov 20 19:57:37 pi2 deCONZ[3980]: 19:57:37:183 trigger rule 170 - Balcony Status <1
```

In v2.04.84:
```
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:329 trigger rule 173 - Balcony Switch On/Off Press (2/2)
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:332 trigger rule 171 - Balcony Status >0
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:337 trigger rule 179 - Balcony On, Dark 2, Day
Nov 20 20:12:43 pi2 deCONZ[25204]: 20:12:43:384 trigger rule 172 - Balcony Switch On/Off Press (1/2)
Nov 20 20:12:43 pi2 deCONZ[25204]: 20:12:43:387 trigger rule 170 - Balcony Status <1
Nov 20 20:12:43 pi2 deCONZ[25204]: 20:12:43:393 trigger rule 176 - Balcony Off
```

manup, 2020-11-21:
I've re-created the setup in a clean environment with just a ConBee and no devices.
Using the same rules and ids as above but replaced the switch with a CLIPSwitch.

1. Trigger
```
PUT /sensors/285/state
{ "buttonevent": 1002 }
```

```
00:52:03:736 rule event /sensors/285/state/lastupdated: 0 -> 0
00:52:03:736 trigger rule 173 - Balcony Switch On/Off Press (2/2)
```

2. Trigger
```
PUT /sensors/285/state
{ "buttonevent": 1002 }
```
```
00:52:24:059 rule event /sensors/285/state/lastupdated: 0 -> 0
00:52:24:059 trigger rule 172 - Balcony Switch On/Off Press (1/2)
```

3. Trigger
```
PUT /sensors/285/state
{ "buttonevent": 1002 }
```
```
00:52:52:697 rule event /sensors/285/state/lastupdated: 0 -> 0
00:52:52:697 trigger rule 173 - Balcony Switch On/Off Press (2/2)
```

I think this looks ok? The rules are only triggered once in my setup.

manup, 2020-11-22:
What kind of switch do you use? I can also imagine that the rules engine works fine but the `buttonevent` is emitted twice.

ebaauw, 2020-11-22:
The switch is a `lumi.sensor_switch.aq2`.  It only sends a single message, but indeed, there's two rule events for `lastupdated`.  Missed that before; good thing Raspbian saves the `syslog` for a week.

I only see a single rule event under v2.05.84.

I have two listeners connected to the web socket, maybe that's related?

Below the detailed log for .88:
```
Nov 20 19:57:35 pi2 deCONZ[3980]: 19:57:35:796 APS-DATA.indication srcAddr: 0x3cae, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 239, rssi: -61
Nov 20 19:57:35 pi2 deCONZ[3980]: 19:57:35:798 #011asdu: 18180a0000100000001001
Nov 20 19:57:35 pi2 deCONZ[3980]: 19:57:35:801 Node data 0x00158d0002008dab profileId: 0x0104, clusterId: 0x0006
Nov 20 19:57:35 pi2 deCONZ[3980]: 19:57:35:803 0x00158D0002008DAB: update ZCL value 0x01/0x0006/0x0000 after 0 s
Nov 20 19:57:35 pi2 dc_eventlog[835]: /sensors/285/state: {"buttonevent":1002,"lastupdated":"2020-11-20T18:57:35.813"}
Nov 20 19:57:35 pi2 dc_eventlog[835]: /sensors/281/state: {"lastupdated":"2020-11-20T18:57:35.843","status":0}
Nov 20 19:57:35 pi2 dc_eventlog[835]: /sensors/280/state: {"lastupdated":"2020-11-20T18:57:35.856"}
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:810 [INFO] - Button 1002 Normal press lumi.sensor_switch.aq2
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:813 ZCL attribute report 0x00158D0002008DAB for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:815 #011payload: 0000100000001001
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:818 don't create binding for attribute reporting of sensor Balcony Switch
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:824 Websocket 127.0.0.1:35298 send message: {"e":"changed","id":"285","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-11-20T18:57:35.813"},"t":"event","uniqueid":"00:15:8d:00:02:00:8d:ab-01-0006"} (ret = 166)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:827 Websocket 127.0.0.1:41256 send message: {"e":"changed","id":"285","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2020-11-20T18:57:35.813"},"t":"event","uniqueid":"00:15:8d:00:02:00:8d:ab-01-0006"} (ret = 166)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:831 discard sensor state push for 285: state/lastupdated (already pushed)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:833 rule event /sensors/285/state/lastupdated: 0 -> 0
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:834 trigger rule 173 - Balcony Switch On/Off Press (2/2)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:837 discard sensor state push for 285: state/buttonevent (already pushed)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:839 discard sensor state push for 285: state/lastupdated (already pushed)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:840 rule event /sensors/285/state/lastupdated: 0 -> 0
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:842 trigger rule 172 - Balcony Switch On/Off Press (1/2)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:844 Websocket 127.0.0.1:35298 send message: {"e":"changed","id":"281","r":"sensors","state":{"lastupdated":"2020-11-20T18:57:35.843","status":0},"t":"event","uniqueid":"/sensors/281"} (ret = 139)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:846 Websocket 127.0.0.1:41256 send message: {"e":"changed","id":"281","r":"sensors","state":{"lastupdated":"2020-11-20T18:57:35.843","status":0},"t":"event","uniqueid":"/sensors/281"} (ret = 139)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:848 rule event /sensors/281/state/status: 0 -> 1
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:851 discard sensor state push for 281: state/lastupdated (already pushed)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:853 discard sensor state push for 281: state/status (already pushed)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:855 rule event /sensors/281/state/status: 1 -> 0
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:856 trigger rule 170 - Balcony Status <1
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:858 discard sensor state push for 281: state/lastupdated (already pushed)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:861 Websocket 127.0.0.1:35298 send message: {"e":"changed","id":"280","r":"sensors","state":{"lastupdated":"2020-11-20T18:57:35.856"},"t":"event","uniqueid":"/sensors/280"} (ret = 128)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:862 Websocket 127.0.0.1:41256 send message: {"e":"changed","id":"280","r":"sensors","state":{"lastupdated":"2020-11-20T18:57:35.856"},"t":"event","uniqueid":"/sensors/280"} (ret = 128)
Nov 20 19:57:36 pi2 deCONZ[3980]: 19:57:35:864 rule event /sensors/280/state/lastupdated: 0 -> 0
```
And for .84:
```
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:322 APS-DATA.indication srcAddr: 0x3cae, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -47
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:322 #011asdu: 181a0a0000100000001001
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:322 Node data 0x00158d0002008dab profileId: 0x0104, clusterId: 0x0006
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:322 0x00158D0002008DAB: added ZCL value 0x01/0x0006/0x0000
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:323 [INFO] - Button 1002 Normal press
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:323 ZCL attribute report 0x00158D0002008DAB for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:323 #011payload: 0000100000001001
[... cut ...]
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:336 Websocket 127.0.0.1:40354 send message: {"e":"changed","id":"280","r":"sensors","state":{"flag":true,"lastupdated":"2020-11-20T19:12:41.332"},"t":"event","uniqueid":"/sensors/280"} (ret = 140)
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:336 Websocket 127.0.0.1:40400 send message: {"e":"changed","id":"280","r":"sensors","state":{"flag":true,"lastupdated":"2020-11-20T19:12:41.332"},"t":"event","uniqueid":"/sensors/280"} (ret = 140)
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:337 rule event /sensors/280/state/flag: 0 -> 1
Nov 20 20:12:41 pi2 deCONZ[25204]: 20:12:41:337 trigger rule 179 - Balcony On, Dark 2, Day
```

manup, 2020-11-22:
Thanks that helps. I think these duplicated button events where mentioned in another issue too. Not sure yet if it is specific for Xiaomi, perhaps this particular switch or a general issue. I'll make some tests.

The database seems to be not the problem here, so I think this doesn't need to be testet for now.

manup, 2020-11-22:
The problems seems that we have two places now where the event is emitted.
The new `button_maps.json` will lead to emit the event in `checkSensorButtonEvent()` as well as the older code in `updateSensorNode()`.

To verify this, as a test can you please remove following from the `button_maps.json` temporarily: 

> /usr/share/deCONZ/devices/button_maps.json

```
        "xiaomiSwitchAq2Map": {
            "modelids": ["lumi.sensor_switch.aq2"],
            "map": [
                [1, "0x01", "ONOFF", "0x0a", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press"],
                [1, "0x01", "ONOFF", "0x0a", "2", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double press"],
                [1, "0x01", "ONOFF", "0x0a", "3", "S_BUTTON_1", "S_BUTTON_ACTION_TREBLE_PRESS", "Triple press"],
                [1, "0x01", "ONOFF", "0x0a", "4", "S_BUTTON_1", "S_BUTTON_ACTION_QUADRUPLE_PRESS", "Quad press"]
            ]
        }
```

After a restart of deCONZ only one buttonevent should be emitted.

If that works I'll extend `updateSensorNode()` to not emit a buttonevent for this modelid (also need to check the other models).

Smanar, 2020-11-22:
> Thanks that helps. I think these duplicated button events where mentioned in another issue too. Not sure yet if it is specific for Xiaomi, perhaps this particular switch or a general issue. I'll make some tests.

ATM I have 2 issues with this problem, but don't think they are related, adn they don't use rules.
- a tuya device that always send double event, but only 1 user, working for other
- a legrand device but it s not always the case, he have the problem every 1/2 days.

ebaauw, 2020-11-22:
That only makes matters worse.  Still two rule events for `lastupdated`, but now the `buttonevent` is 1000 instead of 1002.

```
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:550 APS-DATA.indication srcAddr: 0x3cae, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -49
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:553 #011asdu: 185f0a0000100000001001
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:556 Node data 0x00158d0002008dab profileId: 0x0104, clusterId: 0x0006
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:559 0x00158D0002008DAB: update ZCL value 0x01/0x0006/0x0000 after 0 s
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:604 [INFO] - Button 1000 Normal press lumi.sensor_switch.aq2
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:606 ZCL attribute report 0x00158D0002008DAB for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:607 #011payload: 0000100000001001
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:651 APS-DATA.request id: 115, addrmode: 0x03, addr: 0x00178801104f1238, profile: 0x0104, cluster: 0x0006, ep: 0x01 -> 0x0B queue: 0 len: 5 tx.options 0x00
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:653 Websocket 127.0.0.1:52800 send message: {"e":"changed","id":"285","r":"sensors","state":{"buttonevent":1000,"lastupdated":"2020-11-22T13:30:10.606"},"t":"event","uniqueid":"00:15:8d:00:02:00:8d:ab-01-0006"} (ret = 166)
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:654 Websocket 127.0.0.1:52842 send message: {"e":"changed","id":"285","r":"sensors","state":{"buttonevent":1000,"lastupdated":"2020-11-22T13:30:10.606"},"t":"event","uniqueid":"00:15:8d:00:02:00:8d:ab-01-0006"} (ret = 166)
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:715 discard sensor state push for 285: state/lastupdated (already pushed)
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:716 rule event /sensors/285/state/lastupdated: 0 -> 0
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:722 discard sensor state push for 285: state/buttonevent (already pushed)
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:724 discard sensor state push for 285: state/lastupdated (already pushed)
Nov 22 14:30:10 pi2 deCONZ[6099]: 14:30:10:725 rule event /sensors/285/state/lastupdated: 0 -> 0
```

ebaauw, 2020-11-22:
Checking a triple press, the plugin seems to do OK.  As you can see from the log, the switch uses attribute 0x8000 to report double, triple, and quadruple press.  It reports single press using the _On/Off_ attribute, but sends both values 0 and 1 in a single report.
```
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:912 APS-DATA.indication srcAddr: 0x3cae, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -49
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:913 #011asdu: 18600a00802003
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:914 Node data 0x00158d0002008dab profileId: 0x0104, clusterId: 0x0006
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:918 [INFO] - Button 1005 Triple press lumi.sensor_switch.aq2
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:919 ZCL attribute report 0x00158D0002008DAB for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:920 #011payload: 00802003
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:927 Websocket 127.0.0.1:52800 send message: {"e":"changed","id":"285","r":"sensors","state":{"buttonevent":1005,"lastupdated":"2020-11-22T13:30:14.919"},"t":"event","uniqueid":"00:15:8d:00:02:00:8d:ab-01-0006"} (ret = 166)
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:928 Websocket 127.0.0.1:52842 send message: {"e":"changed","id":"285","r":"sensors","state":{"buttonevent":1005,"lastupdated":"2020-11-22T13:30:14.919"},"t":"event","uniqueid":"00:15:8d:00:02:00:8d:ab-01-0006"} (ret = 166)
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:934 discard sensor state push for 285: state/lastupdated (already pushed)
Nov 22 14:30:15 pi2 deCONZ[6099]: 14:30:14:936 rule event /sensors/285/state/lastupdated: 0 -> 0
```

manup, 2020-11-22:
>That only makes matters worse. Still two rule events for lastupdated, but now the buttonevent is 1000 instead of 1002.

I think this happens due following code:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/sensor.cpp#L552-L555

    "lumi.sensor_switch.aq2".startsWith("lumi.sensor_switch") == true"

So the button map of `lumi.sensor_switch` will be selected since `lumi.sensor_switch.aq2` is not present at a prior position  any longer. `startsWith()` is dangerous in this context.

Can you please remove the "lumi.sensor_switch" entry for the test as well?

ebaauw, 2020-11-22:
OK, after disabling both buttonmap entries, single press works.  But now double, triple, and quadruple press are broken.  After just disabling the first entry (for value "0"), all button actions seem to work.

manup, 2020-11-22:
Good to know, I'll get all Xiaomi switches which we have in the office during the week and make some tests so that `buttonevents?? work as expected.

Dischi, 2020-12-04:
Instead of creating a new issue, I guess hijacking this one is a good idea. ince the last update my Ikea Remotes also stopped sending  updates on the websocket and I also see "discard sensor state push" and "already pushed". I readded the remotes and they work until I restart the deconz service. They also work with deconz-internal rules. They only stopped sending updates over the websocket, Reverting to an older version fixed that problem.

Maybe the isues are related

ebaauw, 2020-12-14:
Still not fixed in v2.07.01?  Workaround (disabling entry in `button_map.json`) still works, though.

Mimiix, 2020-12-16:
@ebaauw Can you confirm later on if this is fixed?

ebaauw, 2020-12-16:
It was fixed in the alpha version already.

closed 2020-12-16 by ebaauw