dresden-elektronik/deconz-rest-plugin #3720: TRADFRI on/off switch (Button) from IKEA of Sweden, FC7C cluster attributes : what for ?

yves19, 2020-11-20:
## Describe the question or issue you are having

Hello.  Just a question about TRADFRI on/off switch (Button) from IKEA of Sweden
This switch has a cluster FC7C "IKEA"  which contains two undocumented rw attributes. Does anybody know what is the purpose of these attributes and the meaning of their values ?

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi and NUC intel 8i5BEH
 - Running method: Raspbian , Win10, Debian 10.5
 - Firmware version: 26660700
 - deCONZ version: (.05.88
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

n/a
## Additional context

n/a

Smanar, 2020-11-20:
Good question !
If someone have added this 2 clusters with the type, there is a reason.

I m curious too ^^.

MattWestb, 2020-11-21:
IKEA is using then then setting up devices with touchlink but its no known what they is doing / used for.
One possible thing is for lights is making the new storad scenes only adjusting the colour and not the on/off state and brightness (all attributes and also windows covers is stored in the scenes is added from the GW or controlling device) then using the 5 button remote (have see that in sniffs for getting the remote working with scenes that was successful done).

closed 2020-12-25 by stale[bot]