dresden-elektronik/deconz-rest-plugin #3726: Device state not exposed from the API

nickpanaiotov, 2020-11-21:
## Describe the question or issue you are having
I have xiaomi zegbee plug and I can see in the logs of the container properties like **power**, **consumption**, **voltage**, **current** and etc. The question is why those are not returned from the API? Is it because the plug is recognized as "light"? Is it a way to get those properties exposed?

**GET** **http://local_adress/api/API_KEY/lights**
```
{
    "1": {
        "etag": "8005d7fb0f2f2f1c8d3b5c8483ce8b8f",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-21T14:54Z",
        "manufacturername": "dresden elektronik",
        "modelid": "ConBee II",
        "name": "Configuration tool 1",
        "state": {
            "reachable": true
        },
        "swversion": "0x26580700",
        "type": "Configuration tool",
        "uniqueid": "00:21:2e:ff:ff:06:1a:7e-01"
    },
    "3": {
        "etag": "7e647a1543a6cb377a57f08137c4f1f3",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2020-11-21T14:52Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.plug.mmeu01",
        "name": "Smart plug 3",
        "state": {
            "alert": "none",
            "on": true,
            "reachable": true
        },
        "swversion": "09-06-2019",
        "type": "Smart plug",
        "uniqueid": "04:cf:8c:df:3c:7b:1d:fc-01"
    }
} 
```

 Not sure what "Configuration tool 1" and if it's important.

## Screenshots
[image]

[image]

## Environment

 - Host system:  Ubuntu server 16.04
 - Running method: Marthoc Docker container
 - Firmware version: 0x26660700
 - deCONZ version: 2.06.00 / 11/15/2020
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

```
deconz           | 14:41:24:841 ZCL attribute report 0x04CF8CDF3C7B1DFC for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
deconz           | 14:41:24:841 0x04CF8CDF3C7B1DFC extract Xiaomi special attribute 0x00F7
deconz           | 14:41:24:841 	64 on/off 1
deconz           | 14:41:24:841 	03 temperature 31 °C
deconz           | 14:41:24:841 	98 power 560.260010 (560)
deconz           | 14:41:24:841 	95 consumption 2.328409 (2328)
deconz           | 14:41:24:841 	96 voltage 2300.000000 (230)
deconz           | 14:41:24:841 	97 current 2435.913086 (2436)
deconz           | 14:41:24:841 	05 RSSI dB (?) 1 (0x0001)
deconz           | 14:41:24:842 	9a unknown 0 (0x00)
deconz           | 14:41:24:842 	08 unknown 278 (0x0116)
deconz           | 14:41:24:842 	07 unknown 0 (0x0000000000000000)
deconz           | 14:41:24:842 	09 unknown 770 (0x0302)
deconz           | 14:41:24:842 	0b unknown 0 (0x0000)
deconz           | 14:41:24:842 	9b Consumer connected (yes/no) 1
```

## Additional context

Smanar, 2020-11-21:
Hello, this device create at least 2 entry, 1 "light" with for exemple on/off, and a second one in "sensor" not visible in phoscon that use the state you miss.

SwoopX, 2020-11-21:
So, in other words: query the API for sensors 😉

nickpanaiotov, 2020-11-22:
Hi, thank you. The thing is that initially it didn't create it, I tried number of times to reset just the plug. I had to reset the whole configuration (rm config volume) which removed all my other sensors. After that when I added the plug as first device two more sensors were created (consumption and power).  

So overall the issue is resolved. But before closing the ticket, I want to ask, is it possible to merge configurations or to manually add configuration? 

When I wandering between the two configs (the old that I deleted and the new one) I saw that in the latter one 

[image]

in **session.default**

```
{
    "ExtAddress": "0x04cf8cdf3c7b1dfc",
    "NodeDescriptor": "AUCOXxF/ZAAALGQAAA==",
    "NwkAddress": "0xa774",
    "SceneX": -2.0,
    "SceneY": 87.0,
    "SimpleDescriptors": [
      "AQQBUQABBwAAAgADAAQABQAGAMD8AgoAGQA=",
      "FQQBUQABAQwAAA==",
      "FgQBUQABAQwAAA==",
      "8uChYQAAAAEhAA=="
    ],
    "UserDescriptor": "Wall Plug"
  }
```

three more descriptors than the original 

```
      "FQQBUQABAQwAAA==",
      "FgQBUQABAQwAAA==",
      "8uChYQAAAAEhAA=="
```

would it had helped if I added them manually (sorry I lost the initial config and I can't post the before state) or session.default is just informational?

Smanar, 2020-11-22:
I think it will be complicated than that, you have data in the zll.db file too.
But why do you want to "merge" configuration ? I can understand users that want more than one config, but I think using and make backup on the last one is enought nope ?

nickpanaiotov, 2020-11-22:
The scenario that I can think of is, what would happen when I get the next sensor and for some reason is partially discovered. And the trick with starting fresh works again. I'll need to add all my sensors again and then to re import them to the correct locations and to reconfigure them on every integration as home assistant, homekit and etc. 

However this is theoretical, you can close the issue. Thank you.

Smanar, 2020-11-22:
If a device is partially discover, just restart the inclusion, and without deleting it.

SwoopX, 2020-12-12:
Closing as suggested by op.

closed 2020-12-12 by SwoopX