dresden-elektronik/deconz-rest-plugin #3729: Phoscon Rule Motion Sensor want to add specified time range #3728

clipse2004, 2020-11-22:
2.05.88

## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: Raspberry
 - Running method: Raspbian / deCONZ
 - Firmware version: 26660700
 - deCONZ version: 2.05.88
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Smanar, 2020-11-22:
Hello, I m not using rules on my side, I have found this issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2010 with some usefull comment, but not easy to understand.

clipse2004, 2020-11-22:
Hello,
Yep I found this too and tried it, but I cannot find my rule to add this.
That’s my problem :/

closed 2020-12-25 by stale[bot]