dresden-elektronik/deconz-rest-plugin #3735: slow response of bulbs and lost commands

capstan1, 2020-11-22:
## Describe the question or issue you are having

## Screenshots

[image]
[image]
[image]

## Environment

 - Host system: Raspberry Pi 3B+
 - Running method: Home Assistent deCONZ Add-on 
 - Firmware version: 26390500
 - deCONZ version: 2.05.88 / 15.10.2020
 - Device: RaspBee I 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context
- 16 device which can act as routers (bulbs, lightstrips, drivers...) from Hue, Tradfri, Gledopto, Müller tint, Osram
- 26 battery powered devices (Aqara switches, Aqara motion sensors, Aqara tempereature sensors, 1 Hue motion sensor, 2 Hue remotes, 2 tradfri remotes
- only 2 lights are controlled by a remote group in deconz (hue, tradfri remote), these usually respond quickly to presses
- as the Aqara switches don't support remote groups they are controlled through node red or via rules (button events) written to deconz using Hue Essentials toggling the lights on button presses

Smanar, 2020-11-22:
Hello, using your logs, can you check, if the delay
- come from the instant where you press the switch and the request is received by node red
- come from the instant node red send the command and the bulb receive it
- come from both.

BTW you can too add light in a group (without remote) and toogle the group using automation (If you don't already do that ?)

capstan1, 2020-11-22:
Thanks for your answer!
I'm not exactly sure where I could check all this together. (Logs from Deconz, HA and Node Red in one place) Do you have any idea?
But overall I _think_ it's a combination of both, because:
a) some switches are not configured in node red, but using hue essentials
b) there is a delay between using the remote and the event getting shown in home assistant (HA receiving remote command) **AND** there is a delay when switching lights using home assistant (bulb receiving command)

I already add various lights in light groups and control them together. I just wanted to say that I have controller groups, where you add lights to a controller directly (those controls even work when deconz is turned off). Those groups tend to work better (possibly because there is a direct connection).

Fruchuxs, 2020-11-23:
I have a similar problem with my enocean friends of hue switch - but the switch is configured directly in deconz. With the hue bridge, the switch always reacts quickly.

In general some lights reacts slowly or stucks on / off or in a very low brightness (hue bulbs). Will open a new ticket in the afternoon.

capstan1, 2020-11-23:
I can try configuring one of the aqara switches in deconz directly if that helps. Just haven't done that because deconz doesn't support setting "toggle" (at least with the aqara switches"

Mimiix, 2020-11-23:
What SD Card are you using? I often see this behavior with a slow/old SD card. Might be worth to check that .

capstan1, 2020-11-23:
Im using a 32GB Samsung Evo Class 10, which is around 6 months old (I guess). Tried it with an sandisk Extreme USB stick yesterday, which made the issue worse.
I ordered a second sd card which is unused, gonna try this one out later!
(Just assumed that the sd card won't affect deconz, but I guess, the rules are written to the container, right?)

Mimiix, 2020-11-23:
It shouldn't. 

The thing is mostly that the read/write is not fast enough to keep up. It might be worth to check the current load on HA with the Glances addon. Can you share a screen with that addon open?

capstan1, 2020-11-23:
Here you go
[image]

Fruchuxs, 2020-11-23:
> What SD Card are you using? I often see this behavior with a slow/old SD card. Might be worth to check that .

SanDisk Extreme microSDHC 32GB 
It is possible to optimize the performance via tempfs?

SwoopX, 2020-11-23:
You wrote

> only 2 lights are controlled by a remote group in deconz (hue, tradfri remote), these usually respond quickly to presses
>
> as the Aqara switches don't support remote groups they are controlled [...]

Has the reaction been differently in the past regarding the button presses? To me, it sound like expected behaviour when not using zigbee groups for switches to switch lights. Had cases here where it took up to 10 secs to put on all lights.

Mimiix, 2020-11-24:
Your ram is almost completely in use. That might cause delays. I also side with Swoop on this one.

capstan1, 2020-11-24:
To make it a bit clearer:
- all lights that need to be controlled together are in respective groups in deconz
A) remotes that allow this are in "remote groups in deconz":
[image]
[image]
B) as aqara switches don't allow these special groups and deconz doesn't allow setting "toggle" for lights (just on/off) I tried to things:
1) getting the deconz event with node red and reacting to it
2) setting deconz rules in Hue Essentials (which I Guess is written somewhere in the deconz container as a rule):
[image]

Both these options (1 and 2) are controlling groups (if applicable) and there's no difference between the options.
Option A) tends to be fast but sometimes still get cought by the error. Also there are times where controlling lights directly from Home Assistant is very slow.

Smanar, 2020-11-24:
I have ever see this kind of issue for badly device, one time for a philips bulb and an osram plug, after removing it the network was fine.
Or another exemple with ikea repeater, but without solution (still a mystery) https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2967

capstan1, 2020-11-24:
Is there any way to debug this? Don't know which device to remove first.
Ordered some tradfri bulbs to exchange others in my system, but trial and error will need a lot of time.

Smanar, 2020-11-24:
Depend of the frequency you have the bug.
If you have it all the time, just power off 50% of your bulb, wait for the zigbee network stabilise, and make tries.
If you don't have issues, you have found a wy to explore, else, the problem is from the 50% resting or it s another problem.

Fruchuxs, 2020-11-25:
> I have a similar problem with my enocean friends of hue switch - but the switch is configured directly in deconz. With the hue bridge, the switch always reacts quickly.
> 
> In general some lights reacts slowly or stucks on / off or in a very low brightness (hue bulbs stucks on low light, deconz lightstrip controllers doesn't reacts). Will open a new ticket in the afternoon.

Maybe my problem is another one. I had a toggle on_off rule on alle Buttons configured in deconz. To switch the lights on, it always works, but since today, the toggle off is a problem.
The Logs say:
```
08:28:00:335 GW firmware version: 0x26660700
08:28:00:808 0x001788010434AEAA error APSDE-DATA.confirm: 0xE1 on task
08:28:06:229 skip binding for attribute reporting of ep: 0x00 cluster 0x0001 (end-device might sleep)
08:28:07:389 GP srcId: 0x01715B1D cmd: 0x12 frameCounter: 6892
08:28:07:390 [INFO] - Button 4000 (null)
08:28:07:393 rule event /sensors/33/state/lastupdated: 0 -> 0
08:28:07:625 GP srcId: 0x01715B1D cmd: 0x16 frameCounter: 6893
08:28:07:625 [INFO] - Button 4002 (null)
08:28:07:628 rule event /sensors/33/state/lastupdated: 0 -> 0
08:28:07:629 trigger rule 7 - Rule TOOGLE_ON
08:28:08:075 0x00212EFFFF038983 error APSDE-DATA.confirm: 0xE1 on task
08:28:08:458 GP srcId: 0x01715B1D cmd: 0x12 frameCounter: 6894
08:28:08:459 [INFO] - Button 4000 (null)
08:28:08:462 rule event /sensors/33/state/lastupdated: 0 -> 0
08:28:08:670 GP srcId: 0x01715B1D cmd: 0x16 frameCounter: 6895
08:28:08:670 [INFO] - Button 4002 (null)
08:28:08:674 rule event /sensors/33/state/lastupdated: 0 -> 0
08:28:08:674 trigger rule 8 - Rule TOOGLE_OFF
08:28:08:836 0x0000000000000000 error APSDE-DATA.confirm: 0xE1 on task
08:28:09:528 Set sensor check interval to 1000 milliseconds
08:28:10:278 dev /dev/ttyAMA0
08:28:10:278 GW firmware version: 0x26660700
08:28:10:472 GP srcId: 0x01715B1D cmd: 0x13 frameCounter: 6896
08:28:10:473 [INFO] - Button 3000 (null)
```

But I have no idea which device is my switch. Maybe it's a good idea to show up the addresses in the deconz web gui.

Smanar, 2020-11-25:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

From the wiki 
>Indicates interference issues when displayed very often. It is issued when a message couldn't be send due the medium (air) is too busy, the radio uses a listen-before-talk mechanism before sending out any message.

capstan1, 2020-11-25:
Thanks! 
The only thing that's possible is interference from my wifi. The emptiest spot is the far right of the spectrum (25 for zigbee). However, I'm not able to change the channel. It always stops.
`22:20:21:105 Websocket disconnected 127.0.0.1:35708, state: 0, close-code: 1000, reason: 
22:20:23:453 channel is 15 but should be 25, start channel change
22:20:23:454 Skip automatic channel change, TODO warn user
22:20:33:453 channel is 15 but should be 25, start channel change
22:20:33:454 Skip automatic channel change, TODO warn user`

Any idea?

Smanar, 2020-11-25:
Lol, I ask  tips for this bug every 15 days ^^, if you have the GUI is easy to solve it, just set correctly the channel using join / quit.
But using phoscon I don't know the magic trick, I m interested too , if someone have the perfect method.

capstan1, 2020-11-26:
So, apparently changing the channel did _something_. All my lights are disconnected 😅 Weirdly I can see their state in deconz (wethere they are on or off) but they won't connect, even after restarts.
[image]

So, gonna use the weekend to repair. As the battery operated devices are still connected, reconnecting only lights should be okay.
Or do someone think I should just reset the Gateway and start from scratch? (Would add a lot more hassle, but if it helps 🤷‍♂️ )

Smanar, 2020-11-26:
Haaoww, you have lost all routers but no end devices ? Strange, it s not the reverse that happen usually ?
You have correctly following the actions in the good order to change the channel ? https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help

Fruchuxs, 2020-11-27:
> https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log
>
> From the wiki

My Pi lies between some USB (Mouse, Keyboard) and Display Port Cables. Seems to interference zigbee. Now I just have "delay" entries.

capstan1, 2020-11-29:
> Haaoww, you have lost all routers but no end devices ? Strange, it s not the reverse that happen usually ?
> You have correctly following the actions in the good order to change the channel ? [...]

So this didn't help, so I tried to reset the raspbee (hit reset gateway, deinstalled deconz addon from HA). Weirdly sometimes it saw the old entities again, so I tried various times.
Yesterday I joined most lights, the delay was still there and huge. Worse: during the afternoon all connection were lost again. What should I do? Gonna try to start over today agin with just some lights, but is there any other idea?

Btw: changed the sd-card to a nearly new one, which didn't help. Also memory and swap stays near 90-100%. Ordered a SSD which will arrive next week.

Smanar, 2020-11-29:
>Also memory and swap stays near 90-100%

Not normal, swoop is using deconz on a raspberry 0 (but he swap to headless when he don't need the GUI)
When you take a look in log you haven't a line that appear more than 1 time by second ? (kind of spam)

Have you tried to power off (using circuit breaker, faster) some device ? to check if you still have the issue ?

capstan1, 2020-12-08:
So, an update now:
1) I repaired all devices to the raspbee
2) moved Home Assistant to a SSD

Sadly, it didn't really help. 
- The time between reactions is huge sometimes, some messages get lost, sometimes the Deconz-UI and/or HueEssentials can't even see which lights are turned on.
- One of my 2 philips bulbs regularly disconnects from my mesh and won't reconnect. It never did this previous to my "channel change disaster".
- I read somewhere that Oscram Plugs and Aqara sensors don't get on along well, so I removed those. Shortly after, my second Hue bulb disconnected and also won't connect until I manually repair it

**Here are some logs:**

> 06:10:26:569 Current channel 11
> 06:10:26:591 Device TTL 1141 s flags: 0x7
> PROTO: CRC error
> 06:10:28:023 binding for cluster 0x0006 of 0x00158D000327DF24 exists (verified by reporting)
> 06:10:28:026 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x00158D000327DF24 (seems to be active)
> 06:10:28:029 binding for cluster 0x0008 of 0x00158D000327DF24 exists (verified by reporting)
> 06:10:28:031 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x00158D000327DF24 (seems to be active)
> 06:10:28:034 binding for cluster 0x0300 of 0x00158D000327DF24 exists (verified by reporting)
> 06:10:28:035 skip configure report for cluster: 0x0300 attr: 0x0007 of node 0x00158D000327DF24 (seems to be active)
> 06:10:28:036 skip configure report for cluster: 0x0300 attr: 0x0003 of node 0x00158D000327DF24 (wait reading or unsupported)
> 06:10:28:037 skip configure report for cluster: 0x0300 attr: 0x0004 of node 0x00158D000327DF24 (wait reading or unsupported)
> 06:10:28:412 ZCL configure reporting rsp seq: 78 0x00158D000327DF24 for ep: 0x01 cluster: 0x0300 attr: 0x0008 status: 0xC1
> 06:10:28:465 verified group capacity: 6 and group count: 2 of LightNode 0x00158d000327df24

> PROTO: CRC error
> 10:02:09:846 device state timeout ignored in state 3
> 10:02:11:147 0x00178801081C4ACD error APSDE-DATA.confirm: 0xA7 on task
> 10:02:25:893 Current channel 11
> 10:02:25:919 Device TTL 2400 s flags: 0x7
> PROTO: CRC error
> 10:02:39:847 device state timeout ignored in state 3
> 10:02:44:347 binding for cluster 0x0000 of 0x000B57FFFE250728 exists (verified by reporting)
> 10:02:44:349 binding for cluster 0x0006 of 0x000B57FFFE250728 exists (verified by reporting)
> 10:02:44:352 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0x000B57FFFE250728 (seems to be active)
> 10:02:44:355 binding for cluster 0x0008 of 0x000B57FFFE250728 exists (verified by reporting)
> 10:02:44:358 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0x000B57FFFE250728 (seems to be active)
> PROTO: CRC error
> 10:02:44:856 verified group capacity: 255 and group count: 2 of LightNode 0x000b57fffe250728

> 08:02:50:961 Device TTL 931 s flags: 0x7
> 08:03:12:967 dev /dev/ttyAMA0
> 08:03:12:970 GW firmware version: 0x26390500
> 08:03:12:971 GW firmware version is up to date: 0x26390500
> 08:03:15:370 ZCL attribute report 0x086BD7FFFE0008EC for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
> 08:03:15:556 0x00178801030C3CD6 error APSDE-DATA.confirm: 0xE9 on task
> 08:03:16:348 0x7CB03EAA00ACDDC4 error APSDE-DATA.confirm: 0xE9 on task
> 08:03:16:819 ZCL attribute report 0x00158D000327DF24 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
> 08:03:50:932 Current channel 11
> 08:03:50:954 Device TTL 871 s flags: 0x7
> 08:03:54:647 0x00178801030C3CD6 error APSDE-DATA.confirm: 0xE9 on task
> 08:04:50:932 Current channel 11
> 08:04:50:954 Device TTL 811 s flags: 0x7
> 08:04:57:014 [INFO] - No button handler for: lumi.remote.b186acn01 endpoint: 0x01 cluster: BASIC (0x0000) command: 0x0A payload[0]: 001
> 08:04:57:015 ZCL attribute report 0x00158D0002B8113D for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 

So this **PROTO: CRC error** is new. Didn't find much, googling it. Anyways: do you think I should exchange the ikea bulbs fro tradfri ones, as those are the ones disconnecting. It's working faster right now (with both disconnected) but it's a period too short to be sure.

Mimiix, 2020-12-08:
08:03:54:647 0x00178801030C3CD6 error APSDE-DATA.confirm: 0xE9 on task
08:03:15:556 0x00178801030C3CD6 error APSDE-DATA.confirm: 0xE9 on task
08:03:16:348 0x7CB03EAA00ACDDC4 error APSDE-DATA.confirm: 0xE9 on task

These are the ones i'm interested in. 

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

capstan1, 2020-12-08:
one of those is the bulb that disconnected various times (Hue), the other I can't identify (maybe its one of the Osram sockets).

I tried to listen to deconz_events in HA when pushing a button and it's mixed. From 10 pushes, 1-2 get delayed. Sometimes the event arrives late in HA, sometimes it arrives fast, but the lights needs time to react.

hdejongh, 2020-12-30:
after 2 weeks of frustration "why doesn't this solve it self" i just posted this message on the community of home-assistant (i deleted it and reposted it here):

Hello:

I have a Dell VEP 1485 running ESXI 6.7 with a conbee2 in a usb port.
On that i have a home assistant VM and the conbee2 usb stick is mapped to the virtual machine.
I had multiple aqara sensors connected to it.
Also in the same house i had 2 HUE bridges with multiple lights/remotes connect.

Last week i moved all hue lights/remotes to the conbee2 and removed all hue bridges.

I have a living room with 13 lights(all hue). My problem is simple when i active a scene or automation not all lights respond directly it can take up to seconds until they respond.
I didn’t have that problem with my hue bridges (which were also connected to HA). My lights would always respond instantly.

Btw the same problem also happens in smaller rooms with only 5 lights…

I have no idea why i get slow responds. I have changed the zigbee channel from 15 to 25 but that didn’t make a difference…

how can i get a better responds?

hdejongh, 2020-12-30:
this is what i see in the logs:
17:42:43:649 delay sending request 225 dt 0 ms to 0x0017880100DD380A, ep: 0x0B cluster: 0x0300 onAir: 1
17:42:43:650 delay sending request 228 dt 0 ms to 0x0017880101138AB6, ep: 0x0B cluster: 0x0008 onAir: 1
17:42:43:650 delay sending request 229 dt 0 ms to 0x0017880101138AB6, ep: 0x0B cluster: 0x0300 onAir: 1
17:42:43:651 	0x0017880100F9563D force poll (2)
17:42:43:661 5 running tasks, wait
17:42:43:679 5 running tasks, wait
17:42:43:694 5 running tasks, wait
17:42:43:707 failed to add task 189598 type: 5, too many tasks
17:42:43:708 5 running tasks, wait
17:42:43:710 5 running tasks, wait
17:42:43:717 	0x0017880100BD49DD force poll (2)
17:42:43:718 delay sending request 217 dt 1 ms to 0x0017880100F9563D, ep: 0x0B cluster: 0x0300 onAir: 1
17:42:43:720 failed to add task 189601 type: 11, too many tasks
17:42:43:721 failed to add task 189602 type: 5, too many tasks
17:42:43:722 5 running tasks, wait
17:42:43:728 failed to add task 189604 type: 14, too many tasks
17:42:43:729 failed to add task 189605 type: 11, too many tasks
17:42:43:731 failed to add task 189606 type: 5, too many tasks

closed 2021-02-08 by Mimiix

capstan1, 2021-04-01:
As an addition to this thread: what helped me at the end was moving the Pi as far as possible away from my router. Now, everything works like a charm! I just don't get, why this didn't affect me from the beginning.
But hey, it works now!