dresden-elektronik/deconz-rest-plugin #3740: Problem upgrading from 2.05.84

RickyTR, 2020-11-23:
## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: PC
 - Running method: Windows 10
 - Firmware version: (26660700)
 - deCONZ version: (2.05.84)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Mimiix, 2020-11-23:
Logs would be useful! Try to check under help > debug view.

RickyTR, 2020-11-24:
This is the log part related to the http request:

10:50:31:110 HTTP client GET C:/Users/********/AppData/Local/deCONZ/bin/plugins/de_web//api/config
10:50:31:113 	/api/config --> HTTP/1.1 404 Not Found
10:50:31:117 http client (***.***.***.***) handled 1 requests
10:50:31:126 ~http client 0x0657af78
10:50:37:129 HTTP client hdr detected:
GET /api/B46C7F84EB HTTP/1.1
Accept-Encoding: gzip
User-Agent: Jetty/9.4.20.v20190813
Host: 192.168.200.50

10:50:37:133 HTTP client GET C:/Users/********/AppData/Local/deCONZ/bin/plugins/de_web//api/B46C7F84EB
10:50:37:135 	/api/B46C7F84EB --> HTTP/1.1 404 Not Found
10:50:37:138 http client (***.***.***.***) handled 1 requests
10:50:37:142 ~http client 0x0657b6c8
10:50:43:613 CTRL got nwk update id 1

Let me know if you need logs from another source.

Mimiix, 2020-11-24:
Also, if you are certain that its the wrong subnet: You can simply navigate to the IP of where deconz is hosted. That's where phoscon is.

If phoscon doesn't find the stick there, it simply isnt started.

I'd like to have logs from APS,ZDP, ZCL and INFO

RickyTR, 2020-11-24:
Already tried to navigate to that IP but it seems the gateway is not discovered. As you can see it's grayed out.

[image]

This is part of the logs you asked for:

11:53:31:871 APS-DATA.request id: 53, addrmode: 0x03, addr: 0x84182600000d121a, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
11:53:31:934 APS-DATA.confirm id: 53, status: 0x00 SUCCESS
11:53:31:938 APS-DATA.confirm request id: 53 -> confirmed, timeout 1606215211
11:53:31:959 APS-DATA.indication srcAddr: 0xc7f4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 79, rssi: -84
11:53:31:962 APS-DATA.indication request id: 53 -> finished
11:53:31:965 APS-DATA.request id: 53 erase from queue
11:53:38:052 Mgmt_Lqi_req zdpSeq: 41 to 0x7CB03EAA0A07A978 start index 0
11:53:38:053 APS-DATA.request id: 59, addrmode: 0x03, addr: 0x7cb03eaa0a07a978, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
11:53:38:095 APS-DATA.confirm id: 59, status: 0x00 SUCCESS
11:53:38:096 APS-DATA.confirm request id: 59 -> confirmed, timeout 1606215218
11:53:50:732 Current channel 20
11:53:50:748 Device TTL 5545 s flags: 0x7
11:53:59:993 APS-DATA.request id: 74, addrmode: 0x02, addr: 0xfffd, profile: 0x0000, cluster: 0x0000, ep: 0x00 -> 0x00 queue: 1 len: 11 tx.options 0x00
11:53:59:996 send NWK_addr_req to 0x7CB03EAA00B1DAA8, last seen 0 s, last seen by neighbors 0 s
11:54:00:672 APS-DATA.confirm id: 74, status: 0x00 SUCCESS
11:54:34:529 APS-DATA.indication srcAddr: 0xc7f4, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 79, rssi: -84
11:54:38:053 aps request id: 59 prf: 0x0000 cl: 0x0031 timeout (confirmed: 1) to 0x7CB03EAA0A07A978 (0xC115)
11:54:38:146 aps request id: 59 finished, erase from queue
11:54:38:428 Mgmt_Lqi_req zdpSeq: 81 to 0x00212EFFFF04E1CA start index 0
11:54:38:431 APS-DATA.request id: 100, addrmode: 0x03, addr: 0x00212effff04e1ca, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
11:54:38:455 APS-DATA.confirm id: 100, status: 0x00 SUCCESS
11:54:38:458 APS-DATA.confirm request id: 100 -> confirmed, timeout 1606215278
11:54:38:518 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 88, rssi: 8
11:54:38:520 APS-DATA.indication request id: 100 -> finished
11:54:38:521 APS-DATA.request id: 100 erase from queue
11:54:38:522 ZDP status = 0x00 -> SUCCESS
11:54:38:523 ZDP Mgmt_Lqi_rsp zdpSeq: 81 from 0x00212EFFFF04E1CA total: 3, startIndex: 0, listCount: 1
11:54:38:524     * neighbor: 0x84182600000D121A (0xC7F4), LQI: 79, relation: 0x02 rxOnWHenIdle: 1
11:54:41:242 Mgmt_Lqi_req zdpSeq: 84 to 0x00212EFFFF04E1CA start index 1
11:54:41:246 APS-DATA.request id: 104, addrmode: 0x03, addr: 0x00212effff04e1ca, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
11:54:41:270 APS-DATA.confirm id: 104, status: 0x00 SUCCESS
11:54:41:273 APS-DATA.confirm request id: 104 -> confirmed, timeout 1606215281
11:54:41:339 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 15, rssi: 122
11:54:41:341 APS-DATA.indication request id: 104 -> finished
11:54:41:343 APS-DATA.request id: 104 erase from queue
11:54:41:344 ZDP status = 0x00 -> SUCCESS
11:54:41:346 ZDP Mgmt_Lqi_rsp zdpSeq: 84 from 0x00212EFFFF04E1CA total: 3, startIndex: 1, listCount: 1
11:54:41:348     * neighbor: 0x7CB03EAA0A07A978 (0xC115), LQI: 41, relation: 0x02 rxOnWHenIdle: 1
11:54:44:053 Mgmt_Lqi_req zdpSeq: 87 to 0x00212EFFFF04E1CA start index 2
11:54:44:056 APS-DATA.request id: 108, addrmode: 0x03, addr: 0x00212effff04e1ca, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
11:54:44:079 APS-DATA.confirm id: 108, status: 0x00 SUCCESS
11:54:44:082 APS-DATA.confirm request id: 108 -> confirmed, timeout 1606215284
11:54:44:139 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 15, rssi: 122
11:54:44:141 APS-DATA.indication request id: 108 -> finished
11:54:44:143 APS-DATA.request id: 108 erase from queue
11:54:44:144 ZDP status = 0x00 -> SUCCESS
11:54:44:146 ZDP Mgmt_Lqi_rsp zdpSeq: 87 from 0x00212EFFFF04E1CA total: 3, startIndex: 2, listCount: 1
11:54:44:148     * neighbor: 0x00158D0002D4826F (0x0427), LQI: 255, relation: 0x01 rxOnWHenIdle: 0
11:54:47:429 Mgmt_Lqi_req zdpSeq: 90 to 0x84182600000D121A start index 0
11:54:47:432 APS-DATA.request id: 112, addrmode: 0x03, addr: 0x84182600000d121a, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 1 len: 2 tx.options 0x00
11:54:47:544 APS-DATA.confirm id: 112, status: 0x00 SUCCESS
11:54:47:548 APS-DATA.confirm request id: 112 -> confirmed, timeout 1606215287
11:54:47:567 APS-DATA.indication srcAddr: 0xc7f4, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 87, rssi: -83
11:54:47:570 APS-DATA.indication request id: 112 -> finished
11:54:47:573 APS-DATA.request id: 112 erase from queue
11:54:47:575 ZDP status = 0x00 -> SUCCESS
11:54:47:577 ZDP Mgmt_Lqi_rsp zdpSeq: 90 from 0x84182600000D121A total: 6, startIndex: 0, listCount: 3
11:54:47:579     * neighbor: 0x7CB03EAA0A07A978 (0xC115), LQI: 50, relation: 0x02 rxOnWHenIdle: 1
11:54:47:581     * neighbor: 0x00212EFFFF04E1CA (0x0000), LQI: 34, relation: 0x02 rxOnWHenIdle: 1
11:54:47:583     * neighbor: 0x00158D0002D4861C (0xB6EE), LQI: 102, relation: 0x01 rxOnWHenIdle: 0
11:54:50:242 Mgmt_Lqi_req zdpSeq: 92 to 0x84182600000D121A start index 3
[... cut ...]
11:54:50:325 APS-DATA.indication request id: 115 -> finished
11:54:50:327 APS-DATA.request id: 115 erase from queue
11:54:50:329 ZDP status = 0x00 -> SUCCESS
11:54:50:332 ZDP Mgmt_Lqi_rsp zdpSeq: 92 from 0x84182600000D121A total: 6, startIndex: 3, listCount: 3
11:54:50:334     * neighbor: 0x00158D0002E6B578 (0x7D7D), LQI: 30, relation: 0x01 rxOnWHenIdle: 0
11:54:50:336     * neighbor: 0x00158D0002CA139F (0x2EFA), LQI: 44, relation: 0x03 rxOnWHenIdle: 0
11:54:50:338     * neighbor: 0x00158D0002CA0E8F (0x21C3), LQI: 78, relation: 0x01 rxOnWHenIdle: 0

SwoopX, 2020-11-30:
> 10:50:31:110 HTTP client GET C:/Users/--------/AppData/Local/deCONZ/bin/plugins/de_web//api/config
> 10:50:31:113 /api/config --> HTTP/1.1 404 Not Found
> 10:50:31:117 http client (...) handled 1 [...]

The GET request looks horribly wrong in the very first line, it should be as mentioned in line 6. Not sure if that's just an error in the debug output or really trying that...

MissMusic, 2020-12-01:
I have the exact same issue, im stuck on 2.05.84.
@RickyTR  Did you get it resolved?

Mimiix, 2020-12-01:
@RickyTR Just wait a bit. It ungreys after a while. Also: try the inprivate mode in chrome.

MissMusic, 2020-12-01:
@Mimiix Wait for how long?
I have waited for 15+ minutes now, still the same error 404 in the logs over and over again.

I have tried a private browser tab, no change. It's stuck on "please wait while searching for Gateways". The REST API connections from HomeAssistant also fails.

The GUI (Windows app) connects to the GW and it finds all the nodes, but it's like the config is blank. The nodes have no names, and none of the web interfaces.

```

14:40:20:932 HTTP client GET C:/Program Files (x86)/deCONZ/bin/plugins/de_web//api/config
14:40:20:933 	/api/config --> HTTP/1.1 404 Not Found
14:40:20:934 http client (192.168.130.7) handled 1 requests
14:40:20:934 ~http client 0x082bcee8
14:40:22:039 HTTP client hdr detected:
GET /api/config?_=1606830021681 HTTP/1.1
Host: 192.168.1.240
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:83.0) Gecko/20100101 Firefox/83.0
Accept: */*
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
DNT: 1
Connection: keep-alive
Referer: http://192.168.1.240/pwa/login.html

14:40:22:041 HTTP client GET C:/Program Files (x86)/deCONZ/bin/plugins/de_web//api/config
14:40:22:041 	/api/config --> HTTP/1.1 404 Not Found
14:40:22:042 http client (192.168.130.7) handled 1 requests
14:40:22:042 ~http client 0x082bcee8
14:40:25:347 HTTP client hdr detected:
GET /api/config?_=1606830025044 HTTP/1.1
Host: localhost
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.67 Safari/537.36
Accept: */*
Sec-Fetch-Site: same-origin
Sec-Fetch-Mode: cors
Sec-Fetch-Dest: empty
Accept-Encoding: gzip, deflate, br
Accept-Language: en-US,en;q=0.9
Cookie: session=08fa4fc974816623421620154c440475; path=/

14:40:25:349 HTTP client GET C:/Program Files (x86)/deCONZ/bin/plugins/de_web//api/config
14:40:25:349 	/api/config --> HTTP/1.1 404 Not Found
14:40:25:350 http client (127.0.0.1) handled 1 requests
14:40:25:350 ~http client 0x082bebe0
14:40:25:955 HTTP client hdr detected:
GET /api/config?_=1606830025603 HTTP/1.1
Host: 192.168.1.240
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:83.0) Gecko/20100101 Firefox/83.0
Accept: */*
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
DNT: 1
Connection: keep-alive
Referer: http://192.168.1.240/pwa/login.html
Cookie: path=/; session=77fa36bb796f7ac733d76744574a3075

14:40:25:956 HTTP client GET C:/Program Files (x86)/deCONZ/bin/plugins/de_web//api/config
14:40:25:957 	/api/config --> HTTP/1.1 404 Not Found
14:40:25:957 http client (192.168.130.7) handled 1 requests
14:40:25:958 ~http client 0x082bf7b0
14:40:27:074 HTTP client hdr detected:
GET /api/config?_=1606830026716 HTTP/1.1
Host: 192.168.1.240
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:83.0) Gecko/20100101 Firefox/83.0
Accept: */*
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate
DNT: 1
Connection: keep-alive
Referer: http://192.168.1.240/pwa/login.html
```

Mimiix, 2020-12-01:
@MissMusic If you've  read my comment closely: I replied to Ricky, not to your issue. Swoop did reply to your issue and you ignored it.

 I do not appreciate issue hijacking and I'd appreciate if you opened your own issue.

RickyTR, 2020-12-01:
Waiting for a couple of minutes did not help and I already tried the incognito mode before posting here. I tried also to remove completely the Deconz app and install everything from scratch but with the same result. I don't know what else to try...

Mimiix, 2020-12-01:
@RickyTR What happens if you try on port 8080? I just had the same issue ... After going to my ip :8080 it was there.

MissMusic, 2020-12-01:
@Mimiix 
Yea, sorry i got it mixed up.

I did not ignore Swoops reply, but it does not include any further questions or any solutions as far as i understand.

I tried a clean install now and i still get the same HTTP errors and can't get the Phoscon app to work even locally with cleared cache or incognito mode.

I also tried port 8080 as you suggested but got no response. Has the port changed? Phostcon have always been on port 80 earlier.

Mimiix, 2020-12-01:
@MissMusic But you sure did ignore mine.

Please, again, open a seperate User question issue.

RickyTR, 2020-12-01:
@Mimiix just tried with the latest version (2.7.00) with the same result. I checked 8080 port also but there's nothing binded on that port.

Mimiix, 2020-12-03:
The thing is: IF you are on your local ip, Phoscon is running but the stick is not working. However, i think we need @YKO-de for this to confirm.

RickyTR, 2020-12-04:
@Mimiix how can be that the stick is not working if the Deconz GUI correctly sees it and all the associated devices? And also if I uninstall the not working version and install 2.5.84 again everything is working again (I’m using it tight now...). Can be something related to the firmware or the device driver?

MissMusic, 2020-12-04:
I'm 99% sure the "wrong subnet" reported have lead this thread astray. 
My take on that is that Phoscon gets no response on the internal API request, most likely because of the butched API-URL (no idea why it is that way). After some time not "finding" any GW that responds, Phoscon triggers a search on all available networks, and therefore also the "wrong subnet".

My system show the exact same behavior and by now I'm convinced me and @RickyTR have the exact same issue.

Here is a recap of my data.
Environment:
    Host system: PC
    Running method: Windows Server 2019
    Firmware version: (26390500)
    deCONZ version: (2.05.88)
    Device: (ConBee I)
    Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
    Is there any other USB or serial devices connected to the host system? Yes

deCONZ GUI on 2.05.88 (or any other version above 2.05.84): (Notice that GW is connected, so is all nodes,  but no names applied).

[image]

Opening the browser on the host, and navigating to its local IP: (http://192.168.1.240/pwa/login.html):

[image]

After a while it gives up, dropping the displayed ConBee1, and starts a search on all networks for a GW.
Clearing browser cache or running in a private window makes no difference.

When opening the Phoscon web page (http://localhost), the HTTP log in the Debug starts to show:

`09:46:39:952 HTTP client GET C:/Program Files (x86)/deCONZ/bin/plugins/de_web//api/config
09:46:39:952 	/api/config --> HTTP/1.1 404 Not Found
09:46:39:953 http client (192.168.1.240) handled 1 requests
09:46:39:953 ~http client 0x07f4b900
09:46:44:973 HTTP client hdr detected:
GET /api/config?_=1607071604972 HTTP/1.1
Host: 192.168.1.240
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.67 Safari/537.36
Accept: */*
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9

09:46:44:974 HTTP client GET C:/Program Files (x86)/deCONZ/bin/plugins/de_web//api/config
09:46:44:974 	/api/config --> HTTP/1.1 404 Not Found
09:46:44:975 http client (192.168.1.240) handled 1 requests
09:46:44:975 ~http client 0x07f4d4d8
09:46:49:995 HTTP client hdr detected:
GET /api/config?_=1607071609994 HTTP/1.1
Host: 192.168.1.240
Connection: keep-alive
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/87.0.4280.67 Safari/537.36
Accept: */*
Accept-Encoding: gzip, deflate
Accept-Language: en-US,en;q=0.9`

Repeating over and over forever.

Closing deCONZ, running the 2.05.84 installer, and starting deCONZ brings everything back to normal:

[image]

[image]

IPSAttain, 2020-12-06:
Hi at all,

i have exactly the same behaivior.
My Enviroment are:
Host system: PC
Running method: Win 10
Firmware version: (26580700)
effected deCONZ version: (all above 2.05.84)
Device: (ConBee II)
Do you use an USB extension cable: (yes)

If i install deConz above 2.05.84 all Nodes are unnamed, connection with phoscon and my home automation system via API as well, is not possible.
I tryed 2.05.86, 2.05.88, 2.06.00 and 2.07.00. All the same

If I install .84, this brings everything back to normal.

brunoz78, 2020-12-11:
We have exactly the same problem.
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3518
Up to version 2.05.84 everything works fine. After that no more. (v2.05.86 - 2.07.01).
If from version 2.05.84 the old DLL (\bin\plugins\de_resp_plugin.dll) is copied to the new version 2.07.01 and then the new version deconz.exe is started, everything is correct again.
Some change must have happened in version 2.05.84 to 86 which causes the error.

tklyger, 2020-12-11:
Hello,

I have the same problem. However, I realised that up to version 2.05.84 the "REST API PLUGIN" plugin is no longer started in deconz application or at least it is no longer listed under the context menu. Apparently the plugin is not loaded at all.

RickyTR, 2020-12-15:
> We have exactly the same problem.
> #3518
> Up to version 2.05.84 everything works fine. After that no more. (v2.05.86 - 2.07.01).
> If from version 2.05.84 the old DLL (\bin\plugins\de_resp_plugin.dll) [...]

I tried your solution and it's working. The web interface shows a wrong version with the correct date (2.05.84 / 12/8/2020).

IPSAttain, 2020-12-16:
I would like to thank manup for fix this issue. I confirm that with Version 2.8.0 everything works as it should.

Mimiix, 2020-12-16:
@RickyTR Should be fixed in 2.8.0!

MissMusic, 2020-12-17:
@Mimiix I'm running 2.8.0 since yesterday, starts up fine and no issues so far. Thank you!

closed 2020-12-18 by YKO-de