dresden-elektronik/deconz-rest-plugin #3746: Issue while adding a Heiman HS1SA-E

TomStarren, 2020-11-23:
## Describe the question or issue you are having
I'm trying to add a Heiman HS1SA-E to deCONZ but it's failing via de webUI. I've tried it via the deCONZ app but there I can see the device but it seems that it's not configured correctly. What I've tried is writing the IAS_CIE_ADDRESS but this fails. I'm pretty new to deCONZ so it could be that I don't know all the information of tricks which can be used.

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi 3
 - Running method: Raspbian
 - Firmware version: 2660700
 - deCONZ version: 2.0.58
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
Let me know what is needed

## Additional context
I'm within a couple of meters of the Raspberry and multiple IKEA Repeaters are also in the network

Thank you in advance!

Smanar, 2020-11-23:
Remember you probably need to make the device awake when sending command.
Can you check the model ID, to be sure it s not a new one pls ? Cluster 0x0000, press the read button (whith making the device alive) if the table is empty.

TomStarren, 2020-11-23:
Here's the screenshot. 
[image]

Smanar, 2020-11-23:
I m perhaps wrong but I think it  s a new version, I can't see it in the code. can explain your problems

TomStarren, 2020-11-23:
That could indeed be an issue, is there anything I need to do for it to be added?

SwoopX, 2020-11-23:
It is no new device but been included like forever. Also, the CIE address should be automatically written during pairing. If you wanna do that manually, you indeed need to make the device awake first.

In doubt, start debug logging and than reset and repair the device. The log might give some further insights.

TomStarren, 2020-11-23:
fixed, thank you!

closed 2020-11-23 by TomStarren

Smanar, 2020-11-24:
>It is no new device but been included like forever.

@SwoopX But I can't see it in whitelist ? what is the trick ?