dresden-elektronik/deconz-rest-plugin #3748: USB SSD boot issue

itfhunter, 2020-11-24:
## Describe the question or issue you are having

Hi,

I installed my rasp pi with on a SD card with that http://deconz.dresden-elektronik.de/sd-card-image-beta/Phoscon_Gateway_Homebridge_2020-10-16.xz ISO. Everything works so far. I want to migrate the SD card to an SSD so I made a copy if the SD to the SSD with ImageWriter.  I use an Y cable as power source for the SSD and an additional power supply (http://shorturl.at/ivDLZ & http://shorturl.at/bBH25 only to make sure that the power supply isn't my issue). Everything still works if I start the rasp with boot from SD card and the SSD connected. But if I remove the SD card an start the rasppi, the conbee II doesn’t work somehow and no device can connect to the rasppi. 

Unfortunately, I'm quite new to rasppi/linux and didn't find a way to verify what the issue is and the link to switch debugging on has only an empty wiki site (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-switches). To make short summery

1) SD card boot + USB SSD works completely
2) USB SSD boot everything works except the conbee II device

Any idea why the Conbee II device don’t want to work
## Screenshots

## Environment

 - Host system: Raspberry Pi 
 - Running method: Raspbian
 - Firmware version: (26610700)
 - deCONZ version: (2.06.00)
 - Device:  RaspBee II
 - Do you use an USB extension cable: yes - ssd
 - Is there any other USB or serial devices connected to the host system? If so: http://shorturl.at/ivDLZ & http://shorturl.at/bBH25

## deCONZ Logs

Couldn'T create a log and "deCONZ GUI: click Help > debug view" on the website doesnt exists
## Additional context

Mimiix, 2020-11-24:
Hi!

Logs are in deCONZ, not phoscon 😅

itfhunter, 2020-11-24:
Ok, then I dont have the gui since I installed the headless image most likely..... and not the HUE version as I thought (had to reinstall the ISO several times during the starting phase...) How to enable the logs then ?

Mimiix, 2020-11-24:
@itfhunter https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-debug-flags

That explains a bit :)

itfhunter, 2020-11-24:
ty, tried it but 
---
:~ $ /usr/bin/deCONZ --dbg-info=2
QCoreApplication::arguments: Please instantiate the QApplication object first
qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.
:~ $
---

I'll need to install a xserver and so on then. this will takes some time :)

itfhunter, 2020-11-25:
Log is now attached, I started vnc and open deconz gui (sry, I'm a noob with linux and it took a bit longer)

Smanar, 2020-11-25:
And why not just installing a complete os on the SSD and use backup/restaure feature from phoscon ?

itfhunter, 2020-11-25:
Because I have also installed openhab, unifi and grafana already. If it wouldnÄT be "only" deconz/phosocn I would do it

itfhunter, 2020-11-25:
What I also tried is to reconnect a device. this is also not working with USB SSD boot

Smanar, 2020-11-25:
If you are using openhab I think you are using a real OS, so you can just install deconz like a standard application, if it s the only one that not working (don't forget make backup first)

itfhunter, 2020-11-25:
I also tried it but it didn't work either since... I dont remember anymore but I tried to get in touch with Dresden via mail. Unfortunately they didn'T replied me. That was the reason why I used the Phoscon ISO. But I will try to make a clean installation and get everything up and running again. Is there a way to get in touch with someone if I encounter any issues then ? (maybe via voice?)

Smanar, 2020-11-25:
Not me, sorry, not skilled enought to talk english ^^.
But if you have error message or can describe the problem I can try  to help you, you can meet to other users on discord, can be faster than github.

itfhunter, 2020-11-25:
German is also possible for me ^_^.
I already tried to make a clean phoscon iso installation on the SSD. still don'T work. So I'm somehow puzzled

Smanar, 2020-11-25:
You have an error message ?
You have a firmware number displayed on phoscon/gateway ?
Headless or desktop ?

itfhunter, 2020-11-25:
Version: 2.05.85 / 15.10.2020
Firmware: 26610700
[log](https://filetransfer.io/data-package/Ij5adK1L#link)

no error message, see log. Nothing happens ......

ISO
Homebridge-hue + Raspbian Buster
Beta Image mit Desktop und Homebridge-hue basierend auf Debian Buster

Smanar, 2020-11-25:
deconz is perfectly working on your log, try to go at http://phoscon.de/app (if possible on the machine you have installed deconz)

SwoopX, 2020-11-25:
You should take logs with `--dbg-error=2 --dbg-info=2` and for a longer timeframe, preferably deconz start is included.

> no error message, see log. Nothing happens ......

Indeed. But we also do not see you opening the network to search for devices via Phoscon. Can you please make sure that you also get that in the logs as well?

itfhunter, 2020-11-26:
[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5600930/log.txt)

I only know one way to add the device and that is via Phoscon Webinterface. I started the deconz with "/usr/bin/deCONZ --dbg-error=2 --dbg-info=2" and tried 2 times to add a device. Log see above

Smanar, 2020-11-26:
Same, for me deconz is correctly working, you  have nothing on the url I have gived ?

Mimiix, 2020-11-26:
@itfhunter Please add --dbg-aps=2 to it .

itfhunter, 2020-11-26:
[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5604566/log.txt)
log now with /usr/bin/deCONZ --dbg-error=2 --dbg-info=2 --dbg-aps=2

tried 2 times to connect a device

itfhunter, 2020-11-26:
> Same, for me deconz is correctly working, you have nothing on the url I have gived ?

I tried it also there but nothing happened

Smanar, 2020-11-26:
Nothing happened ? you can access to Phosocn ?

itfhunter, 2020-11-26:
I can access it but can'T add any device to it either

winfriedp, 2020-11-26:
In your last log I see sometimes status 0XE1. This indicates radio interference (see [here](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log)).
If you did connect the ssd to the USB3 try USB2 or a very long USB extension cable.

itfhunter, 2020-11-26:
I use Raspee and not a Conbee. Also if I boot from SD Card, with USB SSD connected,the isue is gone. 

But I tried also USB 2 port with the USB SSD without any luck

closed 2020-12-25 by stale[bot]