dresden-elektronik/deconz-rest-plugin #3750: Yale YRD226 ZigBee keypad door lock

zeiphon, 2020-11-24:
## Device

- Product name: Yale Assure Lock Touchscreen Keypad
- Manufacturer: Yale
- Model identifier:  YRD226 TSD3
- Device type : 
  - Lock

This is a new issues as requested by @Smanar here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/579#issuecomment-729018037 for Yale lock YRD226.

Using ZigBee module: AYR202-ZB-HA V261 , Model: YRMZB2

I believe the Yale integration guide will probably still apply, found here: http://maxicom.net/public/Yale%20Locks%20ZigBee%20HA%20Integrators%20Guide%20RevC.pdf
(source: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/579#issuecomment-459896727 )

Thanks!

## Screenshots

[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-11-24:
Ok so first try

So you need to instal deconz on it then take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch yale https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

This code is just the first try, with it (if I m right) you will have a "light" entry, with it you ill be able to open / close.

I m on another door lock in same time https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3152

You will have too (if this pr is working) a ZHAopenclose for all missing settings.

zeiphon, 2020-11-25:
@Smanar thanks for that!

I followed your steps, compiled and copied the plugin but unfortunately the door lock still appears the same (only visible in deconz ui, not phoscon or home assistant etc). Are there any logs etc I could provide to assist?

Smanar, 2020-11-25:
Ok, so idk what you have done, but I don't know someone that have succed in doing that you are trying.
The compilation is platform specific too.

You realy need a real OS, or a virtual one but complete, not hassio for exemple, and you can set hassio to use it after.

Where have you compiled the code ? on the hassio docker ?

zeiphon, 2020-11-25:
Ah. I compiled it outside the hassio docker, which is running on a raspberry pi. Hassio add-ons run in their own docker containers so i just copied the compiled plugin into the container, but I guess that may not work..

Smanar, 2020-11-25:
Yes at startup, you have the plugin version, I m using the 2.06.00 base, but I m not realy sure, and idk wich one version is using the hass plugin now. But like I have said, I m almost sure it can't work.

Why you don't use only the raspberry ? You can use deconz on it using hass on another machine.

AntiqS, 2020-11-26:
Hi guys,
Sorry for posting in this topic, but it's probably if not the same but quite a close request.
I bought Yale YRD 265 lock with Zigbee module, and exactly the same scenario - I can lock unlock the lock from Deconz but it is not available in Phoscon or in Home Assistant.

I am running my Home Assistant on NUC PC (no docker), I am happy to post any logs or give any information if it may help.

AntiqS, 2020-11-26:
[image]

Smanar, 2020-11-26:
Sure can add it with mimic the code but I need at least more
- the manufacture number  (missing on this issue, visible on node info panel)
- the model id (same than second capture)

AntiqS, 2020-11-26:
[image]
[image]

Smanar, 2020-11-27:
So thx a lot, I think I have take the wrong manufacture number for previous device.
I have used 0xAAAA, taken from an other issue (the capture is missing in this one), but you have 0x101d, so I think It can be the problem.

BTW I have updated the code for your device
>    { VENDOR_YALE, "YRD226/246 TSDB", emberMacPrefix }, // Yale YRD226 ZigBee keypad door lock

If you can make try pls ?

zeiphon, 2020-11-28:
Just confirming that the manufacturer code for my lock is also 0x101d:

[image]

Smanar, 2020-11-28:
Ok, thx, so the code you have tried first was wrong, can you make a try with the new one ?

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

zeiphon, 2020-12-01:
@Smanar I've spent a number of hours attempting to create a local copy of the deconz hass add-on (but using your branch) without success so far but will keep trying

Smanar, 2020-12-01:
@zeiphon no, don't loose your time with a deconz hass addon copy.

Just install deconz on a real OS or a Full VM, deconz alone is enought for tests, and you can link HA to the new deconz install too if you want to make try with it.

Smanar, 2020-12-06:
@zeiphon @AntiqS no more informations ?

zeiphon, 2020-12-06:
Sorry I haven't had the chance to continue working on this yet, but will soon (probably in the coming week). Unfortunately running deconz on a separate device isn't an option for me at the moment. I could run it separately on the raspberry pi itself but the tighter integration in the add-on is much more convenient for me to remotely manage.

I might just try testing it separately first to give you some feedback then can work on the add-on later perhaps

Smanar, 2020-12-06:
Yep, was just for test, to confirm it working, if you want it in next beta I need to make PR around the 10th

AntiqS, 2020-12-07:
Do I just insert this code somewhere in the deconz config file?
`{ VENDOR_YALE, "YRD226/246 TSDB", emberMacPrefix }, // Yale YRD226 ZigBee keypad door lock`

Smanar, 2020-12-07:
If it s the first time you are compilng the code > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3750#issuecomment-733130164

this line is already in the code https://github.com/Smanar/deconz-rest-plugin/blob/yale/de_web_plugin.cpp#L419

sunshineautomate, 2020-12-11:
I also have the same locks as AntiqS. It would be fantastic to see them integrated....

Smanar, 2020-12-11:
If you have it, do you have an unix machine to help for integration ?

sunshineautomate, 2020-12-11:
I’m using windows unfortunately... I take it that doesn’t help?

Smanar, 2020-12-11:
Nope, sorry you need an unix machine to compile the code.

AntiqS, 2020-12-21:
Hi Smanar,
Cannot compile the code for some reason, I only have very basic linux knowledge. 
Downloaded Ununtu ISO image and installed on a virtual machine.
Trying:
sudo apt update
Then:
sudo apt install deconz-dev

And basically it says: E: Unable to locate package deconz-dev

Do I need to point to a specific package repository?

Smanar, 2020-12-21:
Yep install deconz on it https://phoscon.de/en/conbee2/install#raspbian

It will write in same time the needed line to make the package command working (step 2 and 3)

AntiqS, 2020-12-27:
Ok, I managed to install it on Ubuntu VM. Silly question - do I need actual deconz stick to run the test, right? It's just not that easy for me to use my production stick here, many devices connected. And will take some time to get another conbee stick to play with.

Smanar, 2020-12-28:
No, don't buy a second conbee just for test.

Devices informations are stored on the host, not on the conbee. So you can switch it from your production to the dev host for some minuts, time to make tests, and switch it again just after.

AntiqS, 2020-12-29:
Ok, I did a quick test, it is discovered in Phoscon app

[image]

[image]

But when I click on the icon my lock just beeps (all buttons flash for 2 seconds) and nothing happens.

Via Deconz I can still lock/Unlock:

[image]

Smanar, 2020-12-29:
Ok so I have checked the code again, and sorry I don't see where it can have a problem ...
Can you enable log (in deconz / help) "info" and "info_l2" and remaking the same manipulation (with phoscon then deconz)

AntiqS, 2020-12-29:
So I just plugged the stick back into my VM, enabled the logs, and tried to reproduce - but it just worked. 

[image]

Attaching some logs just in case, cannot explain why it didn't work the first time.

`13:11:46:683 poll node 00:0d:6f:00:10:fe:8d:be-01
13:11:47:263 poll node 00:0d:6f:00:10:fe:8d:be-01
13:11:47:547 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
13:11:47:557 Websocket 192.168.255.133:36206 send message: {"attr":{"id":"1","lastannounced":null,"lastseen":"2020-12-29T21:11Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:04:f9:22-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:04:f9:22-01"} (ret = 344)
13:11:47:560 Websocket 192.168.255.133:36664 send message: {"attr":{"id":"1","lastannounced":null,"lastseen":"2020-12-29T21:11Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:04:f9:22-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:04:f9:22-01"} (ret = 344)
13:11:47:837 poll node 00:0d:6f:00:10:fe:8d:be-01
13:11:48:395 poll node 00:0d:6f:00:10:fe:8d:be-01
13:11:48:853 Erase task req-id: 56, type: 39 zcl seqno: 36 send time 3, profileId: 0x0104, clusterId: 0x0101
13:11:48:921 poll node 00:0d:6f:00:10:fe:8d:be-01
13:11:49:341 Websocket 192.168.255.133:36206 send message: {"attr":{"id":"2","lastannounced":null,"lastseen":"2020-12-29T21:11Z","manufacturername":"Yale","modelid":"YRD226/246 TSDB","name":"Door Lock 2","swversion":"0x0105002B","type":"Door Lock","uniqueid":"00:0d:6f:00:10:fe:8d:be-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"00:0d:6f:00:10:fe:8d:be-01"} (ret = 318)
13:11:49:342 Websocket 192.168.255.133:36664 send message: {"attr":{"id":"2","lastannounced":null,"lastseen":"2020-12-29T21:11Z","manufacturername":"Yale","modelid":"YRD226/246 TSDB","name":"Door Lock 2","swversion":"0x0105002B","type":"Door Lock","uniqueid":"00:0d:6f:00:10:fe:8d:be-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"00:0d:6f:00:10:fe:8d:be-01"} (ret = 318)
13:11:49:348 APS-DATA.indication from child 0xCA4F
13:11:49:450 poll node 00:0d:6f:00:10:fe:8d:be-01
13:11:49:973 poll node 00:0d:6f:00:10:fe:8d:be-01
13:11:50:427 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
13:11:50:501 poll node 00:0d:6f:00:10:fe:8d:be-01
13:11:50:555 DB save zll database items 0x00000001
13:11:50:556 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('2', 'normal', '00:0d:6f:00:10:fe:8d:be-01', 'Door Lock 2', '65520,1', '1', 'YRD226/246 TSDB', 'Yale', '0x0105002B', '{"attr/id":"2","attr/lastannounced":null,"attr/lastseen":"2020-12-29T21:11Z","attr/manufacturername":"Yale","attr/modelid":"YRD226/246 TSDB","attr/name":"Door Lock 2","attr/swversion":"0x0105002B","attr/type":"Door Lock","attr/uniqueid":"00:0d:6f:00:10:fe:8d:be-01","state/alert":null,"state/on":true,"state/reachable":true}')
13:11:50:562 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('1', 'normal', '00:21:2e:ff:ff:04:f9:22-01', 'Configuration tool 1', '65520', '1', 'ConBee II', 'dresden elektronik', '0x26660700', '{"attr/id":"1","attr/lastannounced":null,"attr/lastseen":"2020-12-29T21:11Z","attr/manufacturername":"dresden elektronik","attr/modelid":"ConBee II","attr/name":"Configuration tool 1","attr/swversion":"0x26660700","attr/type":"Configuration tool","attr/uniqueid":"00:21:2e:ff:ff:04:f9:22-01","state/reachable":true}')
13:11:50:569 DB saved in 14 ms
13:11:51:026 poll node 00:0d:6f:00:10:fe:8d:be-01
13:11:51:556 poll node 00:0d:6f:00:10:fe:8d:be-01`

Smanar, 2020-12-30:
I m seing your device is not in bind list. Yoi haven't state return problem if you use it manualy ?

Have modified the code again, now you will have a sensor too, on it the battery level and the lock state (and in the future the data you want)

So you need to re -include it, or just set phosocn in permit join (add new light) and read basic attribute again (it will trigger an inclusion)

Smanar, 2021-01-03:
No news about the special sensor device ?

AntiqS, 2021-01-05:
Not everything is clear to me what you are asking me to do.
What I have done:
1. In Deconz I set permit to join
[image]
2. In Basic attributes I clicked Read
[image]
3. Then enabled logs
4. In Deconz I clicked Lock, then Unlock
5. Switched to Phoscon and clicked Lock, then Unlock
[image]

Where can I the battery level and the lock state?

Here are the logs:
_16:54:26:338 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:26:895 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:27:420 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:27:948 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:28:473 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:28:002 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:29:526 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:30:055 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:30:580 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:31:109 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:31:685 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:32:264 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:32:839 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:33:397 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
16:54:33:399 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:33:923 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:34:452 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:34:977 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:35:505 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:36:029 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:36:566 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:37:089 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:37:618 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:37:680 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:54:38:195 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:38:770 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:39:347 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:39:895 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:40:420 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:40:559 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:54:40:948 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:41:473 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:42:002 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:42:432 DB save zll database items 0x00000001
16:54:42:435 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('1', 'normal', '00:21:2e:ff:ff:04:f9:22-01', 'Configuration tool 1', '65520', '1', 'ConBee II', 'dresden elektronik', '0x26660700', '{"attr/id":"1","attr/lastannounced":null,"attr/lastseen":"2021-01-05T00:54Z","attr/manufacturername":"dresden elektronik","attr/modelid":"ConBee II","attr/name":"Configuration tool 1","attr/swversion":"0x26660700","attr/type":"Configuration tool","attr/uniqueid":"00:21:2e:ff:ff:04:f9:22-01","state/reachable":true}')
16:54:42:447 DB saved in 14 ms
16:54:42:527 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:43:056 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:43:394 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
16:54:43:407 Master: read param with arg 0x19
16:54:43:417 Current channel 15
16:54:43:429 Device TTL 2807 s flags: 0x7
16:54:43:436 Outgoing frame counter 6806987 (0x0067DDCB)
16:54:43:471 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:54:43:581 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:44:096 APS-DATA.indication from child 0xCA4F
16:54:44:110 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:44:684 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:45:263 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:45:838 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:46:320 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:54:46:395 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:46:921 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:47:450 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:47:974 poll node 00:0d:6f:00:10:fe:8d:be-01
16:54:48:504 poll node 00:0d:6f:00:10:fe:8d:be-01
[... 124 lines cut ...]
16:55:30:894 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:31:423 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:31:951 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:32:476 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:32:941 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:55:33:006 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:33:394 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
16:55:33:530 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:34:058 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:34:581 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:35:109 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:35:683 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:35:815 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:55:36:263 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:36:838 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:37:395 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:37:921 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:38:448 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:38:687 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
16:55:38:692 Websocket 192.168.255.133:43992 send message: {"attr":{"id":"1","lastannounced":null,"lastseen":"2021-01-05T00:55Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:04:f9:22-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:04:f9:22-01"} (ret = 344)
16:55:38:974 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:39:503 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:40:027 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:40:555 poll node 00:0d:6f:00:10:fe:8d:be-01
16:55:41:080 poll node 00:0d:6f:00:10:fe:8d:be-01_

AntiqS, 2021-01-05:
Also found where it beeps - from the lights menu:
[image]
But works fine from the home page:
[image]

AntiqS, 2021-01-05:
[image]

zeiphon, 2021-01-05:
@AntiqS when you select a bulb from the lights menu I think deCONZ/phoscon might pulse the bulb to show that it's selected iirc? Perhaps that could be what's happening since it still thinks the lock is a bulb?

Sorry, I haven't had been able to get back to my lock much to help with the testing (it's not located at my house)

Mimiix, 2021-01-05:
@antiqs if you click it in that lights section, it uses the identifying cluster. The other one is the acutal toggle.

AntiqS, 2021-01-06:
What do you mean "Set phoscon in permit join"? Just click add new lights and while it is searching it kind of enables permit to join? And while searching for new lights go to deconz and read attributes? Do I understand it right?

Mimiix, 2021-01-06:
Yes

AntiqS, 2021-01-08:
Ok, I started searching for new lights in Phoscon app and then read attributes in Deconz:

[image]
[image]
[image]

Logs while searching for lights in Phoscon and trying to unlock/lock from phoscon and then from deconz:

> 16:59:25:524 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:26:051 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:26:576 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:26:720 Websocket disconnected 192.168.255.133:51030, state: 0, close-code: 1001, reason: 
> 16:59:27:104 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:27:394 reset device retries: 4
> 16:59:27:395 reset device apsdeDataRequest success
> 16:59:27:678 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:28:255 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:28:405 Master: read param with arg 0x19
> 16:59:28:417 Current channel 15
> 16:59:28:438 Device TTL 1336 s flags: 0x7
> 16:59:28:464 Outgoing frame counter 6830322 (0x006838F2)
> 16:59:28:670 New websocket 192.168.255.133:51242 (state: 3) 
> 16:59:28:830 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:28:966 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
> 16:59:29:395 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
> 16:59:29:396 reset device wait for confirm timeout.
> 16:59:29:396 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:29:921 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:30:395 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:30:395 Idle timer triggered
> 16:59:30:450 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:30:975 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:31:503 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:31:843 Websocket 192.168.255.133:51242 send message: {"attr":{"id":"1","lastannounced":null,"lastseen":"2021-01-08T00:59Z","manufacturername":"dresden elektronik","modelid":"ConBee II","name":"Configuration tool 1","swversion":"0x26660700","type":"Configuration tool","uniqueid":"00:21:2e:ff:ff:04:f9:22-01"},"e":"changed","id":"1","r":"lights","t":"event","uniqueid":"00:21:2e:ff:ff:04:f9:22-01"} (ret = 344)
> 16:59:31:870 Websocket disconnected 192.168.255.133:51242, state: 0, close-code: 1001, reason: 
> 16:59:32:027 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:32:395 reset device retries: 3
> 16:59:32:395 reset device apsdeDataRequest success
> 16:59:32:555 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:33:078 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:33:184 New websocket 192.168.255.133:51376 (state: 3) 
> 16:59:33:604 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:33:958 Websocket 192.168.255.133:51376 send message: {"attr":{"id":"2","lastannounced":null,"lastseen":"2021-01-08T00:59Z","manufacturername":"Yale","modelid":"YRD226/246 TSDB","name":"Door Lock 2","swversion":"0x0105002B","type":"Door Lock","uniqueid":"00:0d:6f:00:10:fe:8d:be-01"},"e":"changed","id":"2","r":"lights","t":"event","uniqueid":"00:0d:6f:00:10:fe:8d:be-01"} (ret = 318)
> 16:59:34:180 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:34:244 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
> 16:59:34:303 add task 259 type 39 to 0x000D6F0010FE8DBE cluster 0x0101 req.id 73
> 16:59:34:305 Websocket 192.168.255.133:51376 send message: {"e":"changed","id":"2","r":"lights","state":{"alert":null,"on":false,"reachable":true},"t":"event","uniqueid":"00:0d:6f:00:10:fe:8d:be-01"} (ret = 140)
> 16:59:34:313 Websocket 192.168.255.133:51376 send message: {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 93)
> 16:59:34:315 discard group state push for 65520: state/any_on (already pushed)
> 16:59:34:318 Websocket 192.168.255.133:51376 send message: {"e":"changed","id":"1","r":"groups","state":{"all_on":false,"any_on":false},"t":"event"} (ret = 89)
> 16:59:34:321 discard group state push for 1: state/any_on (already pushed)
> 16:59:34:394 reset device wait for confirm timeout.
> 16:59:34:758 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:35:332 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:35:896 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:36:422 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:36:592 DB save zll database items 0x00000001
> 16:59:36:593 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('2', 'normal', '00:0d:6f:00:10:fe:8d:be-01', 'Door Lock 2', '65520,1', '1', 'YRD226/246 TSDB', 'Yale', '0x0105002B', '{"attr/id":"2","attr/lastannounced":null,"attr/lastseen":"2021-01-08T00:59Z","attr/manufacturername":"Yale","attr/modelid":"YRD226/246 TSDB","attr/name":"Door Lock 2","attr/swversion":"0x0105002B","attr/type":"Door Lock","attr/uniqueid":"00:0d:6f:00:10:fe:8d:be-01","state/alert":null,"state/on":false,"state/reachable":true}')
> 16:59:36:595 DB sql exec REPLACE INTO nodes (id, state, mac, name, groups, endpoint, modelid, manufacturername, swbuildid, ritems) VALUES ('1', 'normal', '00:21:2e:ff:ff:04:f9:22-01', 'Configuration tool 1', '65520', '1', 'ConBee II', 'dresden elektronik', '0x26660700', '{"attr/id":"1","attr/lastannounced":null,"attr/lastseen":"2021-01-08T00:59Z","attr/manufacturername":"dresden elektronik","attr/modelid":"ConBee II","attr/name":"Configuration tool 1","attr/swversion":"0x26660700","attr/type":"Configuration tool","attr/uniqueid":"00:21:2e:ff:ff:04:f9:22-01","state/reachable":true}')
> 16:59:36:599 DB saved in 7 ms
> 16:59:36:950 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:37:395 reset device retries: 2
> 16:59:37:396 reset device apsdeDataRequest success
[... cut ...]
> 16:59:49:419 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:49:946 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:50:470 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:50:565 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
> 16:59:50:999 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:51:523 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:52:050 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:52:575 poll node 00:0d:6f:00:10:fe:8d:be-01
> 16:59:53:103 poll node 00:0d:6f:00:10:fe:8d:be-01

Smanar, 2021-01-08:
I think it miss the part just during the inclusion ?
In fact I have modified the code for the api create a ZHAOpenClose entry, like for door sensor.
But I don't it in your log, it s for that I m asking log during the inclusion when deconz create entry.

I don't need capture, you just need to set phoscon in permit join, and read bsic attribute in deconz, to trigger an inclusion, (easier than classic procedure direclty on the device).

But all seem fine on my side, you realy have no more devie for this one, no ZHAbattery entries too ?

AntiqS, 2021-01-11:
maybe I am doing something wrong.. when you update the code - do I need to re-download it or is it automatically updated?
Can you tell me what do you expect to see and where? Cause for me everything looks fine - it shows that I have 4 batteries AA size and power level 100%

Smanar, 2021-01-11:
Yep you need to type

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

But if you can see the battery, you have the sensor, the battery is not in the "light" entry,, so if in your third app (out of deconz) you have the battery level it mean you have the "sensor" entry.

The device create 2 entry, 1 light one, and a sensor one. I need the sensor JSON to look the working value and add the missing one.

Smanar, 2021-01-16:
Hello, I have a little problem ^^, we have decided to make a ZHADoorlock device instead the actual one, a "sensor" and no more "light"
So I have rewritted all the code, if someone  can test it ?

Now the device will be not visible in phoscon.

To use it you will need to use the API using "config/lock"

for exemple with curl
`curl -H 'Content-Type: application/json' -X PUT -d '{"lock": true}' http://192.168.1.1:80/api/AE78FD8547/sensors/16/config`

The code to compile it
```
sudo apt install deconz-dev
git clone --branch doorlock https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

AntiqS, 2021-01-17:
I compiled the code, my lock disappeared from Phoscon as you said.
Trying to use curl command as you asked, it says unauthorized user. I have deconz application running while trying to run curl.

```
user@ubuntu:~/deconz-rest-plugin$ curl -H 'Content-Type: application/json' -X PUT -d '{"lock": true}' http://192.168.255.133:8080/api/AE78FD8547/sensors/16/config
[{"error":{"address":"/sensors/16/config","description":"unauthorized user","type":1}}]
```

Also, when I tried to update deconz-dev it said that I already have the latest version:
`deconz-dev is already the newest version (2.07.01-ubuntu-focal-stable).`

If I cloze Deconz and run Curl command it says that connection refused (tried 80 and 8080 ports)
```
user@ubuntu:~/deconz-rest-plugin$ curl -H 'Content-Type: application/json' -X PUT -d '{"lock": true}' http://192.168.255.133:8080/api/AE78FD8547/sensors/16/config
curl: (7) Failed to connect to 192.168.255.133 port 8080: Connection refused

```

Smanar, 2021-01-17:
Nice thx ^^

>Also, when I tried to update deconz-dev it said that I already have the latest version

Yeah I put this line every time, not usefull

You are using the same api key than your third app ? I think the first line you have used is the good one. Probably a problem in the api key.

AntiqS, 2021-01-17:
I just copied and pasted your code, only replaced IP address and port.
Anything else I need to change in this command?

Smanar, 2021-01-18:
Ha yes ^^, you need to use your own API KEY.
You have probably one in your third app, else you can create a new one.

`curl http://IP:PORT/api/ -X POST -d '{"devicetype": "TestKey"}'`

If you still have autorisation/authentifiction error, go in phosocn/gateway/advanced/authenticate app and retry the command to have the api key.

AntiqS, 2021-01-18:
when you say "third app" - what do you mean?

Smanar, 2021-01-18:
Home automation, domoticz, openHAB, ect .....

AntiqS, 2021-01-29:
can you confirm please if I am doing it right, I am running the command:

curl -H 'Content-Type: application/json' -X PUT -d '{"lock": true}' http://LOCALHOST-IP:8080/api/MY-API/sensors/16/config

Where LOCALHOST IP is IP address of my ubuntu with plugged in deconz stick, MY API - I got from the home assistant for deconz

In this code - what is number 16?

AntiqS, 2021-01-29:
Assuming number 16 is my lock ID, I tried all numbers from 0 to 20, not sure how to get this ID from Deconz.
All numbers give:
`[{"error":{"address":"/sensors/20","description":"resource, /sensors/20, not available","type":3}}]`
Number 1 returns:
[{"error":{"address":"/sensors/1/config/lock","description":"parameter, lock, not available","type":6}}]

Smanar, 2021-01-29:
16 is the device ID, you can find the list on a browser with 
`http://LOCALHOST-IP:8080/api/MY-API/sensors/`

`[{"error":{"address":"/sensors/20","description":"resource, /sensors/20, not available","type":3}}]`
Mean there is no device with this ID, not visible on the previous link

`[{"error":{"address":"/sensors/1/config/lock","description":"parameter, lock, not available","type":6}}]`
Mean the id is good, there isa device with this ID, but not the field "lock" in it, so can be my fault

You can have the complete JSON just for this device on a browser with :
`http://LOCALHOST-IP:8080/api/MY-API/sensors/1`

AntiqS, 2021-02-01:
I tried that link http://LOCALHOST-IP:8080/api/MY-API/sensors/
it only shows me one device id 1, but it's not the lock - not sure what it is, maybe my Philips hue:

[image]

And I still can lock and unlock the lock from Deconz

Smanar, 2021-02-01:
Ok so you are right, it mean the device was not included in the API, but perhaps in deconz ?
I m checking

```
Model id : YRD226/246 TSDB
Manufacture name 0x101D
Mac adress starting by 0x000d6f
```

Ok, so after a verification I have updated the code, forget your model ... 

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

To update it.

Hoping for a new sensor this time.

>And I still can lock and unlock the lock from Deconz

Ha, if you can see it in deconz, you don't need the reset procedure to re-include it in the network, you just need
- Set phoscon in permit join (add new sensor)
- Read again the basic attribute (cluster 0x0000, button "read"), with perhaps need to awake the device

It will trigger the inclusion in the API (if I m right ofc)

But it s strange you can't use deconz to make the device react, you are using the cluster 0x0101 ? It have never worked for your device ?

almperez, 2021-02-03:
Hi @Smanar! I too have a Yale Smart Lock, but mine is a different model. Can you please add it so I can test it here?

[image]

[image]

Many thanks!

Smanar, 2021-02-03:
Sure, Have added `YRD256L TSDB SL`

Same command to compile

```
sudo apt install deconz-dev
git clone --branch doorlock https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

AntiqS, 2021-02-04:
I think we are getting there :)
I set phoscon app to allow to join and then read basic attributes, new device ID appeared:

[image]

`curl -H 'Content-Type: application/json' -X PUT -d '{"lock": true}' http://LOCALHOST-IP:8080/api/MY-API/sensors/2/config`
locks and returns:
`[{"success":{"/sensors/2/config/lock":true}}]`

`curl -H 'Content-Type: application/json' -X PUT -d '{"lock": false}' http://LOCALHOST-IP:8080/api/MY-API/sensors/2/config`
unlocks and returns:
`[{"success":{"/sensors/2/config/lock":false}}]`

Smanar, 2021-02-04:
But the value
config / lock and state / lockstate are not working ? they are "null" on your capture ?

AntiqS, 2021-02-09:
[image]
"Config / Lock" value - changes to false and stays false no matter what I do.
Tried multiple times (img 1)
Then I set permit to join in phoscon app, read basic attributes and tried locking unlocking. Same results (img 2) - Lock false, lockstate null.

[115 more replies]

closed 2021-04-10 by github-actions[bot]