dresden-elektronik/deconz-rest-plugin #3755: IKEA Tradfri is no longer displayed

thueske, 2020-11-25:
Hey friends!

## Describe the bug
I'm using the ConBee II stick with current deCONZ (docker version) on my Banana Pi. Until a few hours ago everything worked fine. But now I have the problem that the lamp can't be found anymore. Via VNC in the user interface it is still displayed, via the API or the Phoscon app it cannot be found and controlled.

## Steps to reproduce the behavior
Good question... I wish I knew :)

## Expected behavior
The lamp can be operated as set up and does not simply disappear.

## Screenshots
[image]
[image]

## Environment

 - Host system:  Banana Pi with Armbian Buster
 - Running method: Docker
 - Firmware version: 26660700
 - deCONZ version: 2.06.00 / 15.11.2020
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
```
deconz    | 16:10:49:302 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:50:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:51:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:52:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:53:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:54:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:55:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:55:322 Current channel 15
deconz    | 16:10:55:335 Device TTL 3900 s flags: 0x7
deconz    | 16:10:56:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:57:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:58:300 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:10:59:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:11:00:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:11:01:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:11:02:299 skip binding for attribute reporting of ep: 0x00 cluster 0x1000 (end-device might sleep)
deconz    | 16:11:02:496 GW firmware version: 0x26660700
deconz    | 16:11:02:498 GW firmware version is up to date: 0x26660700
```

## Additional context
In the logs I could not find anything noticeable, so here are some screenshots.

It is an Ikea Tradfri lamp E27.

Any ideas where the problem is? A bug that needs to be fixed? Or I can't avoid a new installation?

Many greetings
Tobias

thueske, 2020-11-25:
Addition: The lamp can be controlled by the configured switches, but it is not displayed by Phoscon and not controllable by API (Home Assistant).

SwoopX, 2020-11-25:
The only reason I can currently think of is that deconz was shut down right after it discovered the light so that corresponding information could not be written into the DB. On the other hand, while writing this and looking at your screenshot, you already gave it a name and that name is part of the DB's dataset.

What happens if you start light search again in Phoscon and turn on/off the light a couple of times?

jensflorian, 2020-12-03:
I once experienced a situaion where a IKEA Floalt device was not responding after upgrading deconz.  A short interruption of the mains made the light responsive again.

closed 2021-02-08 by Mimiix