dresden-elektronik/deconz-rest-plugin #3756: IKEA TRADFRI remote control, a long press with the middle button deactivates long presses for the left and right buttons

PremiumUsername, 2020-11-25:
## Describe the bug

A long press of the middle button for an IKEA TRADFRI remote control (5 button fw 1.2.223) will change how the left and right buttons work. Initially (if you remove the battery and put it back in) a long press for the left and right buttons result in xxx1 and xxx3 events as you would suspect. But if you long press the middle button until you get 1001, a long press for the left and right buttons will now produce xxx2 events that automatically repeat every 1 second or so.  There is no way to regain the original xxx1 and xxx3 behaviour part from removing the battery or resetting. 

## Steps to reproduce the behavior

1 remove battery and reinsert it

2 long press the left and/or right buttons, it will work just fine, producing xxx1 and xxx3 events

3 long press the middle button, producing 1002 ---> 1001

3 long press the left and/or right buttons, they can now only produce xxx2 events 

## Expected behavior
...
...
3 no change, same as in step 2 

## Screenshots

## Environment

 - Host system: Raspberry Pi
 - Running method:  Home Assistent deCONZ Add-on /
 - Firmware version: 2.05.88
 - deCONZ version: 26390500
 - Device: ConBee I 
 - Do you use an USB extension cable: 
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

This problem existed in at least some previous versions

PremiumUsername, 2020-11-25:
Before I noticed what triggered this behaviour,  I tried to downgrade the firmware of the remote. When that didn't help I upgraded again. I saw the same thing with the older firmware, i.e. loosing xxx1 and xxx3 events after a while, but I'm not sure if the loss was triggered in the same way.

Smanar, 2020-11-25:
Perhpas https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2618 ?

PremiumUsername, 2020-11-26:
> Perhpas #2618 ?

Yes, that seems like the same issue. I'd add though I don't think this was always the case, so I'm not sure this is all firmware related. I have an old HA script that used xxx1 and xxx3 events, I stopped using it at some point for other reasons but it never failed to produce the right responses while I was using it, and that script also used the long press for the middle button.

PremiumUsername, 2020-11-26:
There is apparently possible to fix with SW, [_Replacing ARGS: [] with ARGS: [0] in the device_automation_trigger and restarting home assistant makes the device work correctly. Is this a bug with this device handler?_](https://github.com/zigpy/zha-device-handlers/issues/553)

PremiumUsername, 2020-11-26:
Final comment, I just tried the ZHA integration in HA and it works perfectly with all the right events (even after a long press for the middle button). So it's definitely something that can be fixed with software. 

A side note. I never tried the ZHA integration before since I thought all the lights would be lost and needed repairing, but after pairing just one router device, a simple power off/on cycle was required to find the rest. So it was rather painless.

Smanar, 2020-11-26:
This feature is not from the firmware, and configurable using the device ?

MattWestb, 2020-11-26:
@PremiumUsername Then ZHA is initiating the Rasp/CornBee that have in one zigbee network thats was configured and saved in in NVR (Internal memory) and you is not forcing some new network parameter ZHA is fine and letting being configured its starts the coordinator. The coordinator is having all information for the network and also the TC-Link keys and security counters. ZHA only need reading the device config and saving it in the database and HA is using the IEEE as device ID from deCONZ and ZHA so all your devices is not loosing your automations and custom tweeks. Only backside is that HA is saying its deCONZ devices and not ZHA or both (bug in HA).

PremiumUsername, 2020-11-28:
> This feature is not from the firmware, and configurable using the device ?

I think the 'feature' is not from the firmware. I haven't looked at the code, but I think the xxx2 events are SF generated. The underlying events are still xxx1 and xxx3 events.  ZHA gives you something more like raw output (in the args[] when you look at the events in HA), and those do not change based on whether a long press was made beforehand with the middle button.

Smanar, 2020-11-28:
But using the same code, you can enable it or disable it just using the device, so for me this feature was in the device ?

SwoopX, 2020-11-28:
> ZHA gives you something more like raw output (in the args[] when you look at the events in HA), and those do not change based on whether a long press was made beforehand with the middle button.

Deconz also gives you the raw output in vairous ways, when enabled. However, I have to admit that there's room for improvement when there's a successful button mapping. Use the `dbg-aps` switch to get the full aps layer info.

samnewman86, 2020-12-29:
im having the same issue. any news on this?

Mimiix, 2021-01-04:
On 2.7.0 This happens only for a short time(2, 3 seconds?) for me. After that, it starts to work again like it should.

samnewman86, 2021-02-04:
> 
> 
> On 2.7.0 This happens only for a short time(2, 3 seconds?) for me. After that, it starts to work again like it should.

After a long period waiting i still get the x002 event repeated every second, rather than a x001 event.
is there something i need to chnage to correct this. im on deconz 2.7.1

samnewman86, 2021-02-05:
I have now updated the remote to the latest firmware and can confirm this issue is still present. Even an hour after long pressing the middle button, a long press of the side buttons creates repeater x002 events rather than a x001 event

bcutter, 2021-02-06:
> 
> 
> I have now updated the remote to the latest firmware and can confirm this issue is still present. Even an hour after long pressing the middle button, a long press of the side buttons creates repeater x002 events rather than a x001 event

Copy that. Latest firmware.

samnewman86, 2021-02-27:
Still seem to be having this issue

Mimiix, 2021-03-01:
I still don't seem to be able to reproduce this. @ebaauw @SwoopX @Smanar What can we do ?

Smanar, 2021-03-01:
It s something in the firmware no ?
Not possible just avoid long press on middle button ?

ebaauw, 2021-03-01:
> What can we do ?

Don't long-press the middle button.  I think the REST API plugin should never have exposed the 1001 `buttonevent` in the first place.  Also, because it doesn't expose a corresponding 1003.

samnewman86, 2021-03-01:
Apparently it works ok with zha?

Mimiix, 2021-03-01:
> Apparently it works ok with zha?
> […](#)

I have no issues with this remote with deCONZ. I did the long press but after 3 seconds it works fine again. So apparently it works well with deconz too.

samnewman86, 2021-03-01:
Must be the way I'm pressing it then 🤔

ebaauw, 2021-03-01:
It takes about 6 seconds of holding for the dimmer to switch modes.  When you see the 1001 buttonevent, you know it has.

samnewman86, 2021-03-01:
I know, I use for an "All Off" event, that turns off lights, audio, heating
etc when I leave a room permanently. The trouble is I can no longer long
press the left and right buttons

samnewman86, 2021-03-01:
In the original IKEA ecosystem, that action is used to reset lights to all
on 100%. I'm not sure why it would effect the left and right buttons.

ebaauw, 2021-03-01:
Me neither.  Unfortunately, I cannot sniff the interaction with the IKEA hub.  I've got no way to retrieve the network key, as the hub doesn't support regular pairing.

closed 2021-04-03 by github-actions[bot]

Claustn, 2021-04-28:
Sorry to revive an old thread, I just bought 10 of the IKEA  5 button remotes, and have this exact issue... What triggers it for me is after pairing it in Phoscon the GUI prompts you to hold down the middle button for up to 10 seconds, which triggers this behavior. I have had some of them lying around for more than 1 week, and the only thing that help is taking out the battery and putting it back in.

bcutter, 2021-06-24:
Sooner or later this happens again and again. This behaviour basically cuts down the number of usable button events from theoretical 12 to 9. No big deal but those buttons affected are the ones being used a lot, at least in my case - if they would work. In my case those won´t work:
- middle long press
- left long press
- right long press

Can someone please reopen this issue?!?

manup, 2021-06-24:
Are you using the latest version? We had fixes for this problem in [v2.12.1-beta](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.12.1-beta)

bcutter, 2021-06-24:
Latest stable (2.11.5).