dresden-elektronik/deconz-rest-plugin #3757: Tuya Cover - Using wrong logic for reporting "Current Position Lift Percentage"

cobirnm, 2020-11-25:
## Describe the bug

I have several tuya cover switch. It seems that the implementaiton of tuya cover devices is reporting the attribute 0x0008 "Current Position Lift Percentage" using the scale "% open". It seems that Deconz uses for most of it's cover the scale "% close".
A great job has been done implementing these tuya devices (cover was added in #https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3134)
I spooted this issue in Home assistant in https://github.com/home-assistant/core/issues/40627

## Steps to reproduce the behavior

## Expected behavior

@Smanar do you thing that the Deconz logic for "Current Position Lift Percentage" could be applied for tuya covers? I mean start using the scale "% closed". 
This could be done by applying something in the code like "Current Position Lift Percentage" =100-(Present Current Position Lift Percentage)

Both me and @Kane610 have made extensive testing on this under #https://github.com/home-assistant/core/issues/40627

Do you think it's possible to fix it?
Thank you in advance
## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2020-11-26:
>do you thing that the Deconz logic for "Current Position Lift Percentage" could be applied for tuya covers? I mean start using the scale "% closed".

It couldn't, it need, ebaauw logic is now a rule ^^.

But the rule is for exemple

```
on= true
bri = 255
open = false
lift = 100%
shutter closed
```

And from your log (from the other issue) you have
```
{"bri":254,"lift":100,"on":true,"open":false,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:f7:18-01"}
{"bri":0,"lift":0,"on":false,"open":true,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:f7:18-01"}
```

So all seem fine ?

cobirnm, 2020-11-26:
What I have is the other way round:

100 is fully open
0 is fully closed

cobirnm, 2020-11-26:
Just to be clear When I have this in deconz the cover is actually fully open:

[image]

If I execute the command in deconz to open the cover it goes to 100% as in the picture above. 

To close it's the other way round (0%) (idon't have a picture right now).

For me this logic was right until I found out of the rule in Deconz! (% closed) This was the reason I created an issue in Home Assistant since these were my first zigbee cover devices!

Smanar, 2020-11-26:
Yep, but values you are seing in deconz are not the values gived by the api. HA don't use values from deconz but values from the API.

Do you have a json returned by deconz with bad values, or good values but the covering device in bad state ?

cobirnm, 2020-11-26:
when I execute Up/open I get the following: (cover fully open)

`{"e":"changed","id":"17","r":"lights","state":{"bri":254,"lift":100,"on":true,"open":false,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:f7:18-01"}`

When executing down/Close: (cover fully closed)

`{"e":"changed","id":"17","r":"lights","state":{"bri":0,"lift":0,"on":false,"open":true,"reachable":true},"t":"event","uniqueid":"ec:1b:bd:ff:fe:32:f7:18-01"}`

Smanar, 2020-11-27:
So there is a problem yes.
Your model id  is "TS130F" ? There is no hack for this one, so it s realy the pure zigbee protocol.
Are you sure the problem is not from the setting, remember you have a command to reserve the side, or just the wiring, have you tried to reverse the brown and the black wire ?

cobirnm, 2020-11-29:
Yes my model is TS130F. 
I have been playing with the reverse command. When I set it for the first 2 times the cover moves it works as it should. Then I loose the calibration and everything goes back to the starting point...

So my findings are:
1) trun on  motor reversal
the buttons on the switch do not reverse (they stay the same)
the position in deconz is correct (before was 100, after is zero)
The position on Home assistant is correct.

2) move the cover down executing down command. (to fully closed) 
the cover goes down
the position reports 100 (this is ok) (in deconz)

3) move the cover up
the cover opens fully 
the position reports 0  (this is ok) (in deconz)

4) move the cover down (in deconz)
the cover goes down by more or less 30 % and reports 0 (it looks line it makes some kind of reset on the switch)

5) move up ( deconz)
If goes up and shows 100%

6) move down
Again it does not go down  (not calibrated) and shows 0%

P.S. I have tried also using tuya hub for setting the reversal  and calibration. When I re-add the switch to deconz there is no difference in the reposting of the position.
Is it possible to correct the position in the code by adding some kind of combination of manufacture/moddel? I have 3 of these switches and they all behave like this
[image]

cobirnm, 2020-11-29:
I know I'm always insisting with this issue but can't we have something like we already have in window_covering.cpp for Xiaomi (please note that I'm not a coder)

```
                // Reverse value for Xiaomi curtain 
                if (lightNode->modelId().startsWith(QLatin1String("lumi.curtain")) || 
                   (lightNode->modelId() == QLatin1String("Motor Controller")) )
                {
                    lift = 100 - lift;
                }
```

Smanar, 2020-11-29:
Yep it s possible but we realy need to be sure the device is realy reversed, else it break the configuration on other device

>I have been playing with the reverse command. When I set it for the first 2 times the cover moves it works as it should. Then I loose the calibration and everything goes back to the starting point

Ok, nice issue, and it don't happen when you use the original gateway ?

>I have tried also using tuya hub for setting the reversal and calibration. When I re-add the switch to deconz there is no difference in the reposting of the position.

This is normal, the setting are reset when you change the gateway

>Regarding the wiring if I reverse it the open button will close the cover and the close button will open the cover.

Ok so not the wiring.

Can try this code

```
sudo apt install deconz-dev
git clone --branch tuya_cov https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

cobirnm, 2020-11-29:
>Ok, nice issue, and it don't happen when you use the original gateway ?

No it does not happen with the original gateway

>This is normal, the setting are reset when you change the gateway

That is not really true.  The settings that I make using the tuya gateway stay in the device when I pair it with deconz. Originaly I used this to calibrate the device and then use it with deconz.

>Can try this code

I will try the code but noobie question: How can I unistall it after testing?

cobirnm, 2020-11-29:
I get an error when compiling:

```
rest_lights.cpp: In member function ‘int DeRestPluginPrivate::setWindowCoveringS                       tate(const ApiRequest&, ApiResponse&, TaskItem&, QVariantMap&)’:
rest_lights.cpp:1710:32: error: expected unqualified-id before ‘->’ token
            (taskRef.lightNode->->manufacturer() == QLatin1String("_TZ3000_egq7y                       6pr")) ||
                                ^~
Makefile.Release:1058: recipe for target 'release/rest_lights.o' failed
make[1]: *** [release/rest_lights.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/pi/deconz-rest-plugin'
Makefile:38: recipe for target 'release' failed
make: *** [release] Error 2
```

Smanar, 2020-11-29:
Corrected, sorry typo

cobirnm, 2020-11-30:
First of all sorry for the late reply. I had to learn how to install all of this! (I did manage to install the dev on docker container!)

Your code works like a charm! Now everything is working as it should with standard home assistant deconz integration!
Do you think the can go to the next release of Deconz, or should it be tested by someone else? (I tried to find what you have changed but wsa not able to do it!)

Smanar, 2020-11-30:
Bad luck, the last date for PR was today, sorry.
I will make the PR now, but it will be surely in next version, the Beta in 15 days now.

And yes It s model specific, so I don't think we need to wait for another return.

cobirnm, 2020-11-30:
I will skip next release until this PR is released! 
I can only thank you for all your hard work!

Smanar, 2020-11-30:
Lol, you do more work than me.
It make more work to test the code for you than editing it for me.

closed 2020-12-07 by manup

axellebot, 2020-12-08:
I don't know where to ask this. Is there a way to edit the travel time of the cover of TS130F ?
[image]

Smanar, 2020-12-08:
It s a device without level check that use timed move to simulated a set level command ?
You have nothing for calibration ?

axellebot, 2020-12-08:
From what I know it's possible to setup time travel through the official (tuya/zemismart/whatever) App. But I don't wanna buy a gateway to set it up. Looks like there is attributes for this purpose but only readable according to deconz (and some time not even readable) :/

Smanar, 2020-12-08:
Your device use tuya cluster or covering cluster ?

cobirnm, 2020-12-08:
You need to edit geral.xml to make te attribute RW (read and right). In a couple of hours I can give you the details (I need to be on a computer)

axellebot, 2020-12-08:
Putting info here :

<details>
  <summary>From map :</summary>
  
[image]
</details>

<details>
  <summary>From Basic section :</summary>
[image]
</details>

<details>
  <summary>From Widnows covering section :</summary>
  
[image]

</details>

> You need to edit geral.xml to make te attribute RW (read and right). In a couple of hours I can give you the details (I need to be on a computer)

FYI : I'm accessing Deconz through Home-assistant

Smanar, 2020-12-08:
Take a look in the cluster 0x0102, you wil find "calibration", but he is right on some version this line is read only.

cobirnm, 2020-12-08:
@axellebot What you are looking for is attribute 0xF001. If in the access colunm you have RW as in the picture below you are  good to go.

[image]

If you don't have RW then you need to edit general.xml file as in the code: (change on line 3 bellow from `access="r"`  to `access="rw"`.

```
        </attribute-set>
        <attribute-set id="0xf001" description="Tuya Window Covering Setting">
          <attribute id="0xf001" name="Calibration" type="enum8" default="0" required="m" access="rw">
            <value name="Start" value="0"></value>
            <value name="End" value="1"></value>
          </attribute>
          <attribute id="0xf002" name="Motor Reversal" type="enum8" default="0" required="m" access="rw">
            <value name="Off" value="0"></value>
            <value name="On" value="1"></value>
          </attribute>
        </attribute-set>
```

Now for calibration:

1) under deconz double click the 0xF001 and select "start" from the dropdown list and press "write" button
Now your cover should enter in calibration mode
[image]

2) on the cover switch press the button to fully open the cover (on my stwich it's the left button)
3) when the cover is fully open press the "stop" button (on my switch it's the midle button)
4) press the close button and wait for the cover to fully close (on my switch it's right button)
5) As soon as the cover is fully closed press the "stop" button
6) Go to deconz and change from "start" to  "end" and press "write" button
[image]

That's it , your cover should be calibrated and ready to use!

P.S. I have noticed that my covers some times do not open fully. This happens after setting different positions (20%, 35%, 80%....). To solve this I fully close the cover and then the positions are correct again!

TimSoethout, 2020-12-08:
Maybe this is not the place, but is it possible to change `general.xml` in the "hosted" deconz on Home Assistant Core (https://github.com/home-assistant/core/tree/dev/homeassistant/components/deconz)?
My other option is to wait until the updated deconz version is released for Home Assistant Core.

axellebot, 2020-12-08:
I don't really understand why I don't have the same attributes, also no "tuya" section and they are gray 😞 
I guess I can do the same trick with "limit" attributes but seems also broken.

<details>
  <summary>From "Window covering setup" section :</summary>

[image]
[image]

</details>

> Maybe this is not the place, but is it possible to change `general.xml` in the "hosted" deconz on Home Assistant Core?
> My other option is to wait until the updated deconz version is released for Home Assistant Core.

I'm able to edit the `xml` file with docker's shell (after installing `vi`).

cobirnm, 2020-12-08:
You don’t have manufacture name, only model. Maybe @Smanar can help with this.

axellebot, 2020-12-08:
<details>
  <summary>From HA Zigbee integration : </summary>

```json
{
  "node_descriptor": "NodeDescriptor(byte1=1, byte2=64, mac_capability_flags=142, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=0)",
  "endpoints": {
    "1": {
      "profile_id": 260,
      "device_type": "0x0202",
      "in_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0102"
      ],
      "out_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_ltiqubue",
  "model": "TS130F",
  "class": "zigpy.device.Device"
}
```

</details>

<details>
  <summary>From Deconz  :</summary>

| MAC | NWK | Nane | Model | Vendor | Version |
|------|--------|-------|--------|----------|---------|
| MAC@    | 0x6efe | Windows Covering | TS130F | Heiman |  |

</details>

Smanar, 2020-12-09:
It s in the cluster "covering" the 0x0102.

If you have not the "tuya windows covering setting" it mean you have an old deconz version.

If the attribute are grayed it mean your device don't support them, so hopping for they are not grayed for yours ^^
Else you can too take a  look in the device manual, if you need a zigbee remote or 433 mhz one.

axellebot, 2020-12-10:
I updated Deconz plugin from HA and I'm now able to calibrate it but I have 2 main issues :
- Calibration is based on closing time travel. And because I'm using "roller shutter" the time travel is slower while going "up" than going "down". For now I based my calibration on the longest travel time using a simple stopwatch but I hope there is a way to calibrate both way.
- HA is not updated properly : still displaying "up arrow" after being fully opened

Smanar, 2020-12-10:
I realy need to be sure the device is reversed, so same question than in this issue.
Pls check wire, JSON, using command direclty in deconz, , the zigbee command to set the direction (just close the command you have used for calibration) I need too the model ID/Manufacture name

cobirnm, 2020-12-10:
It looks like the same issue I reported. If that is the case then the correction will be only available in next deconz version

Smanar, 2020-12-10:
@cobirnm for you it s already available, no ?

cobirnm, 2020-12-10:
I have it because I compiled the code you provided

axellebot, 2020-12-11:
The behavior using `deCONZ 6.6.0` with HA plugin :
- Up/Open : Opening
- Down/Close : Closing
- Lift Percentage :
  - 100% : Fully opened
  - 0% : Fully closed

Smanar, 2020-12-11:
No if lift = 100 % the shutter need to be closed

```
on= true
bri = 255
open = false
lift = 100%
shutter closed
```

axellebot, 2020-12-11:
> No if lift = 100 % the shutter need to be closed

That's not what is happening with `deCONZ 6.6.0` from HA plugin.

Is there a way to calibrate "opening travel time" ?

cobirnm, 2020-12-12:
The travel time is calibrated on the switch itself. If you state that the cover takes longer to go up than to go down then on step 5 of the procedure I sent you, then you can press the stop button some seconds after the cover stops moving.
On a side note: if your motor does not take the same time to go up and to go down, it looks to me that you don’t have an enough powerful motor and you should replace it!

axellebot, 2020-12-12:
That's what I have done. I calibrated travel down time based on travel up time.

mathwern, 2020-12-28:
I believe I have exactly the same issue but with a different manufacturer (also Tuya): "_TZ3000_vd43bbfq"

The messages when the shutter is fully open:
id: "13"
lastannounced: null
lastseen: "2020-12-28T18:43Z"
manufacturername: "_TZ3000_vd43bbfq"
modelid: "TS130F"
name: "whatever"
swversion: null
type: "Window covering device"
uniqueid: "5c:02:72:ff:fe:bb:e3:a8-01"
e: "changed"
id: "13"
r: "lights"
t: "event"
uniqueid: "5c:02:72:ff:fe:bb:e3:a8-01"

e: "changed"
id: "13"
r: "lights"
state: object
bri: 254
lift: 100
on: true
open: false
reachable: true
t: "event"
uniqueid: "5c:02:72:ff:fe:bb:e3:a8-01"

And the message when fully closed:
e: "changed"
id: "13"
r: "lights"
state: object
bri: 0
lift: 0
on: false
open: true
reachable: true
t: "event"
uniqueid: "5c:02:72:ff:fe:bb:e3:a8-01"

The arrows up/down in HA trigger the correct movement, but the status inverted. Would it maybe be possible to add an option in  deconz to invert the status (something like invert_cover, similarly to motor reversal).

Smanar, 2020-12-28:
>The arrows up/down in HA trigger the correct movement, but the status inverted. Would it maybe be possible to add an option in deconz to invert the status (something like invert_cover, similarly to motor reversal).

There is one in the API
` PUT, PATCH /api/<apikey>/lights/<id> with {"reverse":true}`

Can you try it, using the API or deconz itself ? To be sure it s reversed in the device itself (But remember it will reverse too the manual mode.)

If it don't work I will make the PR for your device too.

mathwern, 2020-12-28:
Reversing the motor doesn't help, I've tried (then up is down and down is up; the open/close status is OK, but the arrows aren't)

Smanar, 2020-12-29:
Ok so can you try this code ? (if you have unix machine ?)

```
sudo apt install deconz-dev
git clone --branch tuya_cov2 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

mathwern, 2020-12-30:
I will try, I'm not too sure how this plays with the Docker installation.
Is it enough to replace the libde_rest_plugin.so in the various folders ./var/lib/docker/.../usr/share/deCONZ/plugins?

Smanar, 2020-12-30:
I don't think it s possible with docker. You need to compile and install on the same (virtual) machine.
I never see someone succed with that.

cobirnm, 2020-12-30:
It’s possible to do it on docket. I did it but to be honest I lost almost 3 hours installing dependencies just to be able to compile.

mathwern, 2021-01-03:
Test OK!

Sorry, took me forever to get a testing environment up.
Your fix seems fine, fully opening the cover in HA gives me lift: 0 / open: true and fully closing gives lift: 100 / open: false

Smanar, 2021-01-03:
Nice thx, the PR is submitted.

bastshoes, 2021-01-07:
Hello @Smanar! Today I've recived switch with slightly different manufacturename - "_TZ3000_xzqbrqk1".
It's operational, but reports inverted state:
```
{
    "etag": "764fa3a33fc0f76a34ff105416aa8f9b",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-01-07T19:01Z",
    "manufacturername": "_TZ3000_xzqbrqk1",
    "modelid": "TS130F",
    "name": "Window covering device 5",
    "state": {
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:ca:a1:9b-01"
}
```
As you see lift and open attributes are correct but "on" attribute is inverted.
Could you please include this model also in your fix?

Smanar, 2021-01-08:
"on" can be the opposite of "open".
The shutter is open or closed (full) with this json ?

bastshoes, 2021-01-09:
Fully closed:
```
{
    "etag": "f87afb7a1f6b8f6dd123917c311ff40f",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-01-09T11:55Z",
    "manufacturername": "_TZ3000_xzqbrqk1",
    "modelid": "TS130F",
    "name": "Window covering device 5",
    "state": {
        "bri": 0,
        "lift": 0,
        "on": false,
        "open": true,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:ca:a1:9b-01"
}
```
Fully opened:
```
{
    "etag": "bc96261da15da44c98525e05912f4dc3",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2021-01-09T11:56Z",
    "manufacturername": "_TZ3000_xzqbrqk1",
    "modelid": "TS130F",
    "name": "Window covering device 5",
    "state": {
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
    "swversion": null,
    "type": "Window covering device",
    "uniqueid": "5c:02:72:ff:fe:ca:a1:9b-01"
}
```

Smanar, 2021-01-09:
Added https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4089

mathwern, 2021-01-09:
Wouldn't it make sense to add the option to the cover device in deCONZ to allow changing the behaviour without hard coding each manufacturer name (like for motor reversal)?
Tuya seem to come up with a new one every other week :/

Smanar, 2021-01-09:
lol, yeah boring.

There is already a tuya command to reverse motor side, usable with the api too, but some user have issue with it, it s realy hardware dependent, it s for that I generaly ask user to test with the command first https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3757#issuecomment-734918989

Smanar, 2021-02-06:
@bastshoes do you have the manufacture and a product reference for your device ? (_TZ3000_xzqbrqk1)

bastshoes, 2021-02-09:
Hello @Smanar!

First of all thank you fixing issue with my model. Now it works fine.

Regarding product itself. This is Lonsoho Zigbee curtain switch - https://aliexpress.ru/item/4001095065015.html?spm=a2g0s.9042311.0.0.264d33edvi02BK&sku_id=10000014381513687

And here is screeshot of basic cluster
[image]

Smanar, 2021-02-09:
Nice, thx a lot ^^
It s because deconz will use product name later (easier to reconize device) so I need to translate all TZ3000_XXXXXX in product name.

ch1spas2, 2021-02-13:
Hello,
I have the same issue, with this model _TZ3000_ltiqubue (https://www.aliexpress.com/snapshot/0.html?orderId=3009066202103412) and the motor reversal doesn't solve it.
There is my screenshot of the basic cluster:
[image]
Thank you so much for your time I apreciate it.

Smanar, 2021-02-13:
Hello, can you just check if you have same value or exactly the reverse for all ?

```
    "state": {
        "bri": 254,
        "lift": 100,
        "on": true,
        "open": false,
        "reachable": true
    },
```

ch1spas2, 2021-02-14:
I don't know how to retrive this info but I can give you this one, maybe it can be as useful for you.
[image]
[image]

[43 more replies]