dresden-elektronik/deconz-rest-plugin #3761: Device name

Ajowan84, 2020-11-26:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

Ajowan84, 2020-11-26:
Hi,
This switch is detected correctly but like a light.

Product name : MoesGo ZigBee Smart Light Switch
Manufacturer : MoesGo
Model : DS-112
Device type : Switch

[image]

[image]

Smanar, 2020-11-27:
Wired switches are light, battery switches are sensors.

It don't work correclty too ?

Ajowan84, 2020-11-27:
Yes, it work correctly in my home automation app.
But in Phoscon app, with a wired switch as light i can't associate this switch with a light in a group.

Configuring groups in Phoscon is very useful, it allows you to continue to control the lights if the home automation application down.

Smanar, 2020-11-27:
>i can't associate this switch with a light in a group.

Normal, you can't ^^
You can associate it like a bulb, not a switch, this device receive command, not send them.

Ajowan84, 2020-11-27:
It's a switch not a bulb !
It can send commands when I press one of the buttons on the switch

Smanar, 2020-11-27:
It send command at wich one device ?
This device receive command to open / close / dim the output, like a bulb.

Ajowan84, 2020-11-28:
It send command in my home automation app.
In my home automation app, i can command bulb, module...
In Phoscon, i can't do it because the switch is not detected as a switch but as a light bulb ...

Smanar, 2020-11-28:
It send command or notification ? It send a zigbee command to your home automation system, or it s deconz that intercept the zigbee notification and send it using http to your home automation ?
Can you send the "on" command or the the "set level" command to a battery device switch ?
Can you set them to a bulb ? then at your wired switch ? So it s a bulb.

But it need to work too in phoscon, nothing is changing when you turn it on/off in your home automation ?

Ajowan84, 2020-11-28:
Thank you very much for your help Smanar.
But i don't understand why you say that device is a bulb.
It's a wall switch !!!
Look at the pictures on my first post...

Smanar, 2020-11-29:
It's managed like a light, from zigbee side.
Deconz know only 2 big family "sensor" and "light", to make short "sensor" send order, and "light" reveive order.

With a battery switch you can send order to a buld or a wired switch to turn on the light. The reverse is not possible

Battery switches are "sensor"
Wired switches are "light"

Sonoff relay are "light" too, even they haven't light.

Ajowan84, 2020-11-29:
If the wireds switches are considered as bulbs in Phoscon, impossible to associate them in a group with bulbs to drive them...

Ajowan84, 2020-11-29:
If the wired switches are considered as bulbs in Phoscon, impossible to associate them in a group with real bulbs to drive them...
I don't understand why because it's a wall switch.
But if it's managed like a light, from zigbee side, i will not use it as a rescue in Phoscon.

Smanar, 2020-11-29:
>If the wired switches are considered as bulbs in Phoscon, impossible to associate them in a group with real bulbs to drive them...
I don't understand why because it's a wall switch.

You are right, but tuya stuff generaly don't use the standard cluster for on/off the 0x0006 but a Tuya cluster the 0xEF00.
And the second one can 't work in group.

I can't be sure, because the cluster 0xEF00 can be invisible, but as you have the basic one 0x0000 on the 3 endpoint, I think it s fake endpoint, so this device can't work in zigbee group.

To be sure just use the cluster 0x0006 with deconz, it will not work if your device use tuya cluster.

Ajowan84, 2020-12-13:
Thank you again Smanar for this information

closed 2020-12-13 by Ajowan84