dresden-elektronik/deconz-rest-plugin #3762: Lidl Zigbee devices

canedje, 2020-11-27:
## Device

- Product name: different Zigbee devices from lidl, like wallplug and Smart Motion sensor.
- Manufacturer: SilverCrest.
- Model identifier: SmartPlug, Smart Motion Sensor.
- Device type : Please remove all unrelated device types. 
  - Light
  - Sensor (PIR)
  - Switch
  - Other: 

## Screenshots

Mimiix, 2020-11-27:
Missing screenshots

canedje, 2020-11-27:
I did add the link to the lidl page with all devices. Is that not good enough?

Mimiix, 2020-11-27:
Basic cluster still missing ^^

wovka88, 2020-11-27:
The Lidl Motion Sensor should be the Heiman Motion Sensor: http://www.heimantech.com/product/?type=detail&id=6

canedje, 2020-11-27:
I did find out that the Walplug is type TS011F from Heiman and is available in the GUI of the Deconz/Phoscon
So that is added already.
Tjhe Smart Motion Sensor is not.
I will try to find the basic cluster and add over here

ebaauw, 2020-11-27:
I accidentally walked into a LIDL today, and came home with a Gateway, a X-Mas lightstrip, a set of LED spotlights, and a smart USB extension cord.

The gateway runs on Zigbee channel 15, as now does my sniffer.  The LIDL app requires e-mail registration and uses some cloud service.  Need to ditch that as quickly as possible.  There doesn't seem to be a way to find out or change the Zigbee channel.

The spotlights are regular GU10 spots, with luminaries holding them.  You need to provide your own electricity wires.  Controlling them from the LIDL app, the gateway seems to send regular _On/Off_, _Level Control_, and _Color Control_ commands, but also some _Color Control_ commands I haven't seen before, which aren't recognised by Wireshark either.  They don't have the _Manufacturer Specific_ bit set in the ZCL header, though.  The spot replies with a _Report Attributes_.  I haven't seen any binding or attribute reporting configuration commands, though.  
As I would have hoped, the remaining time uses _On with Timed Off_.  I suspect the schedules are implemented on the gateway.

The USB extension cord seems to have three endpoints with _On/Off_ cluster.  When controlling all outlets at once, the app sends three _On/Off_ commands, one for each endpoint.  Each endpoint responds with a _Report Attributes_ as well.  When controlling the extension cord manually, each outlet sends an _Attribute Report_, but now the gateway replies with a _Default Response_.  I don't see anything weird here.

The X-Mas lightstrip consists of 200 LEDs, which seem to be individually addressable at least from the controller.  I think LEDs are chained, connecting the data in of the next LED to the data out of the previous LED, like the Neopixel or Pimoroni Blinkt! (see Homebridge RPi).  When setting the strip to a single colour, the new colour "travels through" the strip.  I'm not sure of this detailed level of control is exposed over Zigbee, though.  
The strip seems to be controlled from cluster 0xEF00, again no _Manufacturer Specific_ bit.  Any change (be it white, colour or scene) results in a 0x00 command, followed by one or more 0x01 commands.  The commands have a variable payload, up to 56 bytes.  Even combined, the payloads doesn't seem to be large enough to hold the RGB and brightness values per LED.  Presumably, the controller uses the payload to generate the serial signal.

Mimiix, 2020-11-27:
>I accidentally walked into a LIDL today. 

No one does that accidentally 😂

ebaauw, 2020-11-27:
The extension cord looks reasonable, and more or less what we expected.  _Groups_ and _Scenes_ clusters per endpoint.  Reports capacity for 50 scenes, probably shared between the endpoints - need to test.  Doesn't report group capacity.
[image]

_On/Off_ clusters don't support _PowerOn On/Off_.  No _Level Control_, so (presumably) doesn't accept _Go to Level with On/Off_, limiting direct control by controllers that send that command.  I have no idea what _Light mode_ means or does.  @Smanar, would you know?
[image]

The _Basic_ clusters on the 2nd and 3rd endpoints don't seem to be implemented: all attributes report as not supported.  REST API doesn't like that and reports `null` for `manufacturer name` and `modelid`.  I don't kwow what `powerup` means or does.  According to the code, that should only be exposed for dresden devices?  @manup?

```
  "5": {
    "etag": "81535d186187f9aaade083d3ca14340c",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-11-27T14:40Z",
    "manufacturername": "_TZ3000_1obwwnmq",
    "modelid": "TS011F",
    "name": "On/Off plug-in unit 5",
    "powerup": 7,
    "state": {
      "alert": "none",
      "on": true,
      "reachable": true
    },
    "swversion": null,
    "type": "On/Off plug-in unit",
    "uniqueid": "ec:1b:bd:ff:fe:b6:2a:0b-01"
  },
  "10": {
    "etag": "8102d22f7ee0f9440dd298680e85884e",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-11-27T14:40Z",
    "manufacturername": "Unknown",
    "modelid": null,
    "name": "On/Off plug-in unit 10",
    "powerup": 7,
    "state": {
      "alert": "none",
      "on": true,
      "reachable": true
    },
    "swversion": null,
    "type": "On/Off plug-in unit",
    "uniqueid": "ec:1b:bd:ff:fe:b6:2a:0b-02"
  },
  "11": {
    "etag": "cde6ef3b28d7385fa393da998b9d13bb",
    "hascolor": false,
    "lastannounced": null,
    "lastseen": "2020-11-27T14:39Z",
    "manufacturername": "Unknown",
    "modelid": null,
    "name": "On/Off plug-in unit 11",
    "powerup": 7,
    "state": {
      "alert": "none",
      "on": true,
      "reachable": true
    },
    "swversion": null,
    "type": "On/Off plug-in unit",
    "uniqueid": "ec:1b:bd:ff:fe:b6:2a:0b-03"
  }
```

manup, 2020-11-27:
Indeed the `powerup` is only used by FLS-PP lp, seen that the other day as well, it should only be exposed for FLS.
However I've found this:
```
        case VENDOR_DDEL:
            return prefix == deMacPrefix ||
                   prefix == silabs3MacPrefix;    // 
```

I'll make the check stricter so it doesn't include non DE devices and checks for the actual manufacturer code of node descriptor.

@Smanar does the manufacturer name also encode the modelid?

For all the weird Tuya modelids and manufacturer names I suggest we use JSON files which maps them to sane strings perhaps also modelids.

**Example lookup**

*Pseudocode*
```c++
// returns a sane manufacturer name, or the original value if none is found
QString lookupManufacturerName(QString basicClusterManufacturerName, QMap manufacturerLookup);

lookupManufacturerName("_TZ3000...", mnMap); // ---> "Tuya"
```

Smanar, 2020-11-27:
Yep, it s my project for future beta, bored to add lines like that every 2 days :
```
    { VENDOR_EMBER, "TS0302", silabs3MacPrefix }, // Tuya curtain switch
    { VENDOR_EMBER, "TS0041", silabs3MacPrefix }, // Tuya wireless switch
    { VENDOR_EMBER, "TS0041", silabs5MacPrefix }, // Tuya wireless switch
    { VENDOR_EMBER, "TS0042", silabs3MacPrefix }, // Tuya wireless switch
    { VENDOR_EMBER, "TS0043", silabs3MacPrefix }, // Tuya wireless switch
    { VENDOR_EMBER, "TS0043", silabs8MacPrefix }, // Tuya wireless switch
    { VENDOR_EMBER, "TS0044", silabs9MacPrefix }, // Tuya wireless switch
```

I will make a fonction that check all vendor, all mac adress, and special model id.

>@Smanar does the manufacturer name also encode the modelid?
There is different hardware for tuya device, for exemple https://docs.tuya.com/en/iot/device-development/module/zigbee-module/zigbeetyzs11module?id=K989rik5nkhez

Some hardware have a real model id, other have just TSXXXX, for the second one the manufacture name is <HARDWARECART>_<X><MODELID>

Smanar, 2020-11-27:
> The Basic clusters on the 2nd and 3rd endpoints don't seem to be implemented: all attributes report as not supported. REST API doesn't like that and reports null for manufacturer name and modelid. I [...]

@ebaauw I think this device have only 1 endpoint, the 2 other are the fake endpoint created for tuya switches.

ebaauw, 2020-11-27:
> I think this device have only 1 endpoint, the 2 other are the fake endpoint created for tuya switches.

Good call, @Smanar!  The endpoints are real alright, but the _Basic_ clusters on endpoints 0x02 and 0x03 aren't.

What makes the code treat this as a Tuya device?  Other than the weird values for _Manufacturer Name_ and _Model Identifier_ there's nothing Tuya about this device.

```
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x677b
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x677b
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
ZigBee Device Profile, Active Endpoint Response, Nwk Addr: 0x677b, Status: Success
    Sequence Number: 77
    Status: Success (0)
    Nwk Addr of Interest: 0x677b
    Endpoint Count: 4
    Active Endpoint List
        Endpoint: 1
        Endpoint: 2
        Endpoint: 3
        Endpoint: 242
```
```
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x677b
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x677b
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
ZigBee Device Profile, Simple Descriptor Response, Nwk Addr: 0x677b, Status: Success
    Sequence Number: 42
    Status: Success (0)
    Nwk Addr of Interest: 0x677b
    Simple Descriptor Length: 22
    Simple Descriptor
        Endpoint: 1
        Profile: Home Automation (0x0104)
        Application Device: On/Off plug-in unit (0x010a)
        Application Version: 0x0001
        Input Cluster Count: 5
        Input Cluster List
            Input Cluster: Basic (0x0000)
            Input Cluster: Identify (0x0003)
            Input Cluster: Groups (0x0004)
            Input Cluster: Scenes (0x0005)
            Input Cluster: On/Off (0x0006)
        Output Cluster Count: 2
        Output Cluster List
            Output Cluster: OTA Upgrade (0x0019)
            Output Cluster: Time (0x000a)
```
```
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x677b
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x677b
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
ZigBee Device Profile, Simple Descriptor Response, Nwk Addr: 0x677b, Status: Success
    Sequence Number: 43
    Status: Success (0)
    Nwk Addr of Interest: 0x677b
    Simple Descriptor Length: 16
    Simple Descriptor
        Endpoint: 2
        Profile: Home Automation (0x0104)
        Application Device: On/Off plug-in unit (0x010a)
        Application Version: 0x0001
        Input Cluster Count: 4
        Input Cluster List
            Input Cluster: Identify (0x0003)
            Input Cluster: Groups (0x0004)
            Input Cluster: Scenes (0x0005)
            Input Cluster: On/Off (0x0006)
        Output Cluster Count: 0
```
```
IEEE 802.15.4 Data, Dst: 0x0000, Src: 0x677b
ZigBee Network Layer Data, Dst: 0x0000, Src: 0x677b
ZigBee Application Support Layer Data, Dst Endpt: 0, Src Endpt: 0
ZigBee Device Profile, Simple Descriptor Response, Nwk Addr: 0x677b, Status: Success
    Sequence Number: 44
    Status: Success (0)
    Nwk Addr of Interest: 0x677b
    Simple Descriptor Length: 16
    Simple Descriptor
        Endpoint: 3
        Profile: Home Automation (0x0104)
        Application Device: On/Off plug-in unit (0x010a)
        Application Version: 0x0001
        Input Cluster Count: 4
        Input Cluster List
            Input Cluster: Identify (0x0003)
            Input Cluster: Groups (0x0004)
            Input Cluster: Scenes (0x0005)
            Input Cluster: On/Off (0x0006)
        Output Cluster Count: 0
```

ebaauw, 2020-11-27:
For completeness:
[image]

zladukas, 2020-11-28:
> Indeed the `powerup` is only used by FLS-PP lp, seen that the other day as well, it should only be exposed for FLS.

@manup and there are possible values for FLS-PP lp, defined in resource.h, used to restore last known state after power loss?

Smanar, 2020-11-28:
> What makes the code treat this as a Tuya device? Other than the weird values for _Manufacturer Name_ and _Model Identifier_ there's nothing Tuya about this device.

Welcome in tuya world ^^

It's a long story, to make short "lot of tuya device" have exactly the same information during pairing
- Device id = DEV_ID_SMART_PLUG
- Manufacture number = None or Ember
- Same MAC address zoo
- Same clusters

And it can be for switch, bulb but covering or thermostat too. It realy not possible to reconize a device using the actual code.

The only way to reconize tuya device is the manufacture name, it will be possible later with the new descriptor code made by manup (it use json file), but he want to change something again, and not enought time to take a look on it yet.

So ATM in the same fonction, you will see below a "blacklist" and "whitelist".
Tuya device need 2 inclusions, the first one retreive informations on the device (like manufacture name) and can create badly entry. The second one look in the database the good missing information and using this blacklist/whitelist, create the good entry.

hbkoenig, 2020-12-01:
thanks to the deconz 2.07.00 (beta) changelog I've learned about the LIDL smart home products (on sale in german stors starting wednesday Dec 3rd).

they also offer RGB light bulbs (both E27 and E14) for 14,95 EUR, but I can't find these mentioned in this issue.

are they already supported ?  if not, any chance/plans for support RSN ?
if it helps I'm willing to buy some bulbs day after tomorrow and provide all needed infos...

hbkoenig, 2020-12-01:
link for the germen LIDL offers this week (I'm not working for or related with LIDL;)
https://www.lidl.de/de/home-smart-home-ab-03-12/c29863

Smanar, 2020-12-01:
Yep, can be usefull, I have never see a tuya bulb.

canedje, 2020-12-01:
I just updated to 2.07.00
I did try to pair my Lidl motion sensor.
It gives a ready sign in the phoscon app GUI (green balk with ready) but is not showing up as device/ sensor in the GUI.
In VNC and HA I do see a binary sensor showing up.
It is only not reacting on movement.

Smanar, 2020-12-01:
The device is not visible in phoscon, you need to check direclty the api or using third app.
But strange it don't owrk on HA, all values are good on your device.

As you have deconz, you can see something special in log ?
Have you tried to re-include it ?

canedje, 2020-12-01:
I did re include several times with all the same results.
I also did restart HA. No effect.
The log is showing no errors or no warnings

Mimiix, 2020-12-01:
I have one too and it works with the latest deconz. It won't show in phoscon tho.

Smanar, 2020-12-01:
IDK what happen, perhaps different versions, will check tommorow

SwoopX, 2020-12-01:
The sensor might not be enrolled (under IAS zone cluster). Double check this pushing the physical button on the device once (just like the start of a reset, 1 sec) and then immediately read the IAS zone cluster's attributes.

Smanar, 2020-12-02:
But why it work fine on one user, and not for an other ?
This part use "waiting list" ?
I realy don't see something special.

Have you tried with making the device awake during inclusion, like for xiaomi.

canedje, 2020-12-02:
yes
I tried everything
I can imagine

Smanar, 2020-12-02:
Can you retreive all logs during inclusion ?
But it seem you have lot of devices ? The file can be hudge.

canedje, 2020-12-02:
the Log of the Docker while pairing:
17:21:48:478 GW firmware version is up to date: 0x26660700

17:22:05:883 created username: 2135F90D23, devicetype: Phoscon#B1920x947
17:22:07:072 New websocket 192.168.2.28:53036 (state: 3)
17:22:07:391 Websocket disconnected 192.168.2.28:53036, state: 0, close-code: 1001, reason:
17:22:08:534 New websocket 192.168.2.28:53037 (state: 3)
17:22:08:534 New websocket 192.168.2.28:53037 (state: 3)
17:22:19:658 Websocket disconnected 192.168.2.28:53037, state: 0, close-code: 1001, reason:
17:22:20:769 New websocket 192.168.2.28:53039 (state: 3)
17:22:43:461 Current channel 15
17:22:43:469 Device TTL 2239 s flags: 0x7
17:22:43:828 ZCL attribute report 0x04CF8CDF3C78B547 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
17:22:43:461 Current channel 15
17:22:43:469 Device TTL 2239 s flags: 0x7
17:22:43:828 ZCL attribute report 0x04CF8CDF3C78B547 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
17:22:46:946 send permit join, duration: 59
17:22:46:946 send GP proxy commissioning mode
17:22:48:645 ZCL attribute report 0xBC33ACFFFE6974D5 for cluster: 0x0001, ep: 0x01, frame control: 0x08, mfcode: 0x0000
17:22:55:946 Tuya debug 7 : Missing manufacture name for 0x842e14fffe61e09b
17:22:55:947 DB SELECT manufacturername FROM nodes WHERE mac LIKE '84:2e:14:ff:fe:61:e0:9b%' COLLATE NOCASE: _TZ3000_kdi2o9m6
17:22:55:947 Tuya debug 7 : Missing manufacture name for 0x842e14fffe53002b
17:22:55:948 DB SELECT manufacturername FROM nodes WHERE mac LIKE '84:2e:14:ff:fe:53:00:2b%' COLLATE NOCASE: _TZ3000_kdi2o9m6
17:22:58:441 ZCL attribute report 0x842E14FFFEF5A27E for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000
17:22:58:441 ZCL attribute report 0x842E14FFFEF5A27E for cluster: 0x0000, ep: 0x01, frame control: 0x18, mfcode: 0x0000
17:23:04:991 0xF6CD nwk changed to 0xE9AD
17:23:04:991 device announce 0xBC33ACFFFE6974D5 (0xE9AD) mac capabilities 0x80
17:23:04:992 set fast probe address to 0xBC33ACFFFE6974D5 (0xE9AD)
17:23:04:992 FP indication 0x0000 / 0x0013 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:04:992 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:04:992 device announce 0xBC33ACFFFE6974D5 (0xE9AD) mac capabilities 0x80
17:23:05:847 [1] get node descriptor for 0xbc33acfffe6974d5
17:23:05:945 Tuya debug 7 : Missing manufacture name for 0x842e14fffe61e09b
17:23:05:945 DB SELECT manufacturername FROM nodes WHERE mac LIKE '84:2e:14:ff:fe:61:e0:9b%' COLLATE NOCASE: _TZ3000_kdi2o9m6
17:23:05:946 Tuya debug 7 : Missing manufacture name for 0x842e14fffe53002b
17:23:05:946 DB SELECT manufacturername FROM nodes WHERE mac LIKE '84:2e:14:ff:fe:53:00:2b%' COLLATE NOCASE: _TZ3000_kdi2o9m6
17:23:06:073 FP indication 0x0000 / 0x8002 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:06:073 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:06:073 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) cluster 0x8002
17:23:06:074 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) clear timeout on cluster 0x8002
17:23:06:079 [2] get active endpoints for 0xbc33acfffe6974d5
17:23:06:073 FP indication 0x0000 / 0x8002 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:06:073 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:06:073 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) cluster 0x8002
17:23:06:074 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) clear timeout on cluster 0x8002
17:23:06:079 [2] get active endpoints for 0xbc33acfffe6974d5
17:23:08:020 FP indication 0x0000 / 0x8005 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:08:020 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:08:021 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) cluster 0x8005
17:23:08:021 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) clear timeout on cluster 0x8005
17:23:08:027 [3] get simple descriptor 0x01 for 0xbc33acfffe6974d5
17:23:08:044 FP indication 0x0000 / 0x8005 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:08:044 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:08:044 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) cluster 0x8005
17:23:08:050 wait response fastEnddeviceProbe() 0xBC33ACFFFE6974D5, elapsed 22 ms
17:23:08:094 FP indication 0x0000 / 0x8005 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:08:094 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:08:094 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) cluster 0x8005
17:23:08:101 wait response fastEnddeviceProbe() 0xBC33ACFFFE6974D5, elapsed 73 ms
17:23:09:929 FP indication 0x0000 / 0x8004 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:09:929 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:09:929 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) cluster 0x8004
17:23:09:929 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) clear timeout on cluster 0x8004
17:23:09:936 [3.1] get IAS Zone type for 0xbc33acfffe6974d5
17:23:10:062 FP indication 0x0104 / 0x0500 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:10:062 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:10:062 Clear fast probe timeout for cluster 0x0500, 0xBC33ACFFFE6974D5
17:23:10:068 [4.1] Get manufacturer code
17:23:10:068 [4.2] get basic cluster attr 0x0004 for 0xbc33acfffe6974d5
17:23:11:096 FP indication 0x0104 / 0x0000 (0xBC33ACFFFE6974D5 / 0xE9AD)
[... 65 lines cut ...]
17:23:16:063 06 LQI (?) 4295557120 (0x0100090000)
17:23:16:063 64 temperature 533
17:23:16:063 65 humidity 7955
17:23:16:063 66 pressure 101079 (1011)
17:23:16:063 0a unknown 0 (0x0000)
17:23:16:147 Bind response success for 0xbc33acfffe6974d5 ep: 0x01 cluster: 0x0500
17:23:16:566 CTRL skip polling while permit join is set
[... cut ...]
17:23:18:414 FP indication 0x0000 / 0x8004 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:18:415 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:18:415 ZDP indication search sensors 0xBC33ACFFFE6974D5 (0xE9AD) cluster 0x8004
17:23:19:153 FP indication 0x0104 / 0x0500 (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:19:153 ... (0xBC33ACFFFE6974D5 / 0xE9AD)
17:23:19:449 saved node state in 2 ms
17:23:19:554 sync() in 103 ms
17:23:19:898 Search sensors done
17:23:19:945 send permit join, duration: 0
17:23:19:945 send GP proxy commissioning mode``

Smanar, 2020-12-02:
Lool, Magic.

@SwoopX, I m looking the fonction checkIasEnrollmentStatus(), this one have always a debug output, and this fonction is trigger during sensor creation, do you think it was missing from log ? RConfigPending ? iasZoneStatus ?

SwoopX, 2020-12-02:
It is missing indeed, but only because debug output provided is level 1. The IAS messages require level 2.

canedje, 2020-12-02:
I Also do have trouble pairing a BlitzWolf BW-SHP13.
I do have 3 of them. Two are pairing just fine.
The thirt |I'm already busy for a week not able to pair.
It drives me crazy
It even blowing up my docker app. After a while all my lights are gone and I do need to reinstall a backup to get it running again.
(The lights are not showing up in the deconz app, but are still working in HA???)

derrohrbach, 2020-12-03:
I also bought the ceiling Light from Lidl. Can confirm it is working fine with Home Assistant and Phoscon. It also comes with a remote which pairs fine and shows up in deCONZ, but is not visible to Home Assistant or Phoscon. What info do you need to make it work?

I also got the Motion Detector. It works fine and shows up in Home Assistant, not in Phoscon though. To get it to work I clicked on exec for "Zone Status Change Notification" and "Zone Enroll request" 
[image]

hbkoenig, 2020-12-03:
> they also offer RGB light bulbs (both E27 and E14) for 14,95 EUR, but I can't find these mentioned in this issue.
> 
> are they already supported ? if not, any chance/plans for support RSN ?
> if it helps I'm willing to buy some bulbs day after tomorrow and provide all needed infos...

this morning I bought one E14 and E27 RGB blubs, both work fine with 2.07.00 beta and phoscon.

both show up with Model Identifier (0x0005) TS0505A
Manufacturer Names:

E27: _TZ3000_dbou1ap4
E14: _TZ3000_odygigth

are you interested in any detailed data?
unfortunately uploading screenshots or files into this tracker again does not work for me from company network:-(
if wanted/help I'll add more infos later from home

I'll test power switch and motion sensor later...

Stefan-83, 2020-12-03:
Regarding the Lidl-Lightstripe ([LIVARNO LUX® LED-Band »Zigbee Smart Home«](https://www.lidl.de/de/livarno-lux-led-band-zigbee-smart-home-individuell-teilbar-selbsthaftend/p354570)) and Lidl-Bulbs-G10 ([LIVARNO LUX® Leuchtmittel](https://www.lidl.de/de/livarno-lux-leuchtmittel-lichtfarbensteuerung-zigbee-smart-home/p354568)):
Is or will these device be able to use the [Adaptive lighting on iOS14](https://github.com/ebaauw/homebridge-hue/issues/766) feature?

Thanks and best regards,
Stefan

Smanar, 2020-12-03:
@hbkoenig no, we just need the confirmation they are working natively, like standard bulb, tuya device can be realy exotic ...

hbkoenig, 2020-12-03:
> @hbkoenig no, we just need the confirmation they are working natively, like standard bulb, tuya device can be realy exotic ...

yes, they do -- at least brightness, color temperature, HSB color with phoscon. 

I don't care (and know;) about fading or any other "special" effects though as I'll use them only as green/yellow/red traffic light (plus dimming/brightness) for air quality monitoring (corona airing)...

Smanar, 2020-12-03:
Ha ? they have effect too ?

derrohrbach, 2020-12-03:
@Smanar I can confirm the following devices:

Lidl Smart Plug [SILVERCREST® Steckdose Zwischenstecker Zigbee Smart Home](https://www.lidl.de/de/silvercrest-steckdose-zwischenstecker-zigbee-smart-home/p355168)
- Works, shows up in Home Assistant and Phoscon

Lidl Ceiling Light [LIVARNO LUX® Deckenleuchte »Zigbee Smart Home«](https://www.lidl.de/de/livarno-lux-deckenleuchte-zigbee-smart-home-fernbedienung/p354579)
- Works, shows up in HomeAssistant and Phoscon
- The Remote, which is also included in the Package, does pair to deCONZ, but does **not** show up in HomeAssistant or Phoscon.

Lidl Motion Sensor [SILVERCREST® Bewegungsmelder »Zigbee Smart Home«](https://www.lidl.de/de/silvercrest-bewegungsmelder-zigbee-smart-home-infrarot-sensor-anti-manipulationsalarm/p354561)
- Works, after you send it some commands (See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3762#issuecomment-737848583). Only shows up in HomeAssistant, not in Phoscon

Please tell me what infos you need so we can add support for the remote control. I would like to use it to control some other lights through HomeAssistant.

MarcelMertens, 2020-12-03:
Hi,
these devices came up in german LIDL retails stores today.
I bought one [LIVARNO LUX® Leuchtmittel RGB »Zigbee Smart Home«, dimmbar  ](https://www.lidl.de/de/livarno-lux-leuchtmittel-rgb-zigbee-smart-home-dimmbar/p354569). Paired succesful, but on On/Off and Dim is working.
Color, Color temperature, Saturation are not working. 

Working with Deconz api and iobroker plugin

PicCharr, 2020-12-04:
Also trying to get the motion sensor working. The current stable deconz version does not support it. The beta does support it, but is as already mentioned not shown in the phoscon app. But the API does show it as a tuya sensor, therefore it can be successfully used in openHAB without doing something extra.

Stefan-83, 2020-12-04:
> Regarding the Lidl-Lightstripe ([LIVARNO LUX® LED-Band »Zigbee Smart Home«](https://www.lidl.de/de/livarno-lux-led-band-zigbee-smart-home-individuell-teilbar-selbsthaftend/p354570)) and Lidl-Bulbs-G10 ([LIVARNO LUX® Leuchtmittel](https://www.lidl.de/de/livarno-lux-leuchtmittel-lichtfarbensteuerung-zigbee-smart-home/p354568)):
> Are or will these device be able to use the [Adaptive lighting on iOS14](https://github.com/ebaauw/homebridge-hue/issues/766) feature?
> 
> Thanks and best regards,
> Stefan

I've just received my first Lidl-Bulbs-G10 Ambience (not the RGB).
After easily connecting with ConBee2, the Bulb showed up in HomeKit (over Homebridge_hue) as Heiman TS052A: On/Off, Dimming and Color temperature is working.
And good news: "Adaptive lighting on iOS14" is working as well!

popy2k14, 2020-12-05:
@Stefan-83
Does the led Stripe (led Band) works with current stable deconz 2.5.88?

Thx

KaijuDeluxe, 2020-12-06:
I cant get it to work with Rasbee2 and the Firmware 26680700
Yesterday i added bulbs and it worked, after a reboot everything was gone and i cannot re-add the devices :(

Manlycookiebear, 2020-12-06:
Neither the Power Plug nor the Power Bar support power measurement, right?

sahniel, 2020-12-06:
> Neither the Power Plug nor the Power Bar support power measurement, right?

Right,
On the powerbar you can switch every outlet except for the usb ones on their own but you can't measure the consumption

PicCharr, 2020-12-06:
Does the door sensors run on anybodys setup? Thinking about to ask santa for 6 pieces ... ;)

Tedious01, 2020-12-07:
Sensor is working "partly" in latest beta. Cannot see it in the App, I'm getting Data into FHEM - unlikely everything except movement information :-/

`2020-12-07 08:50:19 HUEDevice LidlBWM_1 batteryPercent: 100
2020-12-07 08:50:19 HUEDevice LidlBWM_1 battery: 100
2020-12-07 08:50:19 HUEDevice LidlBWM_1 reachable: 1`

Mimiix, 2020-12-07:
Hi Guys,

As all devices seem to be integrated properly, i am closing and locking this issue.  

Phoscon does not support this device yet, so please open a device request there to get it integrated in that. 

Thanks!

closed 2020-12-07 by Mimiix