dresden-elektronik/deconz-rest-plugin #3765: Issue and Support Situation

manup, 2020-11-27:
I'd like to provide a frank overview about present support challenges in order to provide a better understanding and a small glimpse into the future.

When problems and bugs are reported it's just natural to want them to get solved. If that doesn't happen in a timely manner the frustration level rises, we understand that. Albeit bugs get fixed regularly it's not possible to fix everything and solve every problem a user might face in a short amount of time.

It's a ongoing challenge at scale as the user number grow rapidly, currently we're at >95K gateways with 5K added each month, given these numbers the amount of support cases is relative small but it's still hard to manage to get everybody happy.

There is the Phoscon support email address, which was working nicely in the beginning but currently we struggle to answer all mails, there is only that much you can answer in one day. We have one very hard working full time support guy and also internal developers jumped in to help, but it has grown that much that it's not possible anymore to answer everything.

But, I've just got the info from HR that we finally get a further support member to extend the team very soon, it was hard to find someone, the job offer was out for months.

Regardless, it's apparent that the email approach doesn't work anymore, therefore we're looking into other directions to address this better, one of them is a separate forum for end users, so this Github issue tracker will become sole about the REST-API and new device requests.

Not so long ago this Github issue tracker was not moderated, and as a result it got messy over time. Luckily Dennis stood up and offered to moderate and bring order into this. It's a hard cut from the old way and I understand it can be seen as too progressive at times. Please understand us here, without going strict and enforcing, for example templates and separate issues for separate problems, we don't stand a chance. Dennis and other moderators and developers provide all this help in their spare time for which I'm deeply grateful. In case of fixing bugs and adding new devices the streamlined approach has helped dramatically.  Thanks to the templates many question which always were needed to be asked are answered beforehand.

Being a moderator or support worker is tough, you got blamed for all sorts of stuff, get all the sh:t, all the frustration and rarely they get thanked for or at least get respected for all the help and work provided. And they need to make tough decisions, closing issues, requesting new issues to be made and asking frustrating questions is part of it. Not everybody knows everything, the Zigbee world is damn complex and we're all learning; it can also occur that things are asked for because of misunderstandings or wrong assumptions, nothing can be perfect, but we're making progress.

What I would like everybody to understand is, we are all just people, no matter if user, moderator, developer or support and as such we all need to keep the discussions calm and polite at a professional level — there shall be no place for disrespect or rudeness from any side, I know this is difficult, but this is the only way to make progress and to figure issues out. In that regard a page will be introduced which describes how and why things get handled they way they are, a set of community rules, and also the reasoning behind some tricky matters, there is no silver bullet but it's a step.

#### It's all about progress

There are some really nasty issues, like the IKEA routing quest in larger networks which goes on for years now, lots of fancy stuff has been tried here, and it's tricky because it can't be reproduced easily. But the work here continues as we try out further things.  In the company we have build a new test network with all sorts of devices in order to better reproduce bugs in real world and large networks.

Another tricky one is the USB issue within Docker containers. It may not look like it from the outside but such things have a high priority and lots of testing and development work *is* going into figuring these out. These "die hard" bugs often take months sometimes years to get right, there are no short cuts for complex problems. The USB/Docker issues for example were only figured out last week after many months and versions with improvements and tests.

Behind the scenes the work on the next major version of deCONZ 3.x has been started a while ago which will come in 2021, this will be a game changer version which solves heaps of issues, I'm looking very forward to it since it also means less support issues.

The documentations for C++ side and the REST-API have seen an overhaul and getting updated to let external clients better facilitate the APIs.

Recently we have introduced release schedules which bring regular beta and stable updates each month, where the stable releases are all about bug fixes, and beta versions include mostly newly supported devices.

In terms of supported devices the situation is very diffuse, there are various manually maintained places which list some devices which work with deCONZ, they are all incomplete. And even for the listed devices it's not clear how well they are supported, do they work with client x/y, how do they look in the REST-API and importantly how reliable are they and do they have known issues?

To address that future versions will use so called device description files, which include all this data. The documentation as well as supported devices list can be generated automatically. They also guide what products are recommended and something like a rating to not run into known issues when extending the home arsenal.

I'm aware this won't please everyone, but I hope it sheds a bit of light at the current state.

– Manu

cbrightly, 2020-12-10:
Thanks for the information, I guess?
Glad to hear you have hired more support staff. It sounds like this should have been done a long time ago; maybe commit  to supporting  your product and customers or end-of-life the Conbee II and save us all the multiple headaches.

Mimiix, 2020-12-10:
@Mattwestb feel free to discuss further here :)

MattWestb, 2020-12-10:
Thanks for the offer but for the moment after https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3051#issuecomment-742545730 not thanks !

And one great happy ending of . . .

ChrisPrefect, 2020-12-10:
Mimiix, you are again escalating instead of de-escalating. Customers are angry. You (or DE) have to accept that and deal with it. Blocking and Locking and Deleting does not help, you are making angry people even more angry. Again: This is the official end user support here.

ChrisPrefect, 2020-12-10:
None of my problems are fixed, even after almost a year of trying to document everything as best as I could and talking to DE and investing a LOT of time here in github issues. Nothing changed so far.

I again have many devices that are just lost and I have to reset them and connect them again. This is frustrating.

If you need logs, send me the commands I need to run on the raspi.

ChrisPrefect, 2020-12-10:
How many DE developers are currently working on the project? Giant and easily reproducible bugs like the broken scenes for Tradfri just have to be fixed within 1-2 weeks. One month max. But it's still not fixed over a year later! The setup of this project is not right. And it is very frustrating for everyone invloved.

Mimiix, 2020-12-10:
I would like to though. I've kept myself on the backline of this stuff happening in the past few weeks. However, i feel that it's time that i say something about it. 

**This reply is solely on my own personal opinion and as a end user.** 

I feel a bit conflicted. Just for two reasons, at the one side people seem to help each other and try to get things sorted. On the other end, people are bashing others and are being disrespectful to each other. So Manuel wrote this issue, talking about how things went. Explaining stuff. Great, right? Well, no response is given. No discussion is being made and yet a lot of complaints are made again. To those i would like to say: either be part of the solution or move on. 

The thing that affects me the most is that people hardly remember there's an actual person on the other side of the screen. Yes, i forget that too sometimes. I try to explain that to myself when i write messages to others: "Maybe that user is having a bad day? Maybe he is just frustrated, i shouldn't take it personally". Well, that's getting harder every time it happens. I am not sure if it's due to COVID or other reasons, but it does happen. This demotivates, but I've got people around me that keep insisting that I'm doing a good job. Hard to believe that sometimes, but okay. 

For example, i stood up for this user. Felt sorry for him and wanted to help him out. Then this user blames me in the end. Why? Because he doesn't like how DE acts and so he rages on me. Asking for details to help him out got replied with _ I don't get paid to research and document bugs here._ Well, help us to help you. The user even demanded me to help him.  Please be advised that 1, i am not obligated to help anyone and 2 wish to be respected. Those requirements were not met and now I am blamed for it. Well, sorry for trying to help i guess.  

Do i think the support thing went out of hand? Yes! Did i tell Manuel? Yes! Is he trying to fix it? Yes! So really, stuff is changing. Maybe not so fast, but it is. With items like this, it is hard to make everyone happy. But it is teamwork and we should work together to make it better. Does playing the blame game help? No, as i learned that myself during my time here. But hey, i am open for feedback and trying to be there for the users. 

What i am trying to say: Nobody is perfect. There's a lot of nice people in this community(@Smanar @SwoopX @ebaauw @KodeCR @Thomas-Vos @MattWestb and more!)  that are here to help others out with questions.

A lot of you know my story. I've stepped up to this position because i was frustrated in how deCONZ was handled and managed. Fast forward: A lot has changed, but i am still not happy with everything. For example, Github is not the place for discussions like these so we might have a forum later.  But that's all future work. Step by step we grow!

 I, or anyone else with "powers" has never deleted any comment on this Github or on Discord. I strongly believe in free speech and a opinion is never overruled. Any allegation saying otherwise is a straight up lie.  Locking issues is only done to freeze the situation. Moderation on github is hard as there's limited ways to Seperate issues or discussions. Voices are never muted. 

To @ChrisPrefect i would like to say: In the case of issue (3406), Swoop replied to you and you blame us for not replying. Afterwards, i asked you for the logs (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3406#issuecomment-724785885), and https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3406#issuecomment-733513591. You even reproduced some, but not the ones i wanted(https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3406#issuecomment-733982341). After that, you blamed us _again_. And now you are asking "If you need logs, send me the commands I need to run on the raspi." You know what we need, but i already made it clear to you that i wasn't going to help anymore.  You've said stuff that impacted me way to hard and i wish to not get involved with you. Does that mean I moderate differently on your issues? Nope, same guidelines as any other issue. 

This being said: I like to be part of the solution. Making this amazing thing grow and grow. I've met some amazing people and after covid i hope to meet them some time. Come along the train and be part of the solution! If you feel that you are not heard or if you feel that we won't make it: Take the other train and move away:)

_Be the change that you want to see in this world_

cbrightly, 2020-12-10:
@MattWestb I am the one who wrote the post commenting on how the Dresden Elektronik should do more than just tell us that they are sorry they can't offer consumer support despite taking our money after making claims of a two year warranty. I am also the person who deleted that post. I wasn't censored in any way.
The ConBee II product seems well engineered especially when compared to older generations of DIY/enthusiast grade Zigbee equipment but from the consumer's perspective I currently see a company that either needs to:
- [ ] dedicate more resources to after sales support (i.e. pay dedicated staff to handle technical and customer support) **and** 
- [ ] be 
more upfront about the technical nature of their product
_or_
- [ ] end-of-life their current products or reduce the duration / scope of the (current) 24-month guarantee/warranty 
**and
- [ ] Save us all more headaches 

Although Dresden Elektronik made what seems to be an okay product, they seemingly can't keep up as the space matured.

My ConBee II failed after a year. The real kicker for me was having to find out _why_ the community has become the de facto support department: it's because email and social media posts go unanswered. It effectively doesn't matter **why** I can't get a replacement￼ (it's due to obvious inability to scale their company's model), the only thing that matters is that I can't expect to now own what was sold to me now they as a company and I will likely never get a resolution to my issue. Now Dresden Elektronik is suffering as a company and seemingly also as individuals (its sad to see things get personal; GitHub isn't for finger pointing or infighting but it **is** a valid place to discuss the future of open-source projects); the entire community is also now suffering from their problems. Hopefully they will resolve the problem by growing to address the need for something like their product line. Sadly at this point I would also settle for the crushing honesty of them admitting that the GitHub community is the official yet unpaid support department of Dresden Elektronik, and wish they would end any pretense of there being an entity standing behind the units they sell.

ChrisPrefect, 2020-12-14:
@Mimiix, I don't know you and I don't have anything personal against you. But I don't like it that after I invest my free time to help a company I get "lectured" about how exactly I have to report back the requested information. That's really not my job. Again: This is the official end user support page. It's like when you write a detailed e-mail to a support address, and they respond back, not with a solution, but with a guideline on how I have to format my e-mail to them.

>"On the other end, people are bashing others and are being disrespectful to each other. "

I don't think anyone was bashing another. Users are just annoyed and angry that this system is still full of bugs and not progressing nearly as fast as it should.

> So Manuel wrote this issue, talking about how things went. Explaining stuff. Great, right? Well, no response is given.

Yes, what Manuel wrote was all logical and made sense. There was no need to talk back or comment.

> and yet a lot of complaints are made again. 

Sure we keep complaining! Nothing changed! It was just a description on how the status currently is. We already knew this. But that does not excuse the situation.

> To those i would like to say: either be part of the solution or move on.

Why? Why do end users need to be part of the solution? The product has to be the solution. If the product is not a solution but a problem, then the manufacturer should deliver the solution.

We are already doing much more than we should. We are EXTREMELY patient and we invest A LOT of free time to help the manufacturer solve his problems.

> The thing that affects me the most is that people hardly remember there's an actual person on the other side of the screen.

No, why should we? This is not about a person. This is about a product that was sold by a company. I don't care about people, I care about a working product.

> For example, i stood up for this user. Felt sorry for him and wanted to help him out. Then this user blames me in the end. Why?

Because not one of my problems was fixed. Simple as that.

> Because he doesn't like how DE acts and so he rages on me.

I did not in any way rage on you. I told you that I don't like locked threads and beeing told on how exactly to report information your requested back.

> Well, help us to help you.

I do. I do everything I can. But apparently the logging does not work properly. I checked the logfile, but there was nothing new written to it. And I asked what exactly I should do to help, I didn't even get back correct command line parameters. I DON'T USE LINUX! I can poke around and enter some command lines, but if you want specific log files then you have to tell me what I need to inter in the command line! I said that multiple times. I can't believe that the manufacturer is not able to tell the customer what he needs to run to get the needed logfiles. 

I even offered full access to my setup! It would be very easy to remotely access my site and test all you want to test! We can even talk on the phone and I try to reproduce the bugs directly. Why don't you want my help then? Why does DE refuse to look at the bugs directly? Why do you blame me for not knowing what command line parameters I need to enter on DE system to get the logs you want?

> You know what we need, but i already made it clear to you that i wasn't going to help anymore

Why do you think you are helping ME? DE's product needs help! The bugs should be fixed! It is not MY problem! It is the problem from DE that the product is buggy! DE apparently NEEDS HELP and I offered to give it, but DE didn't even want it! It is not that I make some mistake and need help! It's the other way around!

> This being said: I like to be part of the solution. 

Sure, good for you. I don't have as much free time and energy as you. I can help you and DE up to a certain degree, but not more. After some time I need solutions.

> Making this amazing thing grow and grow.

I didn't really see it grow during the past year. I still have the same bugs.

> I've met some amazing people and after covid i hope to meet them some time.

I am not here for the people. I already have good friends. I am here because my lights don't turn on when I press the light switch!! Simple as that! Nothing else!

cbrightly, 2020-12-14:
I think a lot of the systemic issues would be solved naturally with better delineation between end-user support and the open source project.

Mimiix, 2020-12-14:
> But on the other hand: cbrightly wrote a very good comment, but got not response from you or DE. **Why not?**

@cbrightly removed his comment. We didn't reply on it for that reason. 

>I think a lot of the systemic issues would be solved naturally with better delineation between end-user support and the open source project.

Yes, and that's something to work out in the upcoming year. As we spoke about in private, it's all about expectations and communication. As far as I'm concerned, DE should communicate that discord/git are community support. If there's any other kind, that should be communicated as well.

ChrisPrefect, 2020-12-14:
> @cbrightly removed his comment. We didn't reply on it for that reason.

Uhm, it's still here? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3765#issuecomment-742821720

> it's all about expectations and communication. 

I don't think so. OK if you think that expecting a working system is too much, then yes, those expectations have to be lowered right before someone chooses a Conbee2-Stick.

The expectation is, that bugs will be fixed within weeks, not years. I don't know how this expectation should be handled? Is it not reasonable?

DE does not have any support. So this is the only place to get support. It does not help if DE offers telephone and e-mails support, but they can't do anything about the bugs. I talked for hours on the phone with DE. They are very nice, but could not help in the slightest. 

So it really, really boils down to one thing: Fix the bugs! 

Everything else will solve itself.

mcormick, 2020-12-21:
To hop onto this thread, I myself have seldomly commented anywhere in the issue tracker. Most of the times I come here to read about new progress (monthly releases), or even get new ideas for my home automation just by reading about newly supported devices.

The issue with issue trackers is that it is filled with ... well, issues. And people with ... issues; device issues, mostly. Sometimes threads heat up a little, which is far from constructive, or even fun to read for the casual surfer-by.

My 37 node network is overall functioning as it should, with the occasional quirk at a new release that is often fixed in a next. For myself the Conbee II is somewhat of a holy grail. It's much more versatile than many other brand-bound "hubs", but for the same 40 euro's. For 40 euro's you get a more powerful device, developed by a small company with limited resources. So imo the real issue here is expectancy.

Can you really expect to get more for less, or even demand that your particular issue is more important than other's?
For anyone involved in the development (voluntary and professionally), please don't forget that there always is a silent majority that is really grateful for what you're doing. It deserves to be expressed, every once in a while.

cbrightly, 2020-12-21:
A lot of people are working very hard to address user issues. I hope it goes without saying that (at least my personal) criticism is meant to be constructive.

Also: It is worth mentioning that for those not "in the loop", a lot of discussion, planning, and feedback for this project is happening on Discord. It sounds like volunteers and DE themselves are making huge progress toward resolving all the issues addressed here! Unfortunately, these awesomely laid out plans and any other communication happening on Discord do not make it onto the issue tracker here. It seems to some users that no progress is being made and that there is little discussion of fixing their problems. I understand Discord is used differently as a medium than GitHub and not everything being discussed there is worthy of note to (most) users, but some of the details that are currently being discussed there directly affects users with unresolved and (seemingly) otherwise unaddressed issues. It may be worth considering some type of further consolidation / integration between platforms (i.e. the Discord GitHub bot).

Many thanks to the volunteers and all those contributing and those who are helping the Deconz community grow through discussion and feedback!

ChrisPrefect, 2020-12-21:
@mcormick , How many Tradfri-Devices do you have?

I think many issues are with Tradfri. Sadly, I have over 100 Tradfri-Devices.

It's not at all about "what you get for 40€". Again: I would pay DE 1'000€ immediately if my issues would be fixed.

For me, it's that I have no other option. I need Phoscon. I can't use another Zigbee-system to bind 25 remotes to 20 light groups. That would lead to around 300 individual blockly-scripts in ioBroker, if I would use the TI zigbee developlent board instread of deconz for example.

Would it be possible to port phoscon to ioBroker, so it can be used with the zigbee-adapter or zigbee2mqtt?

> A lot of people are working very hard to address user issues.

Could be, but I don't see any of that :-( The exact same bugs I reported a year ago still exist today. And that's what frustrates me. I see ZERO progress. I might be unlucky that none of the fixes affect me.

@cbrightly, you are right: Part of the frustration comes from the fact that I don't see anything happening. If I knew that someone is, for example, working on the tradfri szenes bug or the alexa groups bug, then I might be able to help, provide tests, test beta-versions or even provide logs. But I created many issues here with a lot of screenshots and tests and I got nothing helpful back.

Mimiix, 2020-12-22:
> I understand Discord is used differently as a medium than GitHub and not everything being discussed there is worthy of note to (most) users, but some of the details that are currently being [...]

I've been trying to figure out how to make this happen. We have a channel in Discord that follows all issues in Github. For github i have no idea how to merge it with Discord. Ideally, i'd have the announcements i make once in a while on a special page. However, I am not show if i can show that anywere on Github. It doesn't have a special "News" section or so. Github is intended for devs and such. We are already "abusing" Github for things that it isn't meant to do. 

For now, the only way i can think of is that i post the announcements i make on Discord, in a Github issue having a label "Announcement" and being closed. Would that be a good temporary solution? I really need some input on this one. Maybe there's some features i don't know about.  In the next year, we are probably getting a forum for all stuff we do on Github except coding/device requests.

ChrisPrefect, 2020-12-22:
Why not provide real end user support?

I sent multiple e-mails tosupport@dresden-elektronik.de but never got a reply.

Why are there no support tickets created for these support request?

You could also send out newsletters to inform end users about the development. There is currentyl ZERO communication from DE to end users.

Discord is not an option. I don't want another messenger/chat tool. Especially not for support questions. I am not looking for a community, I am looking for support. It's impossible to discuss problems in discord. That's what github issues are for.

cbrightly, 2020-12-23:
> Would that be a good temporary solution? I really need some input on this one.

The releases feature on GitHub maybe? …and/or updating README.md or CHANGES.txt? I will have to consider this further I suppose.

> Why not provide real end user support?

ZenDesk or the likes is a popular option that tackles common questions and provides support ticketing as well.

Mimiix, 2020-12-23:
> The releases feature on GitHub maybe? …and/or updating README.md or CHANGES.txt? I will have to consider this further I suppose.
> 
You mean this right? https://github.com/dresden-elektronik/deconz-rest-plugin/releases We already use that for release notes and changes.  

> ZenDesk or the likes is a popular option that tackles common questions and provides support ticketing as well.

This is something DE has to answer. I am not in charge or going to run that.  They use email for support now, so if that doesn't go i don't think ZenDesk or any other ticket system is going to change anything in their behaviour.

lukicsl, 2021-01-05:
@Mimiix did you had a look at [Discourse](https://www.discourse.org/).
I ramped up a living-community forum in no-time. This platform is definitely better for discussion as the github issue tracking.
They have a hosting service at 100$/month. Found other providers starting from 25$.

Mimiix, 2021-01-05:
@lukicsl I did! However, i had trouble getting a email server. I'll ask Manuel to set up one at DE servers so we can mess around with it. How did you do the mailing part?

lukicsl, 2021-01-05:
Mailing is part of the hosting. I did also so with FaceBook. I think so with github is possible too.

devployment, 2021-01-14:
Discourse is great. But why deal with the trouble of installing, maintaining and keeping a mail server secure? I have no idea about the amount of emails we are talking about. But I would go for services like mailgun or mailjet for transactional emails. 

Just saying that using Discourse or not should be depending on getting a mail server or not. :)

KennethLavrsen, 2021-04-17:
Depressing to read this thread. A year ago Deconz was the king of Zigbee with Home assistant's ZHA being feature incomplete and Zigbee2MQTT being a new kid on the block.
Now it seems the two open source alternatives are running away from Deconz.

My observation is that the rest-API is under a lot of development driven by unpaid volunteer people. The Deconz app is dead and Phoscon does not even support the new Philips Hue devices even after 3 months.

Maybe the best would be to change the remaining Deconz and Phoscon apps to full open source. If DE does not invest the development resources in them anyway, that might be better and keep the sale of the Conbee dongles alive.

I can understand that it is impossible for a small company to support every cheap unknown Chinese manufacturers sensor or switch that comes up. But right now not even the devices of the leading Brand (Philips Hue/Signify) are supported 3 months after they have been put on the market. The 4 button Philips Hue dimmer switch has been updated in January with a new model. RWL022. I was driving from store to store to find the old and had to give up. I found one in a kit with a lightbulb and that was it. You can only get the new. And it is not supported yet by Phoscon. I can pair it. But I cannot setup anything in the Phoscon app because it does not show up there. This is a product being neglected.
And if the business situation is that the sale of dongles cannot pay for the development, then we have to accept that fact But then instead of letting the project die, put it in open source.

manup, 2021-04-17:
Hi @KennethLavrsen, I'd like to give a small update here since many things which are happening are not public yet or not easy to spot.

Not in your text, but the topic of this very issue is the support situation which has improved a lot since the creation of this issue, we have now extended our support team and are able to respond much quicker to requests via Email, it's not yet perfect but compared to the overfilled inbox a few months ago a huge step forward.

>My observation is that the rest-API is under a lot of development driven by unpaid volunteer people. The Deconz app is dead and Phoscon does not even support the new Philips Hue devices even after 3 months.

Especially for supported devices in the REST API most of this work is done by the community. Since recently we startet to support community volunteers since all the contributions are a very important and much appreciated part of our products. As deCONZ main developer my focus is to modernize the code base of deCONZ core and the REST API to bring it to a level where it can be maintained and fully tested. And (hopefully) supporting new devices become a breeze, since the currently code base shows is age and is too complex. This work is roughly described in issues and the wiki in https://github.com/dresden-elektronik/deconz-rest-plugin-v2 and the related code in https://github.com/manup/deconz-rest-plugin/tree/device_descriptions

Some of the goals to highlight:
* Reaching maximum possible robustness, as devices will be handled very differently from how it's done currently in deCONZ and in all other projects which suffer similar issues. We need (and can) get to a point where "it just works".
* Fully auto generated device documentation and supported devices lists.
* Each supported device is described in a JSON file aka Device Description File (DDF) which can be written by non C++ developers and automatically testet. This also has the benefit that one device can't introduce bugs to other devices, which is a recurring problem.
* When DDF support emerges we can over time drastically reduce the size of the C++ code base (I recon at least 50% likely more) and in the long run eliminate most of the device specific code from the C++ side.

In this weeks community developer meeting I showed a first demo of how this new approach works and we're looking forward to bring this into shape for the first devices within the next months. I'm very exited for this since it will remove so much pain points on so many fronts :) It will take time to transition currently supported devices, but we'll get there.

Based on that the v2 REST API gets richer and more generic, which allows clients to support new devices with way less code. For example currently in order to support a new switch clients like the Phoscon App and the Home Assistant Integration need to write device specific code to handle all the button events which is tedious and error prone. But with the richer API a switch is self described and a generic client implementation can handle them all. In case of the Phoscon App this allows us to write even a generic UI for switches if there isn't a specific one yet.

>Maybe the best would be to change the remaining Deconz and Phoscon apps to full open source. If DE does not invest the development resources in them anyway, that might be better and keep the sale of the Conbee dongles alive.

Over time more code of deCONZ will be open sourced, the next part is the deCONZ library itself which is heavily used in the REST API plugin. I'd like to extend this for the GUI part as well but it's a bit more complicated and needs some refactoring first. A small core will remain closed source for the foreseeable future because unfortunately the world isn't ready yet to pay for open source.

The Phoscon App itself is under steady development, I have to admit we are way behind schedule to release the new stuff. This is something we are currently restructuring internally to align Phoscon App releases with deCONZ releases to publish work quicker. A glimpse of what's to come can be found in http://phoscon.de/pwabeta there is no plan and ETA but I can imagine the Phoscon App to become open source in future as well.

>And if the business situation is that the sale of dongles cannot pay for the development, then we have to accept that fact But then instead of letting the project die, put it in open source.

We have reached the point where the sales pay for development and we can also give back to the community. Meanwhile the business works well for this rather small niche market, currently there are approx. 130K gateways (ConBee + RaspBee) in the field. It took some years to get here and business can be surprisingly tough to get to a point where it actually works, but as I mentioned in my first post **It's all about progress** I hope I could shed some light on the things which are currently in development, the now 10 years old project isn't dying, it takes it's next step in the evolution :)

KennethLavrsen, 2021-04-17:
I am happy to hear that you are committed to your product. It has been the solid foundation of my smart home and you may have noted that I have been a strong advocat for Deconz on the Home Assistant forum.

I have my personal need covered with respect to the new Philips Hue 4 button switch. But I fear you will loose a lot of customers if you continue not to support the switches from the market leader. You cannot wait till your new platform is created. The new 4 button is in all respects the same as the old with exception of cosmetic looks and its model number. It is still a 4 button remotes with the same features and is presented in the API the same way. It cannot take more than an hour to add it to Phoscon using the existing images from the old 4-button.

I really want to see deconz continue strong. And with the support. As I wrote in another post today. Stop marking bug reports as user questions that you leave open to rot. Create bundle bug reports and mark the similar bug reports as duplicates and close them. It leaves a much better impression with the reporters and it gives you a much better overview of open defects and a central place to communicate. A win-win.

Something I learned the hard way myself when I was release managing some pretty popular open source projects for more than 10 years.

manup, 2021-04-19:
Yes I have noticed your activity in the Home Assistant forum and really appreciate that, thank you :)

I think we have basically the same opinion that popular devices from the market leaders needs to be supported as quickly and good as possible, in our experience from all supported devices only a fraction is used by most people, it's like a 5/95 ratio. For the new Hue Dimmer switch I've asked the team we still don't have this one in house for testing, but try to get one quickly. It should be already in the http://phoscon.de/pwabeta but we couldn't test it on our own yet. Looking forward to get this one quickly in the usual stable and beta releases, same goes for the IKEA shortcut buttons.

> I really want to see deconz continue strong. And with the support. As I wrote in another post today. Stop marking bug reports as user questions that you leave open to rot. Create bundle bug reports [...]

>Something I learned the hard way myself when I was release managing some pretty popular open source projects for more than 10 years.

There are many ways to approach issue handling, to be frank I can't tell which one brings the most benefits for all participants, there are always trade offs. One thing I have learned is that it's impossible to please everyone no matter what, for example we have these strict issue templates which greatly help developers and moderators to not ask the same questions over and over again in order to help, but for some users these are annoying.

But I think that we should be open minded for improvements. As @Mimiix  has mentioned a hopefully significant part of the issue communication difficulties will be addressed better when we get the forum online because we feel that a lot of topics which are currently handled in GitHub issues are better communicated in a forum — time will tell. In Discord the more real time communication has already shown to be very efficient and to solve misunderstandings quickly. Maybe we can schedule a talk to discuss and exchange ideas and a forward path, would be interesting to hear the experiences from your projects and what works well there?

Mimiix, 2021-04-20:
@KennethLavrsen  I am replying here on this comment: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4715#issuecomment-821826029

If you feel that you aren't treated with respect, i'm sorry, but that's not the intention here at all. I think it's how you interpret it. 
The discussion on the labels have been on for ages. We get a lot of user questions and 9 out of 10 issues are simple questions/user errors. To be honest: It's not for them to decide if it's a bug or not. The label itself doesn't matter at all as they don't get a different treatment. I always send them on to the correct people and i read every issue that is created. I can't decide for others.  

For myself, i don't have the time to spend to make those placeholder topics. I'm happy to invite you over to spend the time on them, i don't have the time. I am doing this in my own time when i can.  The funny thing is that most issues happen with the cheaper devices and devices with known to have firmware issues (IKEA, Xiaomi,Aqara, Tuya). 

To think in solutions: I already pinged you in the Discord and i'm happy with any feedback on how to improve. The improvements must be doable with the current time and resources we have. I agree with Manuel on this one: We need a forum.