dresden-elektronik/deconz-rest-plugin #3766: ConBee2 is not visible inside deconz or firmware updater

queeek, 2020-11-28:
## Describe the bug
deCONZ doesn't properly connect the conbee2 gateway and as a result can not connect to ZigBEe devices.

## Steps to reproduce the behavior
plugin a new ConBee II stick (no firmware update was done on my side)
install the latest version of deCONZ on a Raspberry Pi 3 without Docker following the installation instructions.
start with: 
deCONZ --dev=/dev/ttyACM0 -platform minimal --http-port=80 --dbg-info=2 --dbg-zcl=1 --dbg-error=2 dbg-aps=2
(if I don't select the device it blocks fhem from using /dev/ttyAMA0)

Issue appears with other connected USB devices and one additional Homematic gateway connected to the GPIOs (/dev/ttyAMA0) and without (removed all and tested it again) 
also with running fhem in parallel and without fhem running 
deCONZ recognizes the other gateway as RaspBee and it seems it tries to install the firmware on it.

read access rights for /dev/ttyACM0 are set
I can see the device also with lsusb and dmesg

response from GCFFlasher_internal -l
GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee  -> this is the HomeMatic gateway, not a RaspBee ...
/dev/ttyUSB1     | 0x0403 | 0x6001  | A4035xxx   | Generic FTDI -> another Gateway connected through USB
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2259xxx  | ConBee II  -> this is the real ConBee II

In Phoscon I can see the Gateway but it only shows the Product, Vendor and Version. In Firmware i get "Not connected"
Reset gateway through Phoscon fails

if I try to install the latest firmware I get the following error message
GCFFlasher_internal -t 60 -d /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2259xxx-if00 -f deCONZ_ConBeeII_0x26660700.bin.GCF
GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
retry, failed

GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26660700.bin.GCF
GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
retry, failed

## Expected behavior
ConBee2 should be visible as connected and it should be possible to add ZigBee Devices

## Screenshots

## Environment

 - Host system: Raspberry Pi 3
 - Running method: Raspbian 
 - Firmware version: ?
 - deCONZ version: 2.05.88 / 15.10.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes and no (tested both, but mainly with)
 - Is there any other USB or serial devices connected to the host system? If so: Which? (unplugged everything else during some tests, no changes in the result)
lsusb or see result from firmware tool on top:
Bus 001 Device 009: ID 1cf1:0030 Dresden Elektronik 
Bus 001 Device 005: ID 0403:6001 Future Technology Devices International, Ltd FT232 USB-Serial (UART) IC
Bus 001 Device 004: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

## deCONZ Logs
log message directly after start (in this case with --auto-connect as additional parameter, removed it for later tests)
deCONZ --dev=/dev/ttyACM0 --auto-connect -platform minimal --http-port=80 --dbg-info=2 --dbg-zcl=1 --dbg-error=2 dbg-aps=2
17:21:38:259 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
17:21:38:270 CTRL. 3.8.7.117:21:38:436 COM: use stable device path /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2259xxx-if00
17:21:38:463 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
17:21:38:463 dev /dev/ttyAMA0 (/dev/ttyAMA0)
17:21:38:464 COM: /dev/ttyS0 :  (0x0000/0x0000)
17:21:38:464 COM: /dev/ttyUSB0 : USB2.0-Serial (0x1A86/0x7523)
17:21:38:465 COM: /dev/ttyUSB1 : FT232R USB UART (0x0403/0x6001)
17:21:38:465 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
17:21:38:466 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2259xxx-if00 / serialno: DE2259088, ConBee II
17:21:38:466 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
**17:21:39:082 Invalid app argument --auto-connect**
17:21:39:084 parent process -bash
17:21:39:084 gw run mode: normal
17:21:39:084 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
17:21:39:086 sd-card cid: 035344534533324780ffffffff0111ef
17:21:39:118 DB sqlite version 3.8.7.1
17:21:39:125 DB PRAGMA page_count: 42
17:21:39:125 DB PRAGMA page_size: 1024
17:21:39:125 DB PRAGMA freelist_count: 0
17:21:39:125 DB file size 43008 bytes, free pages 0
17:21:39:126 DB PRAGMA user_version: 7
17:21:39:126 DB cleanup

the following log messages are produced continuously: 
(**without** other Devices)
22:43:20:764 failed to reconnect to network try=3
22:43:25:763 failed to reconnect to network try=4
22:43:25:785 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
22:43:25:785 dev /dev/ttyAMA0 (/dev/ttyAMA0)
22:43:25:785 COM: /dev/ttyS0 :  (0x0000/0x0000)
22:43:25:786 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
22:43:25:786 GW firmware update select USB device
22:43:25:786 GW update firmware not found: 
22:43:30:764 Daylight now: nightStart, status: 230, daylight: 0, dark: 1
22:43:30:764 failed to reconnect to network try=5

**With** other devices
22:43:50:501 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
22:43:50:502 dev /dev/ttyAMA0 (/dev/ttyAMA0)
22:43:50:502 COM: /dev/ttyS0 :  (0x0000/0x0000)
22:43:50:503 COM: /dev/ttyUSB0 : USB2.0-Serial ...
22:43:50:503 COM: /dev/ttyUSB1 : FT232R USB UART ...
22:43:50:504 COM: /dev/ttyACM1 : ConBee II (0x1CF1/0x0030)
22:43:50:504 GW firmware update select USB device
22:43:50:504 GW update firmware not found: 
22:43:53:478 failed to reconnect to network try=7

when I start a scan for **lights** the following message appears in the log
API error 950, /**sensors**, Not connected

## Additional context

queeek, 2020-11-28:
firmware update worked in the meantime

> GCFFlasher_internal -t 60 -d /dev/ttyACM0 -f deCONZ_ConBeeII_0x26660700.bin.GCF 
> GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
> Reboot device /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2259xxx-if00 (ConBee II)
> deCONZ firmware version 26580700
> R21B18 Bootloader
> Vers: 2.07
> build: Jun 17 2019
> flashing 164106 bytes: |==============================|
> verify: .
> SUCCESS
> Wait 10 seconds until application starts
> deCONZ firmware version 26660700

but everything else is still the same

queeek, 2020-11-28:
same with version 26670700

Mimiix, 2020-11-28:
Hi

Can you provide a full log file?

SwoopX, 2020-11-28:
Also, do you use HA or docker?

queeek, 2020-11-29:
no HA or Docker

queeek, 2020-11-29:
> Hi
> 
> Can you provide a full log file?

which part are you in special looking for? Its getting quite long, but I added the beginning, and the repeating part already in the original ticket. So if you let me know which part you are additionally interested in I could add it here

SwoopX, 2020-11-29:
Thanks for coming back. With regard to the log file, it might sound a bit strange, but we don't know yet what we're looking for so a log of 5 mins should do. By that, we also caught a pretty rare bug already which has only occurred every now and then.

I'd say just redirect everything to a file and share it here. One additional thing maybe: are you sure neither deconz nor deconz-gui service was running? From the information above I'd conclude that you run the deconz binary via command line previously. Starting it twice has pretty much the experienced effect.

Mimiix, 2020-11-30:
> which part are you in special looking for? Its getting quite long, but I added the beginning, and the repeating part already in the original ticket. So if you let me know which part you are additionally interested in I could add it here

Exactly what Swoop says :) Most of the time, we don't know what we are looking for yet. Some events just look fine alone, but together they look odd. That's why a full log helps best :)

queeek, 2020-11-30:
Hi, i disabled the service fully (had issues upfront with blocked device from deconz with fhem) and started it manually to directly get the logs without saving everything to the SD-card. So it was not running (the GUI version was even never running cause I run the pi fully headless.)
will create the full log later today for you

queeek, 2020-12-02:
Tested it with and without extension cable, and made sure upfront that no other instance is running
started the scan for **lights** after the startup finished
(to make sure the issue is not because of a broken extension cable)
without extension cable

deCONZ --dev=/dev/ttyACM0 -platform minimal --http-port=80 --dbg-info=2 --dbg-zcl=1 --dbg-error=2 dbg-aps=2 --auto-connect=1 
15:15:22:339 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
15:15:22:342 CTRL. 3.8.7.115:15:22:454 COM: use stable device path /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2259088-if00
15:15:22:477 COM: /dev/ttyAMA0 :  (0x0000/0x0000)
15:15:22:477 dev /dev/ttyAMA0 (/dev/ttyAMA0)
15:15:22:478 COM: /dev/ttyS0 :  (0x0000/0x0000)
15:15:22:478 COM: /dev/ttyUSB0 : USB2.0-Serial (0x1A86/0x7523)
15:15:22:479 COM: /dev/ttyUSB1 : FT232R USB UART (0x0403/0x6001)
15:15:22:479 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
15:15:22:480 COM: /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2259088-if00 / serialno: DE2259088, ConBee II
15:15:22:480 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
15:15:22:998 parent process -bash
15:15:22:999 gw run mode: normal
15:15:22:999 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
15:15:22:999 sd-card cid: 035344534533324780ffffffff0111ef
15:15:23:002 DB sqlite version 3.8.7.1
15:15:23:006 DB PRAGMA page_count: 44
15:15:23:006 DB PRAGMA page_size: 1024
15:15:23:006 DB PRAGMA freelist_count: 0
15:15:23:006 DB file size 45056 bytes, free pages 0
15:15:23:006 DB PRAGMA user_version: 7
15:15:23:006 DB cleanup
15:15:23:008 DB create temporary views
15:15:23:013 DB view [0] created
15:15:23:014 DB view [1] created
15:15:23:017 DB view [2] created
15:15:23:019 DB view [3] created
15:15:23:019 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
15:15:23:025 sql exec SELECT key FROM config2
15:15:23:026 sql exec SELECT key,value FROM config2
15:15:23:026 Load config UTC: 2020-11-28T01:09:06 from db.
15:15:23:026 Load config announceinterval: 45 from db.
15:15:23:027 Load config announceurl: https://phoscon.de/discover from db.
15:15:23:027 Load config apiversion: 2.05.88 from db.
15:15:23:027 Load config bridgeid: 0000000000000000 from db.
15:15:23:027 Load config datastoreversion: 93 from db.
15:15:23:027 Load config devicename: ConBee II from db.
15:15:23:027 Load config dhcp: true from db.
15:15:23:027 Load config disablePermitJoinAutoOff: false from db.
15:15:23:028 Load config discovery: false from db.
15:15:23:028 Load config factorynew: false from db.
15:15:23:028 Load config fwneedupdate: false from db.
15:15:23:028 Load config fwupdatestate: idle from db.
15:15:23:028 Load config fwversion: 0x00000000 from db.
15:15:23:028 Load config gateway: 127.0.0.1 from db.
15:15:23:028 Load config group0: 65520 from db.
15:15:23:029 Load config groupdelay: 50 from db.
15:15:23:029 Load config gwpassword: $1$8282jdkm$Y2eozq4GsUxizARIedB4j. from db.
15:15:23:029 Load config gwusername: delight from db.
15:15:23:029 Load config homebridge: not-managed from db.
15:15:23:029 Load config homebridge-pin:  from db.
15:15:23:029 Load config homebridgepin:  from db.
15:15:23:029 Load config homebridgeupdate: false from db.
15:15:23:029 Load config homebridgeupdateversion:  from db.
15:15:23:030 Load config homebridgeversion:  from db.
15:15:23:030 Load config ipaddress: 192.168.8.26 from db.
15:15:23:030 Load config linkbutton: false from db.
15:15:23:030 Load config localtime: 2020-11-28T02:09:06 from db.
15:15:23:030 Load config mac: 38:60:77:7c:53:18 from db.
15:15:23:030 Load config modelid: deCONZ from db.
15:15:23:030 Load config name: ConBee from db.
15:15:23:030 Load config netmask: 255.0.0.0 from db.
15:15:23:031 Load config networkopenduration: 180 from db.
15:15:23:031 Load config ntp: synced from db.
15:15:23:031 Load config otauactive: false from db.
[... 879 lines cut ...]
15:21:44:153 COM: /dev/ttyUSB1 : FT232R USB UART (0x0403/0x6001)
15:21:44:154 COM: /dev/ttyACM0 : ConBee II (0x1CF1/0x0030)
15:21:44:155 GW firmware update select USB device
15:21:44:156 GW update firmware not found: 
15:21:44:583 API error 950, /sensors, Not connected
15:21:45:100 API error 950, /sensors, Not connected
15:21:45:580 API error 950, /sensors, Not connected
15:21:46:061 API error 950, /sensors, Not connected
15:21:46:562 API error 950, /sensors, Not connected
15:21:47:488 API error 950, /sensors, Not connected
15:21:47:570 API error 950, /sensors, Not connected
15:21:48:069 API error 950, /sensors, Not connected
15:21:48:625 API error 950, /sensors, Not connected
15:21:49:073 API error 950, /sensors, Not connected
15:21:49:119 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
15:21:49:120 failed to reconnect to network try=7
15:21:49:595 API error 950, /sensors, Not connected
^C15:21:51:918 shutdown after signal(2)
15:21:51:959 saved node state in 0 ms
15:21:51:968 sync() in 8 ms
15:21:51:969 REST API plugin shutting down
15:21:51:969 DB save zll database items 0x000000C1
15:21:51:971 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('1', 'Daylight', 'Daylight', 'PHDL00', 'Philips', '00:00:00:00:00:00:00:00-01', '1.0', '{"dark":false,"daylight":true,"lastupdated":"2020-12-02T14:19:59.030","status":170,"sunrise":"2020-12-02T07:19:19","sunset":"2020-12-02T15:31:11"}', '{"configured":true,"lat":"51.1854","long":"6.4417","on":true,"sunriseoffset":30,"sunsetoffset":-30}', '', 'normal', '1')
15:21:51:989 DB saved in 19 ms
15:21:52:006 sync() in 16 ms

queeek, 2020-12-02:
just reinstalled it, and still the same

Smanar, 2020-12-02:
Have you tried to upload a new firmware ? the version you want.

Smanar, 2020-12-02:
Can you make the try on another machine, a PC, windows or ubuntu ?
And try to update the firmware again with it ?

queeek, 2020-12-08:
no Windows PC available, could try it on a vanilla Raspbian

queeek, 2020-12-10:
issue was related to raspbian version (used Jessie, now after upgrade to stretch it works)

closed 2020-12-10 by queeek