dresden-elektronik/deconz-rest-plugin #3767: SilverCrest Smart Window or Door Sensor (Lidl)

bvdberg01, 2020-11-28:
## Device

- Product name: Smart Windows or Door Sensor.
- Manufacturer: SilverCrest.
- Model identifier: TY0203
- Device type :
  - Sensor

## Screenshots
**Basic cluster:**
[image]

**Power Configuration Cluster:**
[image]

**Identify Cluster:**
[image]

**IAS Zone Cluster:**
[image]
[image]

**Diagnostics Cluster:**
[image]

**OTAU cluster:**
[image]

**Node info:**
[image]
[image]

Mimiix, 2020-11-28:
@Smanar can you do this one :)?

Smanar, 2020-11-28:
Sure.
@bvdberg01 have you an unix machine for test ?

If yes, you can take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

sudo apt install deconz-dev
git clone --branch SC_door https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

After that you can run deconz and include the device. It will probably not be visible in phoscon, so you can use third app or direclty the api to check the device creation.

Not sure for battery return yet, but I think the rest will be ok, realy few modifications.

closed 2020-11-30 by manup

robertklep, 2020-12-01:
@Smanar just installed v2.0.7-beta, with which I can pair the motion sensor and contact sensor 🥳

However, it doesn't look like the state of the contact sensor is being updated, it remains `"open" : false` even though the sensor is "open".

The zone status attribute of the IAS cluster does change value:
* 0x0020: no tamper, contact closed
* 0x0021: no tamper, contact open
* 0x0024: tamper, contact closed
* 0x0025: tamper, contact open

Mimiix, 2020-12-07:
As this is integrated properly and working now, i am closing off this issue :)