dresden-elektronik/deconz-rest-plugin #3771: Delay, too many Tasks, non responsing FLS-PP lp

Fruchuxs, 2020-11-29:
## Describe the question or issue you are having
One of my dresden Elektronik LED Strip Controllers is not responding most of the time, since the last firmeware update. Other dresden elektronik LED Strip Controller doesn't have this problem (lies near by). I noticed some new "delay"-Log-Messages, which I didn't seen before. 

Info: This only happens if I activate a scene in home Assistant with 8 Lamps! See Logs.

## Screenshots
[image]

## Environment
 - Host system:Raspberry Pi 4B 4G Version
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.05.88
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Homeatic USB Stick
- New SD Card, thanks to BF - SanDisk Extrem Pro
- All Logs logged in tempfs, no IO

## deCONZ Logs
```
01:10:01:307 dev /dev/ttyAMA0
01:10:01:308 GW firmware version: 0x26660700
01:10:11:287 dev /dev/ttyAMA0
01:10:11:288 GW firmware version: 0x26660700
01:10:21:287 dev /dev/ttyAMA0
01:10:21:288 GW firmware version: 0x26660700
01:10:26:083 send move to color temperature 270 to 0x00178801040423C5
01:10:26:083 delay sending request 116 dt 0 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0006 onAir: 1
01:10:26:083 delay sending request 118 dt 0 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0008 onAir: 1
01:10:26:093 send move to color temperature 270 to 0x0017880103A2E9BC
01:10:26:093 delay sending request 116 dt 0 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0006 onAir: 1
01:10:26:093 delay sending request 118 dt 0 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0008 onAir: 1
01:10:26:094 delay sending request 116 dt 0 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0006 onAir: 1
01:10:26:094 delay sending request 118 dt 0 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0008 onAir: 1
01:10:26:094 delay sending request 121 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0006 onAir: 1
01:10:26:094 delay sending request 123 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0008 onAir: 1
01:10:26:094 delay sending request 124 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0300 onAir: 1
01:10:26:108 send move to color temperature 270 to 0x001788010473AFE2
01:10:26:108 delay sending request 116 dt 0 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0006 onAir: 1
01:10:26:108 delay sending request 118 dt 0 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0008 onAir: 1
01:10:26:108 delay sending request 121 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0006 onAir: 1
01:10:26:108 delay sending request 123 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0008 onAir: 1
01:10:26:108 delay sending request 124 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0300 onAir: 1
01:10:26:108 delay sending request 126 dt 0 ms to 0x0017880103A2E9BC, ep: 0x0B cluster: 0x0006 onAir: 1
01:10:26:108 delay sending request 128 dt 0 ms to 0x0017880103A2E9BC, ep: 0x0B cluster: 0x0008 onAir: 1
01:10:26:108 delay sending request 129 dt 0 ms to 0x0017880103A2E9BC, ep: 0x0B cluster: 0x0300 onAir: 1
01:10:26:113 send move to color temperature 270 to 0x0017880106EF280F
01:10:26:113 5 running tasks, wait
01:10:26:133 send move to color temperature 270 to 0x001788010437D103
01:10:26:133 5 running tasks, wait
01:10:26:150 failed to add task 3281 type: 14, too many tasks
01:10:26:150 5 running tasks, wait
01:10:26:164 failed to add task 3287 type: 11, too many tasks
01:10:26:164 failed to add task 3286 type: 14, too many tasks
01:10:26:164 5 running tasks, wait
01:10:26:171 failed to add task 3291 type: 11, too many tasks
01:10:26:171 failed to add task 3290 type: 14, too many tasks
01:10:26:171 5 running tasks, wait
01:10:26:194 5 running tasks, wait
01:10:26:253 0x001788010432F989 error APSDE-DATA.confirm: 0xE1 on task
01:10:26:253 delay sending request 116 dt 1 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0006 onAir: 1
01:10:26:253 delay sending request 118 dt 1 ms to 0x00212EFFFF038983, ep: 0x0A cluster: 0x0008 onAir: 1
01:10:26:253 delay sending request 121 dt 1 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0006 onAir: 1
01:10:26:253 delay sending request 123 dt 1 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0008 onAir: 1
01:10:26:253 delay sending request 124 dt 1 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0300 onAir: 1
01:10:26:253 delay sending request 126 dt 1 ms to 0x0017880103A2E9BC, ep: 0x0B cluster: 0x0006 onAir: 1
01:10:26:253 delay sending request 128 dt 1 ms to 0x0017880103A2E9BC, ep: 0x0B cluster: 0x0008 onAir: 1
01:10:26:253 delay sending request 129 dt 1 ms to 0x0017880103A2E9BC, ep: 0x0B cluster: 0x0300 onAir: 1
01:10:26:253 delay sending request 131 dt 1 ms to 0x001788010473AFE2, ep: 0x0B cluster: 0x0006 onAir: 1
[... 107 lines cut ...]
01:10:28:198 delay sending request 177 dt 1 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0300 onAir: 1
01:10:28:198 delay sending request 179 dt 1 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0006 onAir: 1
01:10:28:198 delay sending request 181 dt 1 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
01:10:28:198 delay sending request 184 dt 1 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0008 onAir: 1
01:10:28:198 delay sending request 183 dt 1 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0006 onAir: 1
01:10:28:198 delay sending request 185 dt 1 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0008 onAir: 1
01:10:28:198 delay sending request 186 dt 1 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0300 onAir: 1
01:10:28:273    0x001788010473AAF1 force poll (2)
01:10:28:286    0x00212EFFFF058959 force poll (2)
01:10:28:286 delay sending request 176 dt 0 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0008 onAir: 1
01:10:28:286 delay sending request 177 dt 0 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0300 onAir: 1
01:10:28:294 delay sending request 176 dt 0 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0008 onAir: 1
01:10:28:294 delay sending request 177 dt 0 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0300 onAir: 1
01:10:28:294 delay sending request 181 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
01:10:28:294 delay sending request 184 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0008 onAir: 1
01:10:28:294 delay sending request 183 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0006 onAir: 1
01:10:28:294 delay sending request 185 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0008 onAir: 1
01:10:28:294 delay sending request 186 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0300 onAir: 1
01:10:28:378    0x001788010473AAF1 force poll (2)
01:10:28:392    0x00212EFFFF058959 force poll (2)
01:10:28:392 delay sending request 177 dt 0 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0300 onAir: 1
```

## Additional context
No additional context. Unhappy with this Issue Bot.

Smanar, 2020-11-29:
Lol, yeah the bot isn't realy friendly ^^.

> This only happens if I activate a scene in home Assistant with 8 Lamps

The scene is a zigbee scene (with zigbee group) or 100% managed by HA ?
And it happen suddenly ? no problem before your firmware update ?

Fruchuxs, 2020-11-29:
Hello! It's 100% managed by HA. No problem before the firmware update. There are more lamps in the network since the update. But the lamps in the log are lamps who are all in the same room with the pi.

Fruchuxs, 2020-11-29:
Added some more informations about my environment: BT and WiFi is completly disabled.
Moved the Raspberry Pi in the mid of the room, the RaspBee II lies free. Same Problem. Tried over the Phoscon Web Gui, can turn them all off with one click but not all on (but no logs, I assume this functionality is just buggy in the web gui hand has no relations to my issue here). If I activate all lights manually the log looks similiar - but without the "too many tasks"-message:
```
13:01:41:744 delay sending request 25 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0006 onAir: 1
13:01:41:744 delay sending request 27 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0008 onAir: 1
13:01:41:744 delay sending request 30 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:41:744 delay sending request 29 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:41:744 delay sending request 31 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:41:782 delay sending request 25 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0006 onAir: 1
13:01:41:782 delay sending request 27 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0008 onAir: 1
13:01:41:782 delay sending request 30 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:41:782 delay sending request 29 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:41:782 delay sending request 31 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:41:883 delay sending request 25 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0006 onAir: 1
13:01:41:883 delay sending request 27 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0008 onAir: 1
13:01:41:883 delay sending request 30 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:41:883 delay sending request 29 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:41:883 delay sending request 31 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:41:958    0x00212EFFFF058959 force poll (2)
13:01:41:982 delay sending request 27 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0008 onAir: 1
13:01:41:982 delay sending request 30 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:41:982 delay sending request 29 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:41:982 delay sending request 31 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:42:019    0x00212EFFFF058959 force poll (2)
13:01:42:079    0x00212EFFFF058959 force poll (2)
13:01:42:082 delay sending request 29 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:42:082 delay sending request 31 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:42:145    0x00212EFFFF058959 force poll (2)
13:01:42:146 delay sending request 31 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:42:182 delay sending request 31 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:42:182 delay sending request 34 dt 0 ms to 0x001788010473AFE2, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:42:182 delay sending request 36 dt 0 ms to 0x001788010473AFE2, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:42:236    0x00212EFFFF058959 force poll (2)
13:01:42:282 delay sending request 34 dt 0 ms to 0x001788010473AFE2, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:42:282 delay sending request 36 dt 0 ms to 0x001788010473AFE2, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:42:291    0x001788010473AFE2 force poll (2)
13:01:42:304    0x00212EFFFF058959 force poll (2)
13:01:42:304 delay sending request 36 dt 0 ms to 0x001788010473AFE2, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:42:364    0x001788010473AFE2 force poll (2)
13:01:42:426    0x001788010473AFE2 force poll (2)
13:01:42:581    0x0017880106EF280F force poll (2)
13:01:42:582 delay sending request 41 dt 0 ms to 0x0017880106EF280F, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:42:647    0x0017880106EF280F force poll (2)
13:01:42:707    0x0017880106EF280F force poll (2)
13:01:43:078    0x001788010437D103 force poll (2)
13:01:43:084 delay sending request 48 dt 0 ms to 0x001788010437D103, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:43:169    0x001788010437D103 force poll (2)
13:01:43:264    0x001788010437D103 force poll (2)
13:01:43:482 delay sending request 55 dt 0 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:43:482 delay sending request 57 dt 0 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:43:582 delay sending request 55 dt 0 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:43:582 delay sending request 57 dt 0 ms to 0x001788010473AAF1, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:43:613    0x001788010473AAF1 force poll (2)
13:01:43:675    0x001788010473AAF1 force poll (2)
13:01:43:734    0x001788010473AAF1 force poll (2)
13:01:44:082 delay sending request 65 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:44:082 delay sending request 67 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:44:182 delay sending request 65 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0006 onAir: 1
13:01:44:182 delay sending request 67 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:44:263    0x00178801040423C5 force poll (2)
13:01:44:282 delay sending request 67 dt 0 ms to 0x00178801040423C5, ep: 0x0B cluster: 0x0008 onAir: 1
13:01:44:353    0x00178801040423C5 force poll (2)
[... cut ...]
13:03:12:859    0x001788010437D103 force poll (2)
13:03:13:417 Current channel 25
13:03:13:433 Device TTL 2965 s flags: 0x7
13:03:13:671 send move to color temperature 267 to 0x00212EFFFF058959
13:03:13:671 delay sending request 186 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0006 onAir: 1
13:03:13:671 delay sending request 188 dt 0 ms to 0x00212EFFFF058959, ep: 0x0B cluster: 0x0008 onAir: 1
13:03:13:671 delay sending request 191 dt 0 ms to 0x00212EFFFF058959, ep: 0x0A cluster: 0x0008 onAir: 1
```

Fruchuxs, 2020-11-29:
In short: the RaspBee II seems overwhelmed. Not good, the Hue Bridge had not such problems.

Fruchuxs, 2020-11-29:
I have the following questions:
- What means "delay"?
  - The Medium is Busy? (liste-before-speak)
  - The UART from the Pi is overwhelmed?
  - The RaspBee is overwhelmed?
  - Or just an info?!
- What means "too many tasks"?
  - The Medium is Busy? (liste-before-speak)
  - The UART from the Pi is overwhelmed?
  - The RaspBee is overwhelmed?
  - Or just an info?!

Smanar, 2020-11-29:
Can you make a try using zigbee group ?
Make a zigbee group with your all 8 light, and instead of sending 8 commands, send only one to group ?

Fruchuxs, 2020-11-29:
The non responsing FLS-PP lp LED Controller was a Bug in HA - the warmwhite part was not reacting because the scene in HA don't saved it.

But the slow response is still a problem and I want to know, which limitations the RaspBee has and why, to understand and analyze my network.

Fruchuxs, 2020-11-29:
> Can you make a try using zigbee group ?
> Make a zigbee group with your all 8 light, and instead of sending 8 commands, send only one to group ?

Phoscon Scenes doesn't produces log entries.

Smanar, 2020-11-29:
I don't understand ? you are using zigbee scene too ?

Fruchuxs, 2020-11-29:
I don't know how "zigbee scenes" works. I maked a scene directly in the phoscon web interface and this worked fine. Did you mean something  else?

Smanar, 2020-11-30:
Yes and no ^^
There is something faster than scene, just "group". With that you can send only 1 request instead of 8, realy usefull to avoid spam.

macmanroy, 2020-12-01:
I'm seeing the same issues when using scenes created with home assistant. If I create an identical scene in phoscon the "delay sending request" message is gone. Can anyone explain how home assistant and phoscon differ in the way they recall scenes?

SwoopX, 2020-12-01:
That's what Smanar is trying to explain. Phoscon makes use of native zigbee scenes here, which HA doesn't.

As an example: Think of a scene which would require 3 state changes on 5 devices:
Phoscon - 1 Command, doing everything simultaneously
HA - 15 commands, doing everything consecutively

Mimiix, 2020-12-01:
> 01:10:26:253 0x001788010432F989 error APSDE-DATA.confirm: 0xE1 on task

Nobody noticed the 0xE1 on first post?

This seems like interference. @Fruchuxs Could you try and unplug the USB that's in it and see then?

macmanroy, 2020-12-01:
@SwoopX Thanks for the explanation, so I assume a zigbee scene stores the state on the end device and then phoscon sends a single group command to recall that scene?

Fruchuxs, 2020-12-01:
> > 01:10:26:253 0x001788010432F989 error APSDE-DATA.confirm: 0xE1 on task
> 
> Nobody noticed the 0xE1 on first post?
> 
> This seems like interference. @Fruchuxs Could you try and unplug the USB that's in it and see then?

This was another problem I fixed. Moved the RaspBee away from other cables (HDMI + DP).

I'm happy to know another feature of phoscon, but this "busy" and "too many tasks" behavior is new. I thought new Lamps in my living room overwhelmed the RaspBee, but I noticed this behavior also in my bedroom with four lamps. This four lamps have always been paired with the RaspBee.

I think this is a problem with a deconz / fw update which should be investigated,
I can't believe, that the RaspBee can't control 4 lamps + 4 color move commands at the same time. Can someone tell me, which conditions in the code triggers this "delay" message and force pull? Maybe I can change something ...

This is the log with four lamps:
```
21:11:39:667 dev /dev/ttyAMA0
21:11:39:667 GW firmware version: 0x26660700
21:11:40:375 send move to color temperature 454 to 0x00178801029A3586
21:11:40:399 send move to color temperature 454 to 0x001788010434AEAA
21:11:40:399 delay sending request 146 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:399 delay sending request 148 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:399 delay sending request 149 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:408 send move to color temperature 454 to 0x001788010436152F
21:11:40:408 delay sending request 146 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:408 delay sending request 148 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:408 delay sending request 149 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:408 delay sending request 151 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:408 delay sending request 153 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:408 delay sending request 154 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:423 send move to color temperature 500 to 0x001788010432F989
21:11:40:424 delay sending request 146 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:424 delay sending request 148 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:424 delay sending request 149 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:424 delay sending request 151 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:424 delay sending request 153 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:424 delay sending request 154 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:424 delay sending request 156 dt 0 ms to 0x001788010436152F, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:424 delay sending request 158 dt 0 ms to 0x001788010436152F, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:424 delay sending request 159 dt 0 ms to 0x001788010436152F, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:454 delay sending request 146 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:454 delay sending request 148 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:454 delay sending request 149 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:454 delay sending request 151 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:454 delay sending request 153 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:454 delay sending request 154 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:454 delay sending request 156 dt 0 ms to 0x001788010436152F, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:455 delay sending request 158 dt 0 ms to 0x001788010436152F, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:455 delay sending request 159 dt 0 ms to 0x001788010436152F, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:455 delay sending request 161 dt 0 ms to 0x001788010432F989, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:455 delay sending request 163 dt 0 ms to 0x001788010432F989, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:455 delay sending request 164 dt 0 ms to 0x001788010432F989, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:497    0x001788010434AEAA force poll (2)
21:11:40:497 delay sending request 146 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:497 delay sending request 148 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:497 delay sending request 149 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:553 delay sending request 146 dt 1 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0006 onAir: 1
21:11:40:553 delay sending request 148 dt 1 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:553 delay sending request 149 dt 1 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:553 delay sending request 153 dt 1 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:554 delay sending request 154 dt 1 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:554 delay sending request 156 dt 1 ms to 0x001788010436152F, ep: 0x0B cluster: 0x0006 onAir: 1
[... cut ...]
21:11:40:753 delay sending request 148 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0008 onAir: 1
21:11:40:753 delay sending request 149 dt 0 ms to 0x00178801029A3586, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:753 delay sending request 154 dt 0 ms to 0x001788010434AEAA, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:753 delay sending request 159 dt 0 ms to 0x001788010436152F, ep: 0x0B cluster: 0x0300 onAir: 1
21:11:40:753 delay sending request 163 dt 0 ms to 0x001788010432F989, ep: 0x0B cluster: 0x0008 onAir: 1
```

Fruchuxs, 2020-12-01:
Btw.: I need to enable phoscon scenes twice. The first command is never been executed,
Noticed this behavior in the deconz Web Gui, but also if I activate the phoscon scene through home assistant.

Fruchuxs, 2020-12-05:
The Issue is still relevant. I also think it relates to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3735#issuecomment-735398061

Smanar, 2020-12-06:
>I can't believe, that the RaspBee can't control 4 lamps + 4 color move commands at the same time

I m agree but atm as workaround try to use zigbee group.

hdejongh, 2020-12-30:
i have exactly the same issue. Its resolved by using deconz (zigbee) groups!
I had a lot of delay messages aswell and unresponsive lights. With zigbee groups it works instantly now.

Fruchuxs, 2021-01-07:
Yeah, zigbee groups are cool, but this just fight symptoms, not the problem itself. If I want to turn of the lights in ONE Room (8 Lamps), I got the same delay problems. Even worse: sometime some lamps stay on (also the Dresden electronic FLS-PP lp: color turned off, white stays on?!). 

But it feels like, that the devs of the firmeware doesn't see any problems. This means for me, that I hit limitations of the hardware, and I can't recommend deconz anymore.
Personally I think there are race conditions in the API, a smart scheduling of command would maybe help, but who knows.

closed 2021-03-05 by github-actions[bot]