dresden-elektronik/deconz-rest-plugin #3777: [Jeedom Deconz plugin] not recognised as Hue Bridge by official Harmony and Hue apps

ArnoM9, 2020-11-30:
## Describe the question or issue you are having

Hi,
I am using a conbee II dongle with deconz plugin in Jeedom on a Raspberry PI 3B+. Everything is working fine in Jeedom, my zigbee and hue bulbs are included and I can also pair the gateway to hueessentials app.
However I would like also to pair the gateway with Hue and Logitech Harmony official  apps. I tried to authenficate and pair the gateway without success : no "Hue bridge" is found.
I search (a lot) and I might have found that these apps are interrogating port 8080 of the gateway for pairing while the  configured gateway port in jeedom is 8484. I tried to change and redirect it without success.

So, I would like to know if my assumptions are correct and if there is a workaround to pair also with Hue and Harmony official apps.

Thank you
Kind regards/MfG

Arno

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Mimiix, 2020-11-30:
Hi!

It is needed to run on 8080 for those apps. I can say it works for Harmony. However, after a change on the Hue side, the hue app doesn't work anymore. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2852

ArnoM9, 2020-11-30:
Hello,
Thank you very much. So my assumption was correct...
How can I change the port to 8080 in the jeedom deconz plugin configuration page ?
I tried without success.
Thanks again

Best regards/MfG

Arno

Mimiix, 2020-11-30:
Hi

Unfortunately, i have no clue. I am not familiar with Jeedom :( 

Kind regards,

Smanar, 2020-11-30:
What happen if you change the setting here ?

[image]

You probably need to restart the "demon" after that.
To check if it work, try to access to phoscon.

ArnoM9, 2020-11-30:
Hello,
I already tried this step but then the gateway is unreachable. In fact 8484 port is hardcoded during local installation of the deconz plugin in Jeedom.
I think I will try to uninstall on local and try to install again directly on the raspberry which hosts my jeedom server without using the plugin. Hopefully I will be able to configure the port number.
Thank you

Smanar, 2020-11-30:
> I already tried this step but then the gateway is unreachable

It mean it  works, no ?

But yes it s probably hard coded, do you have take a look in the file deconz.service

`sudo nano /lib/systemd/system/deconz.service`

But If I remember jeedom don't use service but run deconz with command line ? The problem is files are not open source, so I realy don't know how they do their magic ...

ArnoM9, 2020-12-01:
Hi
Thanks for your feedback.
I finally installed deconz directly on the raspberry (port 8080). I am now able to pair harmony app. However, I still cannot pair the Hue app, the gateway is not recognized.
Do you know if there is a specific pairing operation in order to have my gateway recognized by Hue app ?
I read an older comment in the forum that a SSL connection was necessary between the app and the gateway. I do not know if if is the case...
Thank you again for your support

Arno

Mimiix, 2020-12-01:
I think the Hue app won't work anymore, as they changed something on their end.

ArnoM9, 2020-12-01:
Hello,
I already tried this step but then the gateway is unreachable. In fact 8484
port is hardcoded during local installation.
I think I will try to uninstall on local and try to install again on the
raspberry which hosts my jeedom server.
Thank you

closed 2020-12-31 by stale[bot]