dresden-elektronik/deconz-rest-plugin #3785: Aqara T1 with neutral wire and consuming report DLKZMK11LM

MiAutomations, 2020-11-30:
Hello Team this device is not reporting the measuring data, the device is detected and the switch is working as suppose.
Can you please add the measuring feature to this device? Thank you for your support   

- Product name: Aqara Single Switch Module T1 (With Neutral)
- Manufacturer: Lumi
- Model identifier: lumi.switch.n0agl1
- Device type : 
  - Switch

## Screenshots

[image]
[image]
[image]

SwoopX, 2020-11-30:
Can you please set up the bindings for simple metering and electrical measurement cluster as described here?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/How-to...#manually-bind-server-clusters

Then, please turn the switch on and see if values are generated at all. I don't really trust the exposed clusters by the device, we've seen too many non-working over time. I'd like to ensure that first.

MiAutomations, 2020-11-30:
I executed the binding as described and the values keep in 0 

Please let me know if you need any more info from my side, and thank you for your support

MiAutomations, 2020-12-15:
Is there any update in this device? Do you need any further information?

SwoopX, 2020-12-15:
Sorry, this one drowned somehow. Can you double check the `Analog Input (Basic)` clusters? The values of interest are typically to be found in attribute 0x0055 present value.

Please also do me the favour to right click on the light and read the simple descriptors again. Would be interested to see if electrocal measurement and simple metering clusters disappear again.

flopp999, 2020-12-15:
check this issue #3380

MiAutomations, 2020-12-18:
I just have installed the new update for the plugin  deCONZ 6.6.1 available, and I still without data from the electricity report .
And I have removed the device and re added in order to verify ... Please let me know if you need more info from my side.
Thank you and best regards

MiAutomations, 2020-12-22:
> lso do me the favour to right click on the light and read the simple descriptors again.

Hello and once again thank you for your support 

Please let me know if is this that you need/request 

[image]

Mimiix, 2020-12-22:
@MiAutomations This seems to be a duplicate of #3380. Closing this one, please continue there.

closed 2020-12-22 by Mimiix

MiAutomations, 2020-12-22:
Ok and regarding the info that I share? Is also need to be posted in the open topic?