dresden-elektronik/deconz-rest-plugin #3786: EWelink ZigBee Smart Plug

jfparis, 2020-11-30:
Hi 

Not sure whether the device is supported or partially supported. It does not appears on any list of supported devices but is somehow connected

## Device

- Product name: EWelink ZigBee Smart Plug
- Manufacturer: eWelink
- Model identifier: SA-003=Zigbee
- Device type : Please remove all unrelated device types. 
  - Switch

NB: The device is somehow connected to my conbee II but 
* identified as a "light"
* If I try to edit the name in Phoscon, it starts to switch on /off / on / off until I hit save 

## Screenshots
[image]
[image]

SwoopX, 2020-12-12:
Nothing to do for us here, the device works out of the box.

closed 2020-12-12 by SwoopX

Woll3, 2021-06-20:
Device is working (on/off) but I don't have power meter! How can I add this? Thanks!

Smanar, 2021-06-20:
I don't think it have them, no cluster for that on capture.

jfparis, 2021-06-20:
The model for which I provided the screenshots did not have a power meter in it. @Woll3 might have a new model and should probably submit a new bug / request

Woll3, 2021-06-20:
> 
> 
> The model for which I provided the screenshots did not have a power meter in it. @Woll3 might have a new model and should probably submit a new bug / request

[image]

I think that I have the same unit as you! I think I bought the wrong ones!

TischerT, 2023-09-17:
Sorry to open up this old topic.
I bought an updated version of this plug (SA-030) as the previous model does not seem to be easily available anymore.
I expected that it will just work out of the box as well, but I can't get it connected. Others seem to have it working with Smartthings and Zigbee2MQTT, but of course this does not say anything about the controller they are using.

I'm using an original Raspbee (so I hope I kind of know what I'm doing as I have plenty of other devices working) and assume it's just too old at this point.
I don't expect any work on this to integrate it, just writing down my thoughts for other users.

Smanar, 2023-09-17:
It don't work or you miss the power mesurement ?
Can you show the node picture from deconz ?

TischerT, 2023-09-17:
Thanks for the reply!
Unfortunately, no picture from deconz, because I can't get the plug to connect at all. The plug is in pairing mode, I tell phoscon to scan for new devices, but it's not picked up (neither as light nor as plug). Whatever signals the plug is sending out, the raspbee don't seem to find these valid.
I can connect other new devices to my raspbee, so it's not a problem on this end. However, I can't exclude that the plug that I got is somehow dysfunctional.

Mimiix, 2023-09-17:
> Thanks for the reply! Unfortunately, no picture from deconz, because I can't get the plug to connect at all. The plug is in pairing mode, I tell phoscon to scan for new devices, but it's not picked up (neither as light nor as plug). Whatever signals the plug is sending out, the raspbee doesn't seem to find these valid. I can connect other new devices to my raspbee, so it's not a problem on this end. However, I can't exclude that the plug that I got is somehow dysfunctional.

Can you please open an forum post about this?

Locking this up